Sloeber/arduino-eclipse-plugin #1060: error when installing latest eclipse and arduino plugin

raico-de, 2019-01-11:
If i install the plugin v4 under win64 win7 i cannot finish installation
Eclipse 12/18  (actual version)
I get this error:
Failed to configer Sloeber
Failed to download "https://eclipse.baeyens.it/download/examples_Arduino_1_6_7.zip".
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target

On every new start of eclipse i get this message again

jantje, 2019-01-11:
can you see whether this helps?
http://magicmonster.com/kb/prg/java/ssl/pkix_path_building_failed.html

closed 2019-01-19 by jantje

willemlj, 2019-09-04:
Upgrading to the latest Java Runtime solved this exact problem for me.