Sloeber/arduino-eclipse-plugin #1063: system cannot find path (windows10), recipe for target 'sloeber.ino.cpp.o' failed

EdoMon93, 2019-01-17:
After a fresh install, build for esp8266 seems not to be able to find some path.
I've checked all the paths shown in console and confirmed that all of them actually exist, except for "/bin/xtensa-lx106-elf-g++" which I'm not sure where to find.
Making a fresh project doesn't solve the problem.

Console output:
```
"C:\\Sloeber_4_2\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:/Users/Edo/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta2/tools/sdk/include" "-IC:/Users/Edo/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta2/tools/sdk/lwip2/include" "-IC:/Users/Edo/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta2/tools/sdk/libc/xtensa-lx106-elf/include" "-ID:/Dropbox/AT_EDO/Proyectos/03_Unilever_Monitoreo_Temp/Programas/hTelemetry/Arduino/HolandaLogger/Release/core" -c -Wall -Wextra -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -fexceptions -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0   -DARDUINO=10802 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC" -DLED_BUILTIN=0  -DESP8266   -I"C:\Users\Edo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\cores\esp8266" -I"C:\Users\Edo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\variants\generic" -I"C:\Users\Edo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\ESP8266HTTPClient\src" -I"C:\Users\Edo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\ESP8266httpUpdate\src" -I"C:\Users\Edo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\ESP8266WiFi\src" -I"C:\Users\Edo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\Wire" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"  -o  "sloeber.ino.cpp.o"

El sistema no puede encontrar la ruta especificada.

subdir.mk:18: recipe for target 'sloeber.ino.cpp.o' failed
make: *** [sloeber.ino.cpp.o] Error 1
```

Dropbox is not running btw.

Any idea what could be the problem?

jantje, 2019-01-17:
dropbox? I don't know what you are talking about.
And don't point to your arduino installed packages. It won't work.

closed 2019-01-17 by jantje

EdoMon93, 2019-01-17:
I followed instructions from esp8266 repository found here: [(https://arduino-esp8266.readthedocs.io/en/latest/eclipse/eclipse.html)](url)

> add as private hardware path the Part to the ESP8266, example:
> Windows: C:\Users\[username]\AppData\Roaming\Arduino15\packages\esp8266\hardware

I mentioned Dropbox because sometimes it can cause "Access Denied" problems.

To which packages should I point it then?

Cheers

jantje, 2019-01-18:
Those instructions are for another eclipse tool.
Sloeber is much more like arduino ide see more info here http://sloeber.io