Sloeber/arduino-eclipse-plugin #1064: Change board from ESP8266 to ESP32 in the same problem mess up with libraries

Menion2k, 2019-01-18:
I have a project using ESPAsyncWebserver and it's dependencies, FS, SPIFSS, uRTCLib and AdafruitGFX and SSD1306 libraries
Using few compilation switched based on the definition of "ESP8266" or "ESP32" I can compile fine the porject for both ESP8266 and ESP32 in Arduino suite.
Then if I use the same source code in sloeber 4.3 (standalone installation) I get following problems:

1) The definition of preprocessor defines ESP8266 or ESP32 stuck to the board selected during the creation of the project and does not follow the Build Target, meaning that if you create a new Build Target, in the same project, selecting a different board type, the source indexing is messed up 
2) The libraries get messed up during the switch. Apparently Sloeber is able to select the proper libraries needed, based on #includes and board type, but the compilation fails even if the library is selected. i.e.

```
In file included from ..\display.cpp:10:0:
C:\Users\silveria\sloeber\arduinoPlugin\libraries\Adafruit_SSD1306\1.2.7/Adafruit_SSD1306.h:36:17: fatal error: SPI.h: No such file or directory
compilation terminated.
subdir.mk:27: recipe for target 'display.cpp.o' failed
make: *** [display.cpp.o] Error 1
"C:/Users/silveria/sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

```

But SPI library of ESP32 is present in the project

jantje, 2019-01-18:
I'm using configurations to have different targets. Is build target the same as a configuration?
Sloeber needs to switch to the proper libraries and depending on how you switch configuration this does not always work. Right clicking the project and setting the configuration works fine.
For the indexer to take into account the configuration you need to set the indexing options to use the active configuration.

jantje, 2019-01-21:
>You mean right click->Properties->Arduino->"Arduino board selections"?

[image]

Menion2k, 2019-01-21:
Yes
That one, I made two build configuration and when switching back and forth between ESP8266 and ESP32 I get libraries messed up.
I mean, I understand that the libraries dependecy is tricky, i.e when using ESPAsyncWebserver, ESPAsyncTCP or AsyncTCP must be included based on the target.
But I normally have problems with sketch.ino included libraries, such FS.h and others

Menion2k, 2019-01-22:
Another kind of problem I normally hit when switching active configuration is this one (note: I run sloeber 4.3 standalone on Windows)

```
"C:\\Users\\silveria\\sloeber\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\display.cpp'
'Starting C++ compile'
"C:\Users\silveria\sloeber\arduinoPlugin\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/config" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/app_trace" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/app_update" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/asio" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/bootloader_support" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/bt" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/coap" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/console" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/driver" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp-tls" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp32" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp_adc_cal" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp_event" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp_http_client" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp_http_server" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp_https_ota" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp_https_server" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp_ringbuf" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/ethernet" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/expat" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/fatfs" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/freemodbus" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/freertos" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/heap" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/idf_test" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/jsmn" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/json" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/libsodium" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/log" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/lwip" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/mbedtls" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/mdns" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/micro-ecc" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/mqtt" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/newlib" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/nghttp" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/nvs_flash" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/openssl" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/protobuf-c" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/protocomm" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/pthread" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/sdmmc" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/smartconfig_ack" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/soc" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/spi_flash" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/spiffs" "-IC:\Users\silveria\sloeber\/arduinoPlugin/packages/esp32/hardware/e [... cut ...]
xtensa-esp32-elf-g++: error: unrecognized command line option '-C:\Users\silveria\sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.1\libraries\FS\src'
subdir.mk:33: recipe for target 'display.cpp.o' failed
make: *** [display.cpp.o] Error 1
"C:/Users/silveria/sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

```

Apparently the generated compile command is malformed, but looking to the command itself it seems ok. It is not always at the -I options, it popup randomly at every gcc options. It changes based on "Clean Project" and eventually disappear but without any clear and deterministic action

jantje, 2019-01-22:
Attaching and reattaching libraries is a hard thing-amongst others due to architectural decisions made by arduino-.
My use case for switching configurations always implied to stay on a configuration for a while. In that use case deleting the libraries folder and let Sloeber attach the libraries is not a perfects but a "acceptable solution".

If I would want to build the same source code for 2 boards (without the hassle of having Sloeber reattach the libraries all the time) I would go for 2 projects and linking the code from one project to the other.
This takes away complexity for the indexer (2 separate indexes instead of 1 index with 2 configurations) and Sloeber (Reattaching libraries)
This does imply that each time you add a file to your project you also need to add a link to it from the other project. None ino files can be in a subfolder which allows you to link to the folder taking away the need to relink.

Menion2k, 2019-01-22:
This is actually my workaround :)
Bye

jantje, 2019-01-22:
Then I close this issue.

closed 2019-01-22 by jantje

Menion2k, 2019-01-23:
@jantje anyhow if I were you, I would give a check to what happens with ESP32 Arduino core 1.0.1.
Because even if with a fresh created project with ESP32 target, I always have problem setting and adding libraries.
I need to remove everything, wait for Sloeber to automatically pick up some libraries and then add, all at the same time, the remaining libraries, otherwise I always get in trouble with missing header. And also, when adding during the development a library, it force me to wipe and re-add all together the libraries, otherwise I end up in some missing header.
I believe there is some problem with the gcc invoke command creation when using the esp32 toolchain

jantje, 2019-01-23:
The auto add library function is really "magic"
Years ago I discussed with arduino the way arduino ide adds libraries; because I think it is just crazy.
They look for all header files in all libraries and try to do a match based on header files. The more libraries you use and the more matches of header names between core and libraries the more "unmaintainable" this becomes.
ESP boards are especially vulnerable because of the include wifi.h which exists for esp breakout boards libraries and as part of the core. Once a wrong library attached things only get worse.
That being said I know for sure "something is wrong" after project creation (the same code is used with project properties->arduino) where the indexer is kicking in to soon resulting in missing headers.
The easy way around is turning automatically adding headers off in preferences->arduino.
As to fixing this... why would I?...everyone is asking for help, fixes or improvements but hardly anyone steps in to help/support. Most "requesters" disappear in the masses once their issue is fixed.
I have given a lot of my time and energy for free to help others and get very little in return. There is a point in time when it just is no longer financial viable to continue. Many people have been in this position before me. And many are in the same position. It is up to you to choose in which world you want to live.

Menion2k, 2019-01-24:
Yes I have read your comments and I understand you want a return from your work, but I don't want to comment on this.
My point is that I do not believe the issue is with libraries pickup but there is a problem with the generation of the g++ command when using the ESP32 core (ver 1.0.1 at least)
Bye