Sloeber/arduino-eclipse-plugin #1066: Uploading to nRF52840-DK gives a 'programmer is not responding' error.

BitLox, 2019-01-28:
I'm working with:

nRF52840-DK board
Arduino IDE 1.8.8
Sloeber 4.3.1
development on macOS 10.14.2
It fails when uploading the firmware to the nRF52 via /dev/cu.usbmodem14101 serial port:

`Starting upload
Uploading project "dbdfgb" with "NRFUTIL"
Forcing reset using 1200bps open/close on port : "/dev/cu.usbmodem14101"
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
PORTS { /dev/cu.usbmodem14101,} / { /dev/cu.usbmodem14101,} => {}
Com port is not behaving as expected
Using port "/dev/cu.usbmodem14101" from now onwards
Ending reset

Launching: adafruit-nrfutil dfu serial -pkg /Users/player1/Documents/sloeber-workspace/dbdfgb/Release/dbdfgb.zip -p /dev/cu.usbmodem14101 -b 115200 --singlebank 
Output:
**Cannot run program "adafruit-nrfutil" (in directory "/Users/player1/Documents/sloeber-workspace"): error=2, No such file or directory**
failed to upload "L/dbdfgb/${workspace_loc:/dbdfgb}/Release/dbdfgb.hex"
`

If I extract the command, touch the port, and run manually:
`
SquidBook-Pro:~ player1$ stty -f /dev/cu.usbmodem14101 1200 
SquidBook-Pro:~ player1$ adafruit-nrfutil dfu serial -pkg /Users/player1/Documents/sloeber-workspace/dbdfgb/Release/dbdfgb.zip -p /dev/cu.usbmodem14101 -b 115200 --singlebank
Upgrading target on /dev/cu.usbmodem14101 with DFU package /Users/player1/Documents/sloeber-workspace/dbdfgb/Release/dbdfgb.zip. Flow control is disabled, Single bank, Touch disabled
########################################
###################
[... 20 lines cut ...]
PORTS {/dev/cu.AirPods-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.Squid5-WirelessiAP, /dev/cu.SquidX-WirelessiAPv2, /dev/cu.usbmodem14101, /dev/tty.AirPods-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.Squid5-WirelessiAP, /dev/tty.SquidX-WirelessiAPv2, /dev/tty.usbmodem14101, } / {/dev/cu.AirPods-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.Squid5-WirelessiAP, /dev/cu.SquidX-WirelessiAPv2, /dev/cu.usbmodem14101, /dev/tty.AirPods-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.Squid5-WirelessiAP, /dev/tty.SquidX-WirelessiAPv2, /dev/tty.usbmodem14101, } => {}
PORTS {/dev/cu.AirPods-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.Squid5-WirelessiAP, /dev/cu.SquidX-WirelessiAPv2, /dev/cu.usbmodem14101, /dev/tty.AirPods-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.Squid5-WirelessiAP, /dev/tty.SquidX-WirelessiAPv2, /dev/tty.usbmodem14101, } / {/dev/cu.AirPods-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.Squid5-WirelessiAP, /dev/cu.SquidX-WirelessiAPv2, /dev/cu.usbmodem14101, /dev/tty.AirPods-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.Squid5-WirelessiAP, /dev/tty.SquidX-WirelessiAPv2, /dev/tty.usbmodem14101, } => {}
PORTS {/dev/cu.AirPods-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.Squid5-WirelessiAP, /dev/cu.SquidX-WirelessiAPv2, /dev/cu.usbmodem14101, /dev/tty.AirPods-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.Squid5-WirelessiAP, /dev/tty.SquidX-WirelessiAPv2, /dev/tty.usbmodem14101, } / {/dev/cu.AirPods-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.Squid5-WirelessiAP, /dev/cu.SquidX-WirelessiAPv2, /dev/cu.usbmodem14101, /dev/tty.AirPods-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.Squid5-WirelessiAP, /dev/tty.SquidX-WirelessiAPv2, /dev/tty.usbmodem14101, } => {}
PORTS {/dev/cu.AirPods-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.Squid5-WirelessiAP, /dev/cu.SquidX-WirelessiAPv2, /dev/cu.usbmodem14101, /dev/tty.AirPods-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.Squid5-WirelessiAP, /dev/tty.SquidX-WirelessiAPv2, /dev/tty.usbmodem14101, } / {/dev/cu.AirPods-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.Squid5-WirelessiAP, /dev/cu.SquidX-WirelessiAPv2, /dev/cu.usbmodem14101, /dev/tty.AirPods-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.Squid5-WirelessiAP, /dev/tty.SquidX-WirelessiAPv2, /dev/tty.usbmodem14101, } => {}
PORTS {/dev/cu.AirPods-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.Squid5-WirelessiAP, /dev/cu.SquidX-WirelessiAPv2, /dev/cu.usbmodem14101, /dev/tty.AirPods-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.Squid5-WirelessiAP, /dev/tty.SquidX-WirelessiAPv2, /dev/tty.usbmodem14101, } / {/dev/cu.AirPods-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.Squid5-WirelessiAP, /dev/cu.SquidX-WirelessiAPv2, /dev/cu.usbmodem14101, /dev/tty.AirPods-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.Squid5-WirelessiAP, /dev/tty.SquidX-WirelessiAPv2, /dev/tty.usbmodem14101, } => {}
PORTS {/dev/cu.AirPods-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.Squid5-WirelessiAP, /dev/cu.SquidX-WirelessiAPv2, /dev/cu.usbmodem14101, /dev/tty.AirPods-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.Squid5-WirelessiAP, /dev/tty.SquidX-WirelessiAPv2, /dev/tty.usbmodem14101, } / {/dev/cu.AirPods-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.Squid5-WirelessiAP, /dev/cu.SquidX-WirelessiAPv2, /dev/cu.usbmodem14101, /dev/tty.AirPods-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.Squid5-WirelessiAP, /dev/tty.SquidX-WirelessiAPv2, /dev/tty.usbmodem14101, } => {}
PORTS {/dev/cu.AirPods-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.Squid5-WirelessiAP, /dev/cu.SquidX-WirelessiAPv2, /dev/cu.usbmodem14101, /dev/tty.AirPods-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.Squid5-WirelessiAP, /dev/tty.SquidX-WirelessiAPv2, /dev/tty.usbmodem14101, } / {/dev/cu.AirPods-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.Squid5-WirelessiAP, /dev/cu.SquidX-WirelessiAPv2, /dev/cu.usbmodem14101, /dev/tty.AirPods-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.Squid5-WirelessiAP, /dev/tty.SquidX-WirelessiAPv2, /dev/tty.usbmodem14101, } => {}
[... cut ...]
Upgrading target on /dev/cu.usbmodem14101 with DFU package /private/var/folders/zm/4nmmy__d0n5_m634_w_hm_1w0000gp/T/arduino_build_912906/Blink.ino.zip. Flow control is disabled, Single bank, Touch disabled
Opened serial port /dev/cu.usbmodem14101
Starting DFU upgrade of type 4, SoftDevice size: 0, bootloader size: 0, application size: 29580
Sending DFU start packet
Sending DFU init packet
Sending firmware file
########################################
##################
Activating new firmware

Device programmed.
DFU upgrade took 2.6008460521698s`

jantje, 2019-01-28:
I see 2 issues
1)  Your platform is not properly installed. **Do not point to the arduino installed platforms**
2) the reset seems to fail.

BitLox, 2019-01-28:
1. I did not point the sloeber install at any particular Arduino install. Shall I download and install an older version of Arduino?

2. I’ll see if while it is fumbling about if I can manually reset the port. Where is the code that handles the serial port reset? Is it analogous to the stty reset I used?

jantje, 2019-01-28:
1)
As from your logs:
Arduino tries to upload without fqn to  adafruit-nrfutil 
`/Users/player1/Library/Arduino15/packages/adafruit/hardware/nrf52/0.9.3/tools/adafruit-nrfutil/macos/adafruit-nrfutil`
Where sloeber tries to upload without fqn to  adafruit-nrfutil 
`Launching: adafruit-nrfutil dfu serial -pkg /Users/player1/Documents/s...`
Resulting in Sloeber not finding adafruit-nrfutil 
`**Cannot run program "adafruit-nrfutil" (in directory "/Users/player1/Documents/sloeber-workspace"): error=2, No such file or directory**`

This is typically caused by not having installed the platform using the Sloeber boards manager.

As to using an older Sloeber version...the older the less chance exotic boards will work.

jantje, 2019-01-28:
> Is it analogous to the stty reset I used?

No, Here is the code
https://github.com/Sloeber/arduino-eclipse-plugin/blob/479f1514c427a8e0b73eaa91b001fc49bdf7f5c4/io.sloeber.core/src/io/sloeber/core/communication/ArduinoSerial.java#L37

jantje, 2019-01-29:
Can you provide:
The package json url you used
The platform and version number you used
The board name used 
Image op project properties->arduino
image of preferences->arduino
image of preferences->arduino->platform and boards
image of preferences->arduino->third party url's

BitLox, 2019-01-29:
Sure thing:

https://adafruit.github.io/arduino-board-index/package_adafruit_index.json

Adafruit nRF52 0.9.3
[image]

nRF52840-DK
[image]

[image]

[image]

[image]

[image]

jantje, 2019-01-29:
I tried it here using the smae board/version and I get the fqn
```
Launching: C:\sloeber\/arduinoPlugin/packages/adafruit/hardware/nrf52/0.9.3/tools/adafruit-nrfutil/win32/adafruit-nrfutil.exe dfu serial -pkg C:\workspaces\embedded\issue1066/Release/issue1066.zip -p com1 -b 115200 --singlebank 
Output:
Upgrading target on com1 with DFU package C:\workspaces\embedded\issue1066\Release\issue1066.zip. Flow control is disabled, Single bank, Touch disabled
```
I do not have the board so the upload fails.
It is strange you do not get the fqn.

BitLox, 2019-01-29:
Sorry to be a doofus, what is an fqn?

BitLox, 2019-01-29:
Ok, I did some experimenting.

First off, on the Mac the adafruit-nrfutil throws an error if the environment does not have the locale set.
[http://click.pocoo.org/python3/](http://click.pocoo.org/python3/)

This was hosing up the compile, so I inserted the following in the _platform.txt:100_
```
## Create dfu package zip file
#recipe.objcopy.zip.pattern="{tools.nrfutil.cmd}" dfu genpkg --dev-type 0x0052 --sd-req {build.sd_fwid} --application "{build.path}/{build.project_name}.hex" "{build.path}/{build.project_name}.zip"
recipe.objcopy.zip.pattern=export LC_ALL=en_US.UTF-8 && export LANG=en_US.UTF-8 && "{tools.nrfutil.cmd}" dfu genpkg --dev-type 0x0052 --sd-req {build.sd_fwid} --application "{build.path}/{build.project_name}.hex" "{build.path}/{build.project_name}.zip"
```

This gets me past the compile no problem.
(I did not think this relevant, and was going to open a minor issue, but wait...!)

Now, if I try hard coding the fqn for the adafruit-nrfutil, _platform.txt:129_
```
#tools.nrfutil.cmd=adafruit-nrfutil
tools.nrfutil.cmd=/Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/packages/adafruit/tools/adafruit-nrfutil/macos/adafruit-nrfutil
```

adafruit-nrfutil **executes** but coughs up the same bug as when the original object combination command ran:
```
Launching: /Applications/sloeber.app/Contents/Eclipse//arduinoPlugin/packages/adafruit/tools/adafruit-nrfutil/macos/adafruit-nrfutil dfu serial -pkg /Users/player1/Documents/sloeber-workspace/blinker/Release/blinker.zip -p /dev/cu.usbmodem14101 -b 115200 --singlebank 
Output:
[6822] Failed to execute script __main__
Traceback (most recent call last):
  File "nordicsemi/__main__.py", line 315, in <module>
  File "click/core.py", line 722, in __call__
  File "click/core.py", line 676, in main
  File "click/_unicodefun.py", line 118, in _verify_python3_env
RuntimeError: Click will abort further execution because Python 3 was configured to use ASCII as encoding for the environment.  Consult http://click.pocoo.org/python3/for mitigation steps.

This system lists a couple of UTF-8 supporting locales that
you can pick from.  The following suitable locales where
discovered: af_ZA.UTF-8, am_ET.UTF-8, be_BY.UTF-8, bg_BG.UTF-8, ca_ES.UTF-8, cs_CZ.UTF-8, da_DK.UTF-8, de_AT.UTF-8, de_CH.UTF-8, de_DE.UTF-8, el_GR.UTF-8, en_AU.UTF-8, en_CA.UTF-8, en_GB.UTF-8, en_IE.UTF-8, en_NZ.UTF-8, en_US.UTF-8, es_ES.UTF-8, et_EE.UTF-8, eu_ES.UTF-8, fi_FI.UTF-8, fr_BE.UTF-8, fr_CA.UTF-8, fr_CH.UTF-8, fr_FR.UTF-8, he_IL.UTF-8, hr_HR.UTF-8, hu_HU.UTF-8, hy_AM.UTF-8, is_IS.UTF-8, it_CH.UTF-8, it_IT.UTF-8, ja_JP.UTF-8, kk_KZ.UTF-8, ko_KR.UTF-8, lt_LT.UTF-8, nl_BE.UTF-8, nl_NL.UTF-8, no_NO.UTF-8, pl_PL.UTF-8, pt_BR.UTF-8, pt_PT.UTF-8, ro_RO.UTF-8, ru_RU.UTF-8, sk_SK.UTF-8, sl_SI.UTF-8, sr_YU.UTF-8, sv_SE.UTF-8, tr_TR.UTF-8, uk_UA.UTF-8, zh_CN.UTF-8, zh_HK.UTF-8, zh_TW.UTF-8
The execution of command "adafruit-nrfutil" is done.
```

Aha! proof that now at least it is _TRYING_ to run the command, and sure seems like yea, before it somehow was not executing.

Now the question is, how can we "prepend" the 
`export LC_ALL=en_US.UTF-8 `
and 
`export LANG=en_US.UTF-8 `
to the command?
Just adding them there in the platform as i did for the combiner does not work.

jantje, 2019-01-30:
for the build you can add your environment variables to preferences->C/C++->build->build variables.
Or to project properties->C/C++->build->build variables if you want them project specific.
I'm not sure these will be taken into account when running the uploader though.
But you could set them before starting sloeber

BitLox, 2019-01-30:
Tried both methods, no change.

jantje, 2019-01-30:
Did you delete the release folder before trying? The makefile needs to be recreated or the changes will not be taken into account.

BitLox, 2019-01-30:
Every time I try a new method I use a completely new project.

jantje, 2019-01-30:
> Every time I try a new method I use a completely new project.

Sounds like a good idea.
But setting environments should work as Sloeber heavily relies on that and it has been used as a workaround many times.
So I guess something somewhere is not working.

closed 2019-02-10 by jantje