Sloeber/arduino-eclipse-plugin #1068: Better/uptodate documentation

danobot, 2019-02-06:
I have watched a few of Jan's tutorial videos on Youtube, which are unfortunately out of date.
Sloeber 4.3 bundles lots of things already and makes setup easier, however this information and instructions is not easily accessible.
May I suggest a documentation repository using something like Mkdocs that can be kept up to date with new releases of sloeber?
The benefit of written documentation is:
* easier to update
* searchable
* google indexable
* people can merge improvements and corrections.

There are a ton of small annoying problems I encountered following Jans videos, some of which show stoppers.
I'm a developer myself, and appreciate all the work you put in. Really want to see this become the best IDE for arduino!

jantje, 2019-02-06:
Feel free.
I know many of the video's are partially outdated, however they still contain lots of valuable info.
The sloeber.io website is part of the git repository and can already be modified by the "community".
I don't know this MKDocs things but a quick look told me it is md based. Plenty of the sloeber.io pages are already md based :-)
The current publishing happens with a nightly copy from git to my web service provider. 
How would this MKDocs get published?

danobot, 2019-02-06:
You run a command in your local to generate static html. The tool commits it to a github-pages branch. On GitHub you enable Pages and you get free hosting of documentation

jantje, 2019-02-07:
That would then be another location of documentation.
I'm not sure I like that.
As it is static html I could push it to sloeber.io

danobot, 2019-02-07:
Yeah absolutely, GitHub pages was just a suggestion. If you have a dedicated webserver, there is no need for another location. How should we proceed?

jantje, 2019-02-07:
The dedicated server is http://sloeber.io which redirects to http://eclipse.baeyens.it/
The content of that site is here on github https://github.com/Sloeber/arduino-eclipse-plugin/tree/master/website/WebContent
Anything under that folder is pushed to the server on a nightly basis by the build engine.
I guess the best way forward is that you take a decent look at what is available and what needs updating.

jantje, 2019-02-23:
Any progress?

danobot, 2019-02-23:
No, my priorities have changed.

jantje, 2019-02-23:
Thanks for the update
I'll close the issue then

closed 2019-02-23 by jantje