Sloeber/arduino-eclipse-plugin #1069: Avrdude settings when uploading to attiny85

hreintke, 2019-02-08:
Hi,

I am trying to get upload to attiny85 working with an arduino nano as the programmer.

When using Arduino IDE (which upload OK), I get this output.
```
C:\Utilities\arduino-1.6.13\hardware\tools\avr/bin/avrdude -CC:\Utilities\arduino-1.6.13\hardware\tools\avr/etc/avrdude.conf -v -pattiny85 -cstk500v1 -PCOM3 -b19200 -Uflash:w:C:\Users\Herman\AppData\Local\Temp\arduino_build_447789/sketch_feb08a.ino.hex:i 

avrdude: Version 6.3, compiled on Sep 12 2016 at 17:24:16
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch
```
When using Sloeber I get 
```
Starting upload
Uploading project "Attiny85_I2C" with "AVRDUDE"
no reset because we are using a programmer

Launching: C:\Utilities\sloeber\arduinoPlugin\packages\arduino\tools\avrdude\6.3.0-arduino14/bin/avrdude -CC:\Utilities\sloeber\arduinoPlugin\packages\arduino\tools\avrdude\6.3.0-arduino14/etc/avrdude.conf -pattiny85 -cstk500v1 -PCOM3 -Uflash:w:C:\Userdata\ESP8266\Projects\Attiny85_I2C/Release/Attiny85_I2C.hex:i 
Output:
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x03
```

The only difference in avrdude is the -v and the -b19200 in the commandline.

Is there a way to include these options in the command by updating settings/file so i can test whether that works for correct upload ?

jantje, 2019-02-08:
The -v is verbose so that should not be a problem.
The -b19200 is the baudrate to use and can be the root cause.
Normally Sloeber should give the exact same command as arduino IDE. Are you sure you use the same platform version for both?

hreintke, 2019-02-08:
I know -v is verbose, but could give additional info on the issue.

Same version : 
[image]

[image]

jantje, 2019-02-08:
I tried here
Launching: C:\sloeber\arduinoPlugin\packages\arduino\tools\avrdude\6.3.0-arduino14/bin/avrdude -CC:\sloeber\arduinoPlugin\packages\arduino\tools\avrdude\6.3.0-arduino14/etc/avrdude.conf -pattiny85 -cstk500v1 -Pcom1 -b19200 -D -Uflash:w:C:\workspaces\embedded\issue1069/Release/issue1069.hex:i 
Output:

-b19200 is there on my system

hreintke, 2019-02-09:
There is also an additional -D in the command.

- Wich programmer did you select ?
- From which file are these options taken to create the avrdude command ?

closed 2019-02-09 by hreintke

jantje, 2019-02-09:
As you didn't specify the upload protocol you used I use the default upload protocol because, ... well it is the default.