Sloeber/arduino-eclipse-plugin #1071: Maximum sketch upload speed

cmburns, 2019-02-10:
Hi,

I use sloeber now and created my first sketch (blink) for my ESP8266 (ESP-07, 1MByte Flash).
It worked fine with upload speed 115200.

But if I change upload speed (under project properties-->Arduino Board Selection) to 921600 then I get following error in console view:

> warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed

If I switch back again to 115200 then I can upload the sketch without problems.
My question now is what is the max. upload speed for my ESP8266 (ESP-07)?
How can I make speed 921600 working?

My sloeber version:

>Sloeber, the Eclipse Arduino IDE (plugin version)	4.3.1.201810100607	io.sloeber.feature.feature.group	Sloeber.io

For flashing I use an FTDI adaptor with FTDI chip

>FTDI
1230-C
GO171471
F232RL

Thanks for any help

jantje, 2019-02-23:
Why do you think this is a sloeber issue?

GHPS, 2019-08-21:
>My question now is what is the max. upload speed for my ESP8266 (ESP-07)?

This question is somehow related to this discussion[1] - and totally unrelated to Sloeber.

Please close the issue.

1: https://github.com/esp8266/Arduino/issues/97

closed 2019-08-21 by jantje