Sloeber/arduino-eclipse-plugin #1072: Newbee: Sloeber and OTA - Missing OTA port

cmburns, 2019-02-19:
Hi,

I am totally new to sloeber in combination with OTA. I run sloeber on my Windows 10 professional machine. I have the BasicOTA running on my ESP8266. It shows the IP of my ESP8266.
Then I changed the BasicOTA inside loop() to write something to serial. This changed sketch I wanted to use to test OTA.

What I am missing is the OTA port in the project properties. I only see the COM3.
Python 2.7.13 is installed.

What I further have to do to make the OTA-Pprt available under the ports?
I think there must be a *.py file which establishes the OTA port but I cannot find any  and neither know where to place it. I also beleive I have to setup something inside Eclipse / Sloeber.

Is there any documentation of how to setup OTA for Sloeber?

Thanks a lot!

jantje, 2019-02-20:
The com port is a edit drop box. meaning you can type anything you want there.
As bonjour is not working properly on windows 10 anymore you need to type the ip adress.
For sloeber to know the difference between a IP adress and a local com port you need to add a space and text to the com port.
So for instance 180.180.1.235 my esp board

cmburns, 2019-02-20:
Hi jantje,

thanks for having you here - I really appreciate your help!
As you sggested I entered the IP-Address and space and text inside the Port dropdownbox, like this:

> 192.168.178.44 my esp board

The project name changed to:

>ESP8266_OTA Generic ESP8266 Module :192.168.178.44

Then I compiled the sketch and started the upload, but I got following error:

> Starting upload
Uploading project "ESP8266_OTA" with "ESPTOOL"
no reset because we are using network upload
>Launching: python.exe C:\Program files\sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.5.0/tools/espota.py -i 192.168.178.44 -p 8266 --auth=no_pwd_found_in_code -f C:\sloeber_workspace\ESP8266_OTA/Release/ESP8266_OTA.bin 
Output:
20:57:24 [ERROR]: No response from device
The execution of command "python.exe" is done.

The ESP8266 is up and running, I could ping it and after a manual reset it shows (from the basicOTA sketch):

>Booting
Ready
IP address: 192.168.178.44

I still need some guidance please...

cmburns, 2019-02-21:
@liebman:

Sorry I don't know what to do now. And BTW I don't have any 'build fail'.
Anyway - you changed the file _pom.xml_, I tried to find this file under the sloeber directory and I could find these two:

1. C:\ProgrammeTM\sloeber\plugins\org.eclipse.cdt.core.win32.x86_5.4.1.201806170908\META-INF\maven\org.eclipse.cdt\org.eclipse.cdt.core.win32.x86
2. C:\ProgrammeTM\sloeber\plugins\org.apache.ant_1.10.3.v20180417-1627\META-INF\maven\org.eclipse.orbit.bundles\org.apache.ant

But there is no xml tag like this in these files:

`<tycho.version>1.1.0</tycho.version>`

So I need some light on what to do now.

Thanks...

jantje, 2020-04-21:
I guess this issue can be closed. If not please reopen with current status

closed 2020-04-21 by jantje