Sloeber/arduino-eclipse-plugin #1073: can't create sketchs with curent esp8266 git version

liebman, 2019-02-20:
Using current git esp8266/Arduino Sloeber/Eclipse fails to create project! (At least on macOS) The cpu runs at 100% for a long time then ends with this error:

> Failed to create project
> java.lang.reflect.InvocationTargetException

**To Reproduce**
Steps to reproduce the behavior:
1. install esp8266 core following: https://github.com/esp8266/Arduino/blob/master/doc/installing.rst
2. Start Sloeber/Eclipse
3. Create new  Arduino project
4. Select esp8266 (any variant)
5. Eclipse now using 100% cpu and has to be force quit or wait a long time for error

> Failed to create project
> java.lang.reflect.InvocationTargetException

**Expected behavior**
Project created quickly.

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: macOS 10.14.3 (18D109)
 - Browser N/A
 - Version N/A

jantje, 2019-02-20:
Hi Liebman,
First of all thanks for being a patron.
I'm not sure about what causes this but it looks like something is wrong in the installation.
Let me explain why:
On my test install I have about 500+ boards installed. When I try to create a new sketch it takes a long time.
When in the sketch creation wizard switching between boards.txt takes a long time as well. I think it shouldn't but then again who has 500+ boards installed.
The reason for the delay is that the last page of the wizard is getting repopulated. The last page contains all the possible examples and that means a disk search is done over all installed platforms and libraries, both sloeber managed and private.
On my test system that takes a long time.
Based on that my first guess would be that something in that disk search is not working as expected. There have been bugs before in this area making the search fail but from the top of my head I can't recall a hang.
Something I'm pretty sure of that would cause a hang is creating a link to a parent folder.

Note that if this disk search is the root cause it will happen as soon as the issue on disk exists. I mean it should happen with all boards.txt files and it should happen all the time untill the issue is fixed on disk.
Unless the issue exists in the hardware specific libraries which are part of the currently selected hardware. This is also the reason why switching boards.txt in the sketch wizard causes anew search.

liebman, 2019-02-20:
Interesting....  stack overflow exception......
```
!SESSION 2019-02-19 16:20:20.643 -----------------------------------------------
eclipse.buildId=4.11.0.I20190131-0130
java.version=1.8.0_152
java.vendor=Oracle Corporation
BootLoader constants: OS=macosx, ARCH=x86_64, WS=cocoa, NL=en_US
Framework arguments:  -product org.eclipse.epp.package.cpp.product -keyring /Users/chris.l/.eclipse_keyring
Command-line arguments:  -os macosx -ws cocoa -arch x86_64 -product org.eclipse.epp.package.cpp.product -keyring /Users/chris.l/.eclipse_keyring

!ENTRY org.eclipse.egit.core 1 0 2019-02-19 16:21:11.674
!MESSAGE Using Apache MINA sshd as ssh client.

!ENTRY org.eclipse.launchbar.core 2 0 2019-02-19 16:21:17.007
!MESSAGE Enablement expression is missing for config provider for org.eclipse.cdt.arduino.core.descriptorType

!ENTRY org.eclipse.launchbar.core 2 0 2019-02-19 16:21:17.008
!MESSAGE Enablement expression is missing for config provider for org.eclipse.cdt.debug.core.coreBuildDescriptorType

!ENTRY org.eclipse.launchbar.core 2 0 2019-02-19 16:21:17.008
!MESSAGE Enablement expression is missing for config provider for org.eclipse.cdt.debug.core.coreBuildDescriptorType

!ENTRY io.sloeber.arduino.core 2 0 2019-02-19 16:23:48.743
!MESSAGE Upload will fail due to missing upload port

!ENTRY org.eclipse.equinox.p2.metadata.repository 4 1000 2019-02-19 16:40:37.872
!MESSAGE No repository found at http://download.eclipse.org/tools/cdt/releases/9.7.

!ENTRY io.sloeber.core.ui 4 0 2019-02-19 16:42:11.593
!MESSAGE Failed to create project
!STACK 0
java.lang.reflect.InvocationTargetException
	at org.eclipse.jface.operation.ModalContext.runInCurrentThread(ModalContext.java:451)
	at org.eclipse.jface.operation.ModalContext.run(ModalContext.java:356)
	at org.eclipse.jface.wizard.WizardDialog.run(WizardDialog.java:977)
	at io.sloeber.ui.wizard.newsketch.NewSketchWizard.performFinish(NewSketchWizard.java:99)
	at org.eclipse.jface.wizard.WizardDialog.finishPressed(WizardDialog.java:775)
	at org.eclipse.jface.wizard.WizardDialog.buttonPressed(WizardDialog.java:414)
	at org.eclipse.jface.dialogs.Dialog.lambda$0(Dialog.java:622)
	at org.eclipse.swt.events.SelectionListener$1.widgetSelected(SelectionListener.java:84)
	at org.eclipse.swt.widgets.TypedListener.handleEvent(TypedListener.java:252)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4353)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1512)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1535)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1520)
	at org.eclipse.swt.widgets.Widget.notifyListeners(Widget.java:1324)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4150)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3767)
	at org.eclipse.jface.window.Window.runEventLoop(Window.java:822)
	at org.eclipse.jface.window.Window.open(Window.java:798)
	at org.eclipse.ui.actions.NewProjectAction.run(NewProjectAction.java:118)
	at org.eclipse.jface.action.Action.runWithEvent(Action.java:476)
	at org.eclipse.jface.action.ActionContributionItem.handleWidgetSelection(ActionContributionItem.java:568)
	at org.eclipse.jface.action.ActionContributionItem.lambda$4(ActionContributionItem.java:400)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4353)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1512)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1535)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1520)
	at org.eclipse.swt.widgets.Widget.notifyListeners(Widget.java:1324)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4150)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3767)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1173)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:339)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1062)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:155)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:636)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:339)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:563)
[... 2108 lines cut ...]
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:155)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:636)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:339)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:563)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:151)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:155)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:137)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:107)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:400)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
[... cut ...]
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:659)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:595)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1501)
!SUBENTRY 1 org.eclipse.core.expressions 4 3 2019-02-19 16:42:16.045
!MESSAGE The default variable is not countable. Failed expression: 
CountExpression [ size=0, mode: 5 ANY_NUMBER]
!SUBENTRY 1 org.eclipse.core.expressions 4 3 2019-02-19 16:42:16.045
!MESSAGE The default variable is not countable. Failed expression: 
CountExpression [ size=0, mode: 5 ANY_NUMBER]
```

jantje, 2019-02-20:
The log is weird and as expected at the same time.

`!ENTRY io.sloeber.core.ui 4 0 2019-02-19 16:42:11.593`
Seems like you build your own version so you should be able to debug :-)

`at io.sloeber.ui.wizard.newsketch.NewSketchWizard.performFinish(NewSketchWizard.java:99)`
The code is here
https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.ui/src/io/sloeber/ui/wizard/newsketch/NewSketchWizard.java#L99
This code starts another thread which is not visible in the log. Probably because of logging constraints.

`java.util.regex.Pattern`
This one I have seen before going in a endless loop ending with a stack overflow. Unfortunately the log doesn't tell me where.
Searching leads to following results
[image]

Given the following log entry I think we can exclude serial as the root cause:

```
!ENTRY io.sloeber.arduino.core 2 0 2019-02-19 16:23:48.743
!MESSAGE Upload will fail due to missing upload port
```

I finally end up in MakeEnvironmentString
https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.core/src/io/sloeber/core/tools/Helpers.java#L982
Which implies something new in esp8266 platform.txt or boards.txt file.

jantje, 2019-02-20:
I added a try catch around the code I think that fails.
That will or show the string causing the problem or tell us the problem is elsewhere.
Can you try it?
Own build ->right now
Nightly build ->hopefully tomorrow

liebman, 2019-02-20:
I installed the "plugin" version via marketplace - I'll try to update to nightly tomorrow.

liebman, 2019-02-21:
Nightly update site is still from the 11th...  in any case this is the commit that breaks things:
https://github.com/esp8266/Arduino/commit/9790e1cb7cfa01ac74056fb835e6282af123aba0

jantje, 2019-02-21:
Seems the build sever is still working on the build of 17 feb :-s.
Didn't report any errors
[image]

jantje, 2019-02-21:
The build server is running now. Hopefully it won't take 4 days now :-s
I checked the change but I don't see anything that looks like the root cause. There are changes in the platform.txt so that is kind reassuring that the try catch will lead to the root cause.

jantje, 2019-02-21:
Nightly is now available. Sorry for the delay

liebman, 2019-02-21:
I don't that changed it
```
!ENTRY io.sloeber.core.ui 4 0 2019-02-21 12:07:08.055
!MESSAGE Failed to create project
!STACK 0
java.lang.reflect.InvocationTargetException
	at org.eclipse.jface.operation.ModalContext.runInCurrentThread(ModalContext.java:451)
	at org.eclipse.jface.operation.ModalContext.run(ModalContext.java:356)
	at org.eclipse.jface.wizard.WizardDialog.run(WizardDialog.java:977)
	at io.sloeber.ui.wizard.newsketch.NewSketchWizard.performFinish(NewSketchWizard.java:99)
	at org.eclipse.jface.wizard.WizardDialog.finishPressed(WizardDialog.java:775)
	at org.eclipse.jface.wizard.WizardDialog.buttonPressed(WizardDialog.java:414)
	at org.eclipse.jface.dialogs.Dialog.lambda$0(Dialog.java:622)
	at org.eclipse.swt.events.SelectionListener$1.widgetSelected(SelectionListener.java:84)
	at org.eclipse.swt.widgets.TypedListener.handleEvent(TypedListener.java:252)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4353)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1512)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1535)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1520)
	at org.eclipse.swt.widgets.Widget.notifyListeners(Widget.java:1324)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4150)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3767)
	at org.eclipse.jface.window.Window.runEventLoop(Window.java:822)
	at org.eclipse.jface.window.Window.open(Window.java:798)
	at org.eclipse.ui.internal.navigator.wizards.WizardShortcutAction.run(WizardShortcutAction.java:102)
	at org.eclipse.jface.action.Action.runWithEvent(Action.java:476)
	at org.eclipse.jface.action.ActionContributionItem.handleWidgetSelection(ActionContributionItem.java:568)
	at org.eclipse.jface.action.ActionContributionItem.lambda$4(ActionContributionItem.java:400)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4353)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1512)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1535)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1520)
	at org.eclipse.swt.widgets.Widget.notifyListeners(Widget.java:1324)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4150)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3767)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1173)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:339)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1062)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:155)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:636)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:339)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:563)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:151)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:155)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:137)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:107)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:400)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:659)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:595)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1501)
Caused by: java.lang.StackOverflowError
	at java.util.regex.Pattern$CharProperty.match(Pattern.java:3777)
	at java.util.regex.Pattern$Slice.match(Pattern.java:3972)
	at java.util.regex.Pattern$CharProperty.match(Pattern.java:3777)
	at java.util.regex.Pattern$Slice.match(Pattern.java:3972)
	at java.util.regex.Pattern$CharProperty.match(Pattern.java:3777)
	at java.util.regex.Pattern$Slice.match(Pattern.java:3972)
	at java.util.regex.Pattern$CharProperty.match(Pattern.java:3777)
	at java.util.regex.Pattern$Slice.match(Pattern.java:3972)
	at java.util.regex.Pattern$CharProperty.match(Pattern.java:3777)
	at java.util.regex.Pattern$Slice.match(Pattern.java:3972)
	at java.util.regex.Pattern$CharProperty.match(Pattern.java:3777)
	at java.util.regex.Pattern$Slice.match(Pattern.java:3972)
	at java.util.regex.Pattern$CharProperty.match(Pattern.java:3777)
```

jantje, 2019-02-21:
I didn't expect that to happen.
I'll try here with the git repository.
See whether I can debug the thing

jantje, 2019-02-21:
I think I got it.
Seems like TOOLS.ESPTOOL.CMD contains the variable UPLOAD.VERBOSE 2 times and Sloeber seems to fail on that.
Looking for a way to tell Sloeber not to care about this ;-)

jantje, 2019-02-21:
Project creation and compilation works here but the upload fails with a bizar error
```
Starting upload
Uploading project "dd" with "ESPTOOL"
Starting reset using DTR toggle process
Exception while opening Serial port "".
Continuing to use ""
Ending reset

Launching: C:/Users/jan/git/Arduino/tools/python/python.exe C:/Users/jan/git/Arduino/tools/upload.py C:/Users/jan/git/Arduino/tools/pyserial C:/Users/jan/git/Arduino/tools/esptool --chip esp8266 --port  --baud 57600  version --end --chip esp8266 --port  --baud 57600  write_flash 0x0 L:\test\runtime-EclipseApplication_eng\dd/Release/dd.bin --end 
Output:
Error in command line, need pyserial path as 1st arg and esptool path as 2nd.
The execution of command "C:/Users/jan/git/Arduino/tools/python/python.exe" is done.
```
I'm checking in the change and run the nightly build

liebman, 2019-02-21:
Yea - I get the error on a mac as well.  However if I run the exact command in a shell manually it works. Must be something that sloeber is passing in the env or maybe passing an empty param?

jantje, 2019-02-22:
It doesn't run on the command prompt at my system I guess because I didn't initialize the submodules as in the doc you linked.
Anyway the new nightly is available and should get you there with the latest esp8266 git version.

jantje, 2019-02-23:
Did the latest version work for the sketch creation?
Is the upload issue also there on your system?

liebman, 2019-02-23:
Sketch creation works nicely.  Upload fails.
```
Starting upload
Uploading project "ESPButton" with "ESPTOOL"
Starting reset using DTR toggle process
Continuing to use "/dev/cu.usbserial-6"
Ending reset

Launching: python /Users/chris.l/Dropbox/git/ESPButton/hardware/esp8266com/esp8266/tools/upload.py --chip esp8266 --port /dev/cu.usbserial-6 --baud 115200  version --end --chip esp8266 --port /dev/cu.usbserial-6 --baud 115200  write_flash 0x0 /Users/chris.l/Dropbox/git/ESPButton/ESPButton/Release/ESPButton.bin --end 
Output:
usage: esptool [-h] [--chip {auto,esp8266,esp32}] [--port PORT] [--baud BAUD]
               [--before {default_reset,no_reset,no_reset_no_sync}]
               [--after {hard_reset,soft_reset,no_reset}] [--no-stub]
               [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,version}
               ...
esptool: error: argument operation: invalid choice: '' (choose from 'load_ram', 'dump_mem', 'read_mem', 'write_mem', 'write_flash', 'run', 'image_info', 'make_image', 'elf2image', 'read_mac', 'chip_id', 'flash_id', 'read_flash_status', 'write_flash_status', 'read_flash', 'verify_flash', 'erase_flash', 'erase_region', 'version')
The execution of command "python" is done.
```

Running the command manually works fine:
```
zod: 05:13:37 ESPButton chris.l$ python /Users/chris.l/Dropbox/git/ESPButton/hardware/esp8266com/esp8266/tools/upload.py --chip esp8266 --port /dev/cu.usbserial-6 --baud 115200  version --end --chip esp8266 --port /dev/cu.usbserial-6 --baud 115200  write_flash 0x0 /Users/chris.l/Dropbox/git/ESPButton/ESPButton/Release/ESPButton.bin --end 
esptool.py v2.6
2.6
esptool.py v2.6
Serial port /dev/cu.usbserial-6
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: 2c:3a:e8:4b:d0:53
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 632096 bytes to 402733...
Wrote 632096 bytes (402733 compressed) at 0x00000000 in 35.3 seconds (effective 143.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

It's like an empty command line token/argument is being introduced somewhere before 'version'.

liebman, 2019-02-23:
Ha!  I think I found it.  Upload verbose is empty and I'm guessing your then passing an empty argument that breaks "usage".

tools.esptool.upload.pattern="{cmd}" "{runtime.platform.path}/tools/upload.py" --chip esp8266 --port "{serial.port}" --baud "{upload.speed}" **_"{upload.verbose}"_** {upload.erase_cmd} --end --chip esp8266 --port "{serial.port}" --baud "{upload.speed}" "{upload.verbose}" write_flash 0x0 "{build.path}/{build.project_name}.bin" --end

jantje, 2019-02-23:
Seems that that is indeed the problem.
I think I need to tackle the difference in verbosity in Sloeber versus arduino IDE. Sloeber had this before Arduino IDE and the Arduino IDE way now seems to cause problems.
Anyways: workaround: Create following build variable in the project properties
[image]
Select apply/close
Open project properties->arduino->apply close
Upload should work now.

liebman, 2019-02-23:
Does not like -v.  --trace would work but thats super super verbose...
```
Launching: python /Users/chris.l/Dropbox/git/ESPButton/hardware/esp8266com/esp8266/tools/upload.py --chip esp8266 --port /dev/cu.usbserial-6 --baud 115200 -v version --end --chip esp8266 --port /dev/cu.usbserial-6 --baud 115200 -v write_flash 0x0 /Users/chris.l/Dropbox/git/ESPButton/ESPButton/Release/ESPButton.bin --end 
Output:
usage: esptool [-h] [--chip {auto,esp8266,esp32}] [--port PORT] [--baud BAUD]
               [--before {default_reset,no_reset,no_reset_no_sync}]
               [--after {hard_reset,soft_reset,no_reset}] [--no-stub]
               [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,version}
               ...
esptool: error: unrecognized arguments: -v
```

liebman, 2019-02-23:
Ok '--chip=esp8266' works as it just duplicates an option.

jantje, 2019-02-23:
It is a problem of having it blank resulting in 2 spaces resulting in some problem when starting the upload command. Anything that doesn't break the command should do.

msandres13, 2019-08-11:
Ran into the same problem after #1090 and using nightly. Circumvented by patching upload.py removing all empty commands before call esptool. (see elif removing empty strings)

upload.py from sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.5.2/tools
```python
import sys
import os

sys.argv.pop(0) # Remove executable name
toolspath = os.path.dirname(os.path.realpath(__file__)).replace('\\', '/') # CWD in UNIX format
try:
    sys.path.insert(0, toolspath + "/pyserial") # Add pyserial dir to search path
    sys.path.insert(0, toolspath + "/esptool") # Add esptool dir to search path
    import esptool # If this fails, we can't continue and will bomb below
except:
    sys.stderr.write("Error in command line, need pyserial path as 1st arg and esptool path as 2nd.\n")
    sys.exit(1)

fakeargs = [];
print("Hello")
while len(sys.argv):
    if sys.argv[0] == '--end':
	print("esptool with args" , fakeargs)
        esptool.main(fakeargs)
        sys.argv.pop(0) # Remove --end
        fakeargs = []
    elif sys.argv[0] == '':
        sys.argv.pop(0) # Remove empty param
	print("remove empty args")
    else:
        # We silently replace the 921kbaud setting with 460k to enable backward
        # compatibility with the old esptool-ck.exe.  Esptool.py doesn't seem
        # work reliably at 921k, but is still significantly faster at 460kbaud.
        thisarg = sys.argv.pop(0)
        if thisarg == "921600":
            thisarg = "460800"
        fakeargs = fakeargs + [thisarg]
```

jantje, 2019-10-07:
The update should be at the update site now but due to all kind of eclipse versioning issues my installation got corrupted when updating.
So I'm making a new release based on eclipse 2019-09

closed 2019-10-07 by jantje