Sloeber/arduino-eclipse-plugin #1076: Compilation error #include MBEDTLS_CONFIG_FILE end etc

ilya3737, 2019-02-24:
Hello, jantje! I have a problem. 
I suddenly had trouble compiling code. Up to this point, everything worked predictably .. I spent a lot of time searching for a problem, but I could not solve it.
I tried to put brackets, but nothing helped. Replacing other libraries with WebServer did not help either.

[image]
[image]

jantje, 2019-02-24:
Maybe you can help me with a problem of mine instead of me helping you.
I'm considering suicide in the comming days and I would prefer to jump from a plane or air balloon but as I don't own one of these and I guess most owner of these are aware people wanting to do so I fear that is not an option.
CO seems a good alternative but I don't have a reliable co source so maybe you know a alternative?

ilya3737, 2019-02-25:
I condemn suicides and do not call for solving problems in this way, but I can answer your question. Maybe propane or carbon monoxide will help you?

jantje, 2019-02-25:
co is carbon monoxide. That still leaves me with a reliable source.
I guess you are having a problem with a flavour of issue with #1028

ilya3737, 2019-02-25:
jantje, tthank you so much! That helped!
It turns out that I had previously encountered this problem and the last time I found the page with the answer myself. but I decided to update the library and delete these changes :(

closed 2019-02-25 by ilya3737

ilya3737, 2019-02-25:
just don’t think about suicide!) I really like your program and I made you donate)

jantje, 2019-02-25:
It looks to me you are better in condemning than searching ;-)
https://github.com/Sloeber/arduino-eclipse-plugin/issues?utf8=%E2%9C%93&q=is%3Aissue+MBEDTLS_CONFIG_FILE

ilya3737, 2019-02-26:
I did not condemn you. it is not true. I just did not have enough experience to find the answer myself. I thought that the error is related to another problem. So I asked for help. :)

It is a pity that there were not enough good people around the world who could help your project.
this is very sad news I saw yesterday at patreon.
Be sure to leave the opportunity to thank you for this titanic work.
Аnd leave the opportunity to buy your shirts :)
like many others, I hope that you will come back after some time.

good luck!