Sloeber/arduino-eclipse-plugin #1077: Local debugging not working

dfaulks, 2019-02-27:
**Describe the bug**
After watching the local debugging video the author seems to be debugging a exe file however the compiler does not produce an exe file. From what I can tell the "MAKE" file is not configured to create an executable "exe" file.

**To Reproduce**
Steps to reproduce the behavior:
follow the local debug configuration video

**Expected behavior**
I expect to get and exe file to debug

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 64 bit
 - Browser Safari
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

closed 2019-10-07 by jantje

NuwanR, 2020-05-23:
So what is the resolution of this @jantje ? because I have the same problem the exe doesn't get created

jantje, 2020-05-23:
This is part of my "extensive regression test" and it works fine there.
Can you provide a link to the video you are referring to (might be outdated)
And the build log

NuwanR, 2020-05-23:
https://www.youtube.com/watch?v=ZN-rLSyhBN8 - Debug your arduino code on your local machine

jantje, 2020-05-23:
did you install via the jantje boards?
did you select the jantje board with the debug option?

NuwanR, 2020-05-23:
I used the jantje board with the debug option, which comes under "hardware-master\jantje\avr_local"

jantje, 2020-05-23:
and the build log?

NuwanR, 2020-05-24:
oh..!! now I can't build the project using localdebug configurations its giving an error.

`02:08:14 **** Build of configuration localdebug for project Blink ****
"C:\\Sloeber\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"/bin/g++" -c -g -O0 -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR_LOCAL   -DLOCAL_RUN   -I"C:\arduino_workspace\hardware-master\jantje\avr_local\cores\arduino" -I"C:\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.8.2\variants\standard" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"   -o "sloeber.ino.cpp.o"   -Dsrandom=srand -Drandom=rand
The system cannot find the path specified.
subdir.mk:18: recipe for target 'sloeber.ino.cpp.o' failed
make: *** [sloeber.ino.cpp.o] Error 1
"C:/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

02:08:14 Build Failed. 2 errors, 0 warnings. (took 618ms)

`

NuwanR, 2020-05-24:
Is there any new and complete guide how we can setup the local debugging?

jantje, 2020-05-24:
This is not the log of a boardsmanager installed local debug board

NuwanR, 2020-05-24:
This is the only path it showing as logging in c++ configurations, is there any other place which it logs?

jantje, 2020-05-24:
What I mean is that the log contains
` -I"C:\arduino_workspace\hardware-master\jantje\avr_local\cores\arduino"`
And from that I know the selected board is not a board installed by the boards manager but a board from a private hardware. This also explains why the path is missing in 
`"/bin/g++"`
Which results in 
`The system cannot find the path specified.`
which takes me back to
https://github.com/Sloeber/arduino-eclipse-plugin/issues/1077#issuecomment-633148655

NanduPathak, 2020-05-31:
Hello Jantje
Its good that you have introduce this tool to debug Arduino and ESP boards and also Jantje hardware. 

But I followed all the steps from the video and also looked at the above comments, but unfortunately i did not get how to solve the missing  path "/bin/g++" you mentioned.

My log in debug configuration. Please let me now how can i solve my problem.

'Building file: ..\.ino.cpp'
'Starting C++ compile'
"/bin/g++" -c -g -O0 -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR_LOCAL  -DLOCAL_RUN  -I"C:\debug\git\hardware\jantje\avr_local\cores\arduino" -I"C:\debug\git\hardware\jantje\avr_local\variants\avr:*:standard" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\.ino.cpp"  -o  ".ino.cpp.o"  -Dsrandom=srand -Drandom=rand
The system cannot find the path specified.
subdir.mk:24: recipe for target '.ino.cpp.o' failed

thanks in advance.

jantje, 2020-05-31:
Please provide following info from you system
proof you installed the hardware via the boardsmanager
[image]
proof that is the thing you are using
[image]

Another tip: delete the folder C:\debug\git\hardware\
As you can see there is no git in the command on my system and there should be none on your system
```
16:36:33 **** Build of configuration Release for project issue1077 ****
"C:\\sloeber\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\sloeber.ino.cpp'
'Start de C++ bouw'
"C:\sloeber\arduinoPlugin\packages\Jantje\tools\MinGW\5.3.0/bin/g++" -c -g -O0 -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -DF_CPU=8000000L -DARDUINO=10812 -DARDUINO_AVR_FIO -DARDUINO_ARCH_PC  -DLOCAL_RUN   -I"C:\sloeber\arduinoPlugin\packages\Jantje\hardware\pc\1.0.2\cores\arduino" -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.23\variants\eightanaloginputs" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"   -o "sloeber.ino.cpp.o"   -Dsrandom=srand -Drandom=rand
'Finished building: ..\sloeber.ino.cpp'
' '
```

uzi18, 2020-05-31:
got same problem, when have 2 build configurations
first standard arduino core (with mega 2560 selected) 
second megacore
when 2nd is selected to build, after sloeber restart got exactly same error, need to select first configuration properties and apply, than second and apply, from this moment everything is working

jantje, 2020-05-31:
Switching between configurations is only supported by clicking on the project->configurations->set active

NanduPathak, 2020-06-01:
you are right Sloeber is missing hardware installed.
Can you guide me or share video link showing how to install it
[image]

jantje, 2020-06-01:
@NanduPathak 
I think it is time for you to start with a fresh install with the latest version of sloeber

NanduPathak, 2020-06-01:
Now I can successfully build the project in debug mode.
When I start Debug it asks me to select Process. 
[image]

jantje, 2020-06-01:
This part   is still the same as in the video

NanduPathak, 2020-06-02:
I could able to start code in debug but i am not getting debug point
[image]