Sloeber/arduino-eclipse-plugin #1079: Can not install plug-in into Cevelop (Eclipse-CDT-based IDE)

PeterSommerlad, 2019-03-21:
**Describe the bug**
Trying to install the plugin into Cevelop using the update site, I get missing dependencies:

Cannot complete the install because one or more required items could not be found.
  Software being installed: Sloeber, the Eclipse Arduino IDE 4.3.1.201810101819 (io.sloeber.product 4.3.1.201810101819)
  Missing requirement: GNU MCU C/C++ OpenOCD Debugging 4.3.3.201809300659 (ilg.gnumcueclipse.debug.gdbjtag.openocd 4.3.3.201809300659) requires 'osgi.bundle; ilg.gnumcueclipse.debug.gdbjtag 0.0.0' but it could not be found
  Cannot satisfy dependency:
    From: GNU MCU C/C++ OpenOCD Debugging 4.3.3.201809300659 (ilg.gnumcueclipse.debug.gdbjtag.openocd.feature.feature.group 4.3.3.201809300659)
    To: org.eclipse.equinox.p2.iu; ilg.gnumcueclipse.debug.gdbjtag.openocd [4.3.3.201809300659,4.3.3.201809300659]
  Cannot satisfy dependency:
    From: Sloeber, the Eclipse Arduino IDE 4.3.1.201810101819 (io.sloeber.product 4.3.1.201810101819)
    To: org.eclipse.equinox.p2.iu; ilg.gnumcueclipse.debug.gdbjtag.openocd.feature.feature.group [4.3.3.201809300659,4.3.3.201809300659]

**To Reproduce**
1. Install new software
2. past plugin upatesite: Arduino-Sloeber - http://eclipse.baeyens.it/update/V4/stable
3. select plugin components
4. try to install

**Expected behavior**

Installs properly into Cevelop.

**Screenshots**

**Desktop (please complete the following information):**
 - OS: Macos
 - Browser [e.g. chrome, safari]
 - Version Mojave

**Additional context**
For Cevelop see www.cevelop.com 
It comes with unit testing, improved refactoring and C++ static analysis support

PeterSommerlad, 2019-03-21:
with the GNU MCU Eclipse plugins installed, still a dependency is missing. Mainly because also the GNU MCU Eclipse plugins do not install completely.

org.eclipse.cdt.debug.gdbjtag.core

Need to check if that was abandoned for current CDT or not put into Cevelop

jantje, 2019-12-11:
This is not a "supported" option. The only 'supported' option is the product.
This is because it is a real pain to find all the needed plugins and they all need compatible versions.

closed 2021-07-11 by jantje