Sloeber/arduino-eclipse-plugin #1082: code uploads but does not run to custom stm32 board

Cyberteen, 2019-03-29:
Hello,

I am using eclipse + sloeber configuration to debug arduino programs on my custom stm32 board based on STM32F413RG.

I am using ST-link over SWD for uploading and openOCD for debugging.

So, I am testing a simple blink program using this setup. But when I run it, the led just stays ON and doesn't seem to actually blink. But when I break the code and debug and line-by-line, I am able to see the LED switch ON and OFF every 2 secs as per my code. Why doesn't it blink normally when I just run without breaks?

Also, when uploading using Arduino IDE, the code works fine. So, what am I missing here.? Any timing/clock sync issues? any help appreciated. Thanks in advance. :)

uzi18, 2019-03-29:
what is your stm32 core version in project?

Cyberteen, 2019-03-29:
@uzi18 It says, 1.5.0. That seems to be the latest in my list.

closed 2020-01-18 by jantje