Sloeber/arduino-eclipse-plugin #1083: stm32duino (official) USB CDC won't build with Sloeber

cdrose, 2019-03-31:
Eclipse with Sloeber plugin
STM32F103C8T6 BluePill
Arduino Core STM32 1.5.0 latest
Windows 10

When USB CDC components are selected for stm32duino, Sloeber fails to build the project. Sketches without the USB CDC components build fine, and the Arduino IDE builds USB CDC sketches fine. While I am unable to test this personally, it sounds like the issue may only exist on Windows and Sloeber on Linux works as expected.

I initially raised this on the STM32duino page but it looks like the issue lies with Sloeber, possibly something to do with how platforms.txt is parsed. The compiler flags for the USB configuration are missing the escape character for the quotation marks:

Arduino IDE 1.8.9:
> -DSTM32F103x6 -DUSBCON -DUSBD_VID=0x0483 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"BLUEPILL_F103C6\"" -DHAL_PCD_MODULE_ENABLED -DUSBD_USE_CDC

Eclipse 2018-12 + Sloeber plugin V4:
> -DSTM32F103xB -DUSBCON  -DUSBD_VID=0x0483 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT="BLUEPILL_F103C8"" -DHAL_PCD_MODULE_ENABLED -DUSBD_USE_CDC

platforms.txt:
> build.usb_flags=-DUSBCON {build.usb_speed} -DUSBD_VID={build.vid} '-DUSB_MANUFACTURER={build.usb_manufacturer}' '-DUSB_PRODUCT="{build.board}"' -DHAL_PCD_MODULE_ENABLED 

boards.txt:
> GenF1.menu.pnum.BLUEPILL_F103C8.build.board=BLUEPILL_F103C8

It looks like the goal is to use the build.board parameter which has no quotation marks around it and then add some quotation marks so it works correctly in the string expansion macros in the core. The Arduino IDE correctly handles this and escapes the extra quotation marks while the Sloeber plugin does not.

See: https://github.com/stm32duino/Arduino_Core_STM32/issues/486#issuecomment-477505308

mrv96, 2019-04-27:
Hi, i have the same problem, have you found a temporary solution?

cdrose, 2019-04-28:
Sloeber seems to pull the info from platforms.txt and place it into the build environment variables. I had no luch editing platforms.txt so I changed the build variables instead. Simply add the escape characters for the quotation marks on the USB_PRODUCT variable.

[image]

mrv96, 2019-04-29:
Thank you very much. Now it works. Previously I tried to modify platform.txt without success. I hope that someone will be able to fix this issue also on windows.

dferbas, 2020-01-20:
@cdrose: This workaround is ok only for a certain board.

I succeeded in editing platform.txt. Purpose is to pass board name as a quoted string to allow its concatenation in utils.h. Just modify following line (only 2 defines are listed, all others are left untouched).
```
# USB Flags
# ---------
#build.usb_flags=... '-DUSB_MANUFACTURER={build.usb_manufacturer}'  '-DUSB_PRODUCT="{build.board}"' ...
build.usb_flags=... -DUSB_MANUFACTURER={build.usb_manufacturer}  -DUSB_PRODUCT=\"{build.board}\" ...
```

jantje, 2020-03-10:
This should work now with the nightly out of the box.

closed 2020-04-21 by jantje