Sloeber/arduino-eclipse-plugin #1085: Sloeber.io will expire on 4 june 2019 links will no longer work

jantje, 2019-04-04:
**Describe the bug**
Sloeber.io will expire this year

**To Reproduce**
This is not reproducible

**Screenshots**
[image]

**Additional context**
Sloeber points to sloeber.io for several things in the menu. But also to download examples, libraries, json packages(?). 
Sloeber.io is currently redirecting to baeyens.it/eclipse.
I'm not sure when baeyens.it will expire (been paid for till 15-01-2020)
Baeyens.it/eclipse content is available in this github repository.

Easiest solution is to take over sloeber.io domain and redirect to wherever from there. However that costs money I will not invest.

liebman, 2019-04-04:
This makes me sad.

erdemuncuoglu, 2019-07-06:
It seems that sloeber.io  will live for an other year :D

> Domain Name: SLOEBER.IO
> Registry Domain ID: D503300000040370818-LRMS
> Registrar WHOIS Server: whois.rrpproxy.net
> Registrar URL: http://www.key-systems.net
> Updated Date: 2019-06-04T22:20:25Z
> Creation Date: 2016-06-04T12:23:54Z
> Registry Expiry Date: 2020-06-04T12:23:54Z
> Registrar Registration Expiration Date:
> Registrar: Key-Systems GmbH
> Registrar IANA ID: 269
> Registrar Abuse Contact Email: abuse@key-systems.net
> Registrar Abuse Contact Phone: +49.68949396850
> Reseller:
> Domain Status: ok https://icann.org/epp#ok
> Domain Status: autoRenewPeriod https://icann.org/epp#autoRenewPeriod
> Registrant Organization: jan baeyens
> Registrant State/Province: oostvlaanderen
> Registrant Country: BE
> Name Server: NS1.DREAMHOST.COM
> Name Server: NS2.DREAMHOST.COM
> Name Server: NS3.DREAMHOST.COM
> DNSSEC: unsigned

jantje, 2019-07-10:
Seems like it is still working.
I have not taken any actions so I have no clue what happened.

erdemuncuoglu, 2019-07-10:
> Domain Status: autoRenewPeriod https://icann.org/epp#autoRenewPeriod

Either it may be a hidden blessing from your registrar which they will probably renounce when they won't receive the payment or you subscribed in a way that they will charge the fee from your account/credit card.
You may need to check it.
I never receive such a grace from my registrar. They just suspend it (as a warning shot) before deleting a domain if it is not paid in time. :expressionless:

It was around two years ago when I've first meet SLOEBER (while searching for a professional IDE to develop my hobby projects) since then I could never find a spare time to play with this awesome IDE nor with my ESP32s... (but I'm very close to start again)
work hard, ~~play~~ work hard...
On top of it the mighty developer of SLOEBER already lost his interest because of not so supportive community. [ref#1](https://forum.arduino.cc/index.php?topic=79595.msg4142403#msg4142403), [ref#2](https://forum.arduino.cc/index.php?topic=600197.msg4075118#msg4075118)

Thank you for all your efforts for producing such a beautiful IDE

jantje, 2019-08-22:
As it is still working I updated the end of life date

Theremingenieur, 2019-08-27:
I'd be ready to pay the next sloeber.io domain renewal whenever it will happen and (if needed) to host it on my vps. As a counterpart, I'd like to see the "space in the project path" problem fixed since in the new macOS Catalina, the default user documents folder has been moved to /Users/whoever/Library/CloudStorage/iCloud Drive/Documents. And that space in "iCloud Drive" lets actually all build processes in Sloeber fail...
Deal?

jantje, 2019-08-29:
>And that space in "iCloud Drive" lets actually all build processes in Sloeber fail...

The problem with spaces is multidimensional. One of the main reasons is that make simply is not build to handle spaces, on top of that a different make implementation is used on the different oses, mac, windows and linux.
There is also the added platform dimension because sloeber needs to generate make instructions based on platform.txt files and the makers of these platform.txt files do not care about make.
I found that people knowing make agree that there is no general solution to tackle the space problem. Therefore problems need to be tackled one by one in a three dimensional matrix (dimension 1 : build tools; dimension 2 os; dimension 3 platform)  
As an alternative Sloeber could use a different cdt building engine (which does not rely on make) but that is a whole different can of worms. (I looked into the alternatives and never got my head around it)
What I'm trying to say is that it is complicated. I can't promise I can get things to work with spaces and I can't even test as I don't own a mac.
Being autistic this means I can't accept the deal.
I propose you make a issue for the "iCloud Drive" space issue and we'll go from there.

Theremingenieur, 2019-09-02:
Hi @jantje , thank you for your exhaustive reply. I'll put that issue actually "on hold", seen that macOS Catalina beta 7 does _**again**_ re-arrange the iCloud drive folder locations and paths.

Nonetheless, and independent of that, my offer to host the project and to pay for the domain renewals remains unchanged. I've always supported the project on Patreon until you decided to go away from them (for the well known reasons) and this could be a way for me to continue my support.

Theremingenieur, 2019-09-09:
Looks like with the newest macOS beta, the "space in the project path problem" does not longer exist, even when your Desktop and Documents folders are synced into iCloud. They re-organized the local mirrors of the cloud storage and added symbolic links which makes that the Documents folder is well seen by Sloeber/Eclipse under /Users/yourusername/Documents/ and everything compiles again.

jantje, 2019-09-09:
thanks for the input :-)

MarkAllread, 2019-10-20:
Just a casual user (Toddler), but noticed this on trying to install latest stable MacOS version of Beryllium.

Failed to configer Sloeber
Failed to download "https://eclipse.baeyens.it/download/examples_Arduino_1_6_7.zip".
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target

Wondering if links are in fact broken.

jantje, 2019-10-20:
@MarkAllread 
This is not related. Does the download in the browser work?

MarkAllread, 2019-10-20:
Yes, download works and unzips.  Mac OSX Sierra.
First run where it gets all the packages fails at the point mentioned above and does not complete.
The app starts and I go through the normal project update/edits that you provide, but basically every line of code is flagged as error. 
I have successfully installed V4 previously (a couple of times, maybe 6 months ago) but always ended up going back to 3.
Sorry to hijack the thread with an unrelated issue. Let me know if you want me to move it.

jantje, 2019-10-20:
As it is a v3 issue there is no need to create a issue.

MarkAllread, 2019-10-20:
My issue is with V4, not V3.
Everything I said above is to do with V4. (which I thought was Beryllium)
I have no issue with V3.

"..trying to install latest stable MacOS version of Beryllium...."

MarkAllread, 2019-10-20:
Hi jantje.
I just tried installing the V4 plugin into Eclipse IDE and got exactly the same issue.

jantje, 2019-10-21:
@MarkAllread 
have you seen this?
https://stackoverflow.com/questions/21076179/pkix-path-building-failed-and-unable-to-find-valid-certification-path-to-requ

MarkAllread, 2019-10-21:
Thanks for your help @jantje. I took a look and that's way over my head. I'm just a simple C programmer ;-)...I know nothing about Java.

FYI, I tried installing V4.3 on three different Mac computers and 3 different Mac OSX versions. all with Java 8. I had exactly the same problem on all three.
I don't think I could handle messing around with certificates like that stack overflow post describes.
I think I'll stick to my installed version of Toddler...it works just fine for what I need.
Thanks again.

jantje, 2019-12-23:
Seems like sloeber.io is gone.

erdemuncuoglu, 2019-12-23:
Hosting seems to be expired. What we need to make it live again?

Theremingenieur, 2019-12-23:
I had suggested on last August 27 to provide hosting on one of my servers and to pay for needed domain renewals. As everybody knows, such transfers can not be done without collaboration of the actual domain owner. But nobody has contacted me since then although there was enough time. I'm still listening to requests. @jantje, it's time for some activity from your side!

jantje, 2019-12-23:
@Theremingenieur 
I know. My bad. When the domain remained online, I just assumed I paid for several years and they tried to trick me into paying extra.
Just checked; they/I didn't.

@erdemuncuoglu @Theremingenieur 
I have no clue what can be done nor what I can do.

Theremingenieur, 2019-12-23:
As far as I remember it was either sloeber.io which redirected towards bayens.it or vice versa. Let me know and provide me with a backup of all the content, so that I can set up a similar hosting structure.
I'll then do a domain transfer request for sloeber.io and @jantje will have to allow it. Then, I can setup sloeber.io again, with all the restored content in direct, without the need for eventual redirection to bayens.it.

Theremingenieur, 2019-12-23:
I just see that the domain sloeber.io is property of a company named Key-Systems GmbH. Who is that? I'll need them to allow the domain transfer.
Or we leave sloeber.io as is (with the redirection to baeyens.it) and restore baeyens.it on my server. In that case, @jantje should contact me in private because he is the owner of that domain.

Theremingenieur, 2019-12-23:
Have the website up and running under the (temporary and experimental) subdomain [http://sloeber.theremin.tf](http://sloeber.theremin.tf) until I can put my hands onto the sloeber.io domain.

Theremingenieur, 2019-12-23:
Hell... I do everything (register an auxiliary domain name, buy a SSL certificate, create a new virtual host on my server, do a manual content install from Github) in a hurry to make the website content accessible again. But then, I see that I'm having a huge monologue here and @jantje isn't as reactive as I'd need it... :-(

Theremingenieur, 2019-12-23:
@jantje , pull your head out of the sand and reply! Or did I make all that effort for nothing?

MarkAllread, 2019-12-23:
Hi @Theremingenieur
I'm sure many people appreciate very much what you are trying to do.
It's Christmas...perhaps @jantje is doing his holiday shopping. :-)

jantje, 2019-12-23:
@Theremingenieur 
Been busy all day making chrismas gifts. Only just back online.
I don't think I have the info you request. I'll send you the account info via pm.

jantje, 2019-12-23:
@Theremingenieur 
Message has been send.

Theremingenieur, 2019-12-24:
Thank you! It looks a little bit more complicated than I thought, but we'll get it sorted out. For the moment, we have a temporary workaround on **[https://sloeber.tf](https://sloeber.tf)** and I'll optimize it so that you get the traffic back on eclipse.baeyens.it. But that means that you have to do everything to keep eclipse.baeyens.it alive!

Theremingenieur, 2019-12-24:
And please accept my sincere apologies if I was impatient - I did not realize that people are actually busy with Christmas since I've grown up in a different tradition..

wimjongman, 2019-12-24:
@Theremingenieur are you working on the theremin? I love that instrument and would like to create on. Do you have some public content about it?

Theremingenieur, 2019-12-24:
Hi @wimjongman, yes, I play & teach the Theremin, I'm also one of the rare guys who do Theremin repairs, design of add-on circuits, and custom builds. As a first approach to learn struggling with RF and fractions of a pF, but to get something which is already playable, I recommend building that circuit which I published a few years ago: [http://thereminworld.com/Forums/T/29231/my-new-year-gift-to-tw-a-new-theremin-circuit?Page=0](http://thereminworld.com/Forums/T/29231/my-new-year-gift-to-tw-a-new-theremin-circuit?Page=0)

jantje, 2019-12-24:
@Theremingenieur 
After I closed my patreon account, I seriously reduced my web presence. This process has naturally continued and now it is not uncommon for me not to log on to the web for over a week.

Basically: due to the patreons paying a patreonage I felt obliged to be on line but at the same time I found the total amount I received was way to little to spend so much time. This created a huge amount of stress which was part of the decision to close the patreon page.

Now, when I have energy I do thing around the house. So basically me being off line is good news :-)

So don't worry be happy. No stress: been there, done that, don't want to do it again.

jantje, 2019-12-24:
> Thank you! It looks a little bit more complicated than I thought, but we'll get it sorted out.

I think the only easy solution is to have a sloeber.io web site redirecting to eclipse/baeyens.it
I bought the baeyens.it domain because I didn't want to change email address again (witch each change of provider). So as long as I'm alive I'll probably keep baeyens.it.
Not making any promisees on how long I'll be alive though ;-)
if baeyens.it disappears it should be relative easy to do a global find and replace and mod the redirection.

Theremingenieur, 2019-12-24:
> I think the only easy solution is to have a sloeber.io web site redirecting to eclipse/baeyens.it

Yes, as soon as I can put my hands on that domain to readjust the redirection - investigations continue.
Actually, and as a proof of concept, that's exactly what I did with the **[sloeber.tf](https://sloeber.tf)** domain, so we can observe it already for any side effects.

A happy Christmas and/or Hanukkah to everybody!

erdemuncuoglu, 2019-12-25:
> ... but at the same time I found the total amount I received was way to little to spend so much time.

**The Delusion:** Nearly all issues could be solved by giving or spending some money (which always seems to be big enough).

**The Truth:** You need to give or spend human time and effort mostly.

**The Fun Fact:** You have to get old (spend time and effort) to understand it.

Marry Christmas and Happy New Year.

jantje, 2020-01-08:
@Theremingenieur 
Any update?

Theremingenieur, 2020-01-08:
No update. It’s weird, @jantje. The registrar’s hotline told me that the domain has still a valid registration with them until June,4,2020, but not under your account. They refused to tell me on whose account it is now, so I am blocked for the moment.
Perhaps you can get more info out of them?
If not, I can only wait until the domain expires in June plus the 60 days grace period and then be the first to put my hands on it.

jantje, 2020-01-09:
I can try to get some more info out of the registrar. But I'm far from confident. How do I contact them?

jantje, 2020-06-06:
Seems like the registration has ended.
@Theremingenieur are you still interested in putting your hands on the domain?

Theremingenieur, 2020-06-06:
It has not ended. It was auto-renewed yesterday until June 4 2021... 
Apparently, someone in the US is very interested in not freeing up the domain... :(
[image]
[image]
[image]

jantje, 2020-06-06:
Strange really strange

SLACer-garth, 2021-07-10:
Eclipse marketplace IDE description still has links to sloeber.io, which is now redirecting to a malware site. Can you fix the eclipse marketplace description?

jantje, 2021-07-11:
Thanks for reporting. I fixed the instances I found.
It may take some time to propagate through marketplace (I really do not know)
Can you check whether the instances you found are fixed?
If not report the exact locations?

SLACer-garth, 2021-07-11:
Thanks for fixing it. Yes, the marketplace description now links to the right place.

Jim-2249, 2021-07-12:
There is still a sloeber.io link on the main github page: https://github.com/Sloeber

jantje, 2021-07-12:
@Jim-2249 Thanks
Thinks I fixed it.