Sloeber/arduino-eclipse-plugin #1087: Compilation error in library ESPAsyncWebServer-master

gjanpjetro, 2019-04-12:
In my project I use the ESPAsyncWebServer-master library for an Async Web Server in ESP32 - WROM32 DEVKIT V1. As "Platform folder", I use the sloeber/arduinoPlugin/packages/esp32/1.0.1.
As "Board", I use the DOIT ESP32 DEVKIT V1.
The compiled result, with the arduino IDE, is without any errors and the application works fine.
The Sloeber's app had included by itself the project's libraries from my original Arduino's "libraries" directory.
The other libraries were compiled correctly:  aJson-master, DallasTemperature, Logging, OneWire. Over the ESPAsyncWebServer-master there is also the AsyncTCP-master library, not yet compiled. I hope this will not have any problems.

In Sloeber happen this error:
-------------------------
In file included from C:\Users\gianp\Documents\Arduino\libraries\ESPAsyncWebServer-master\src\WebAuthentication.cpp:24:0:
c:\sloeber\arduinoplugin\packages\esp32\hardware\esp32\1.0.1\tools\sdk\include\mbedtls\mbedtls\md5.h:34:10: error: #include expects "FILENAME" or <FILENAME>
 #include MBEDTLS_CONFIG_FILE
          ^
libraries\ESPAsyncWebServer-master\src\subdir.mk:60: recipe for target 'libraries\ESPAsyncWebServer-master\src\WebAuthentication.cpp.o' failed
make: *** [libraries\ESPAsyncWebServer-master\src\WebAuthentication.cpp.o] Error 1
"C:/sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.
-------------------------

Below the complete output:
-------------------------
Building file: C:\Users\gianp\Documents\Arduino\libraries\ESPAsyncWebServer-master\src\WebAuthentication.cpp'
'Starting C++ compile'
"C:\sloeber\arduinoPlugin\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/config" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/app_trace" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/app_update" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/asio" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/bootloader_support" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/bt" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/coap" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/console" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/driver" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp-tls" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp32" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp_adc_cal" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp_event" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp_http_client" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp_http_server" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp_https_ota" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp_https_server" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp_ringbuf" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/ethernet" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/expat" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/fatfs" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/freemodbus" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/freertos" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/heap" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/idf_test" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/jsmn" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/json" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/libsodium" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/log" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/lwip" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/mbedtls" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/mdns" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/micro-ecc" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/mqtt" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/newlib" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/nghttp" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/nvs_flash" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/openssl" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/protobuf-c" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/protocomm" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/pthread" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/sdmmc" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/smartconfig_ack" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/soc" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/spi_flash" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/spiffs" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/tcp_transport" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/tcpip_adapter" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/ulp" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/unity" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/vfs" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/wear_levelling" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/wifi_provisioning" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/wpa_supplicant" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/xtensa-debug-module" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp32-camera" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/esp-face" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.1/tools/sdk/include/fb_gfx" -std=gnu++11 -fno-exceptions -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wextra -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti -MMD -c -DF_CPU=240000000L -DARDUINO=10802 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="ESP32_DEV" -DARDUINO_VARIANT="doitESP32devkitV1"  -DESP32 -DCORE_DEBUG_LEVEL=1    -I"C:\sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.1\cores\esp32" -I"C:\Users\gianp\Documents\Arduino\libraries\Logging" -I"C:\sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.1\variants\doitESP32devkitV1" -I"C:\Users\gianp\Documents\Arduino\libraries\aJson-master\utility" -I"C:\Users\gianp\Documents\Arduino\libraries\aJson-master" -I"C:\Users\gianp\Documents\Arduino\libraries\DallasTemperature" -I"C:\Users\gianp\Documents\Arduino\libraries\OneWire" -I"C:\Users\gianp\Documents\Arduino\libraries\AsyncTCP-master\src" -I"C:\Users\gianp\Documents\Arduino\libraries\ESPAsyncWebServer-master\src" -I"C:\sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.1\libraries\FS\src" -I"C:\Users\gianp\Documents\Arduino\libraries\Lo [... cut ...]
c:\sloeber\arduinoplugin\packages\esp32\hardware\esp32\1.0.1\tools\sdk\include\mbedtls\mbedtls\md5.h:34:10: error: #include expects "FILENAME" or <FILENAME>
 #include MBEDTLS_CONFIG_FILE
          ^
libraries\ESPAsyncWebServer-master\src\subdir.mk:60: recipe for target 'libraries\ESPAsyncWebServer-master\src\WebAuthentication.cpp.o' failed
make: *** [libraries\ESPAsyncWebServer-master\src\WebAuthentication.cpp.o] Error 1
"C:/sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.
-------------------------

schreibfaul1, 2019-05-20:
Hello gjanpjetro,
this is a problem only in new release 4.3.
As a workaround you can edit platform.txt, add angle brackets around mbedtls/esp_config.h
[image]
Set in proprties/Arduino the Platform folder again, then press <Apply>

Jilocasin, 2019-05-25:
Having the same error, this unfortunately didn't solve my issue.
Temporarily modifying the relevant header files made it work for me though.

See here: https://github.com/Sloeber/arduino-eclipse-plugin/issues/1028#issuecomment-427033193

closed 2019-10-07 by jantje