Sloeber/arduino-eclipse-plugin #1088: I can not compile the project.

gtleon, 2019-04-21:
Hello!
I can not compile the project.
What operating system parameters are needed to compile a project. Thank!

I stick a log file:

[INFO] Scanning for projects...
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[INFO] Computing target platform for MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.3.1-SNAPSHOT @ D:\java\proj\arduino-eclipse-plugin-4_3_1\io.sloeber.feature\pom.xml
[INFO] Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.9/
[INFO] Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.9/
[INFO] Adding repository http://download.eclipse.org/eclipse/updates/4.9
[INFO] Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.9/R-4.9-201809060745/
[INFO] Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.9/R-4.9-201809060745/
[INFO] Fetching p2.index from http://download.eclipse.org/releases/latest/
[INFO] Fetching p2.index from http://download.eclipse.org/releases/latest/
[INFO] Adding repository http://download.eclipse.org/releases/latest
[INFO] Fetching p2.index from http://download.eclipse.org/releases/2019-03/
[INFO] Fetching p2.index from http://download.eclipse.org/releases/2019-03/
[INFO] Fetching p2.index from https://download.eclipse.org/technology/epp/packages/2019-03/
[INFO] Fetching p2.index from https://download.eclipse.org/technology/epp/packages/2019-03/
[INFO] Fetching p2.index from http://download.eclipse.org/releases/2019-03/201903201000/
[INFO] Fetching p2.index from http://download.eclipse.org/releases/2019-03/201903201000/
[INFO] Fetching p2.index from http://download.eclipse.org/nebula/releases/latest/
[INFO] Fetching p2.index from http://download.eclipse.org/nebula/releases/latest/
[INFO] Adding repository http://download.eclipse.org/nebula/releases/latest
[INFO] Fetching p2.index from https://dl.bintray.com/gnu-mcu-eclipse/v4-neon-updates/ (172B)
[INFO] Fetching p2.index from https://dl.bintray.com/gnu-mcu-eclipse/v4-neon-updates/ (172B)
[INFO] Adding repository https://dl.bintray.com/gnu-mcu-eclipse/v4-neon-updates
[INFO] Resolving dependencies of MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.3.1-SNAPSHOT @ D:\java\proj\arduino-eclipse-plugin-4_3_1\io.sloeber.feature\pom.xml
[INFO] {osgi.os=win32, osgi.ws=win32, org.eclipse.update.install.features=true, osgi.arch=x86_64}
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: io.sloeber.feature.feature.group 4.3.1.qualifier
[ERROR]   Missing requirement: org.eclipse.tools.templates.freemarker.java11 1.0.0.201812111206 requires 'osgi.ee; (&(osgi.ee=JavaSE)(version=11))' but it could not be found
[ERROR]   Cannot satisfy dependency: io.sloeber.feature.feature.group 4.3.1.qualifier depends on: org.eclipse.equinox.p2.iu; org.eclipse.cdt.feature.group 9.5.0
[ERROR]   Cannot satisfy dependency: org.eclipse.cdt.feature.group 9.7.0.201903092251 depends on: org.eclipse.equinox.p2.iu; org.eclipse.cdt.platform.feature.group [9.7.0.201903092251,9.7.0.201903092251]
[ERROR]   Cannot satisfy dependency: org.eclipse.cdt.platform.feature.group 9.7.0.201903092251 depends on: org.eclipse.equinox.p2.iu; org.eclipse.tools.templates.freemarker.java11 [1.0.0.201812111206,1.0.0.201812111206]
[ERROR] 
[ERROR] See http://wiki.eclipse.org/Tycho/Dependency_Resolution_Troubleshooting for help.
[ERROR] Cannot resolve dependencies of MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.3.1-SNAPSHOT @ D:\java\proj\arduino-eclipse-plugin-4_3_1\io.sloeber.feature\pom.xml: See log for details -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MavenExecutionException

cfnz, 2019-06-23:
I have the same issue on linux

rupam-deb, 2019-07-19:
I am also with the same issue. Try to solve it for long time.....

uzi18, 2019-07-19:
I need to try to compile, it was working months before =)
What is your command line to compile ?

rupam-deb, 2019-07-19:
My system is Linux mint and using Terminator.

jantje, 2019-07-21:
I build it 2 days ago on windows. Worked fine.
These problems are mostly temporary hickups.

rupam-deb, 2019-07-22:
From my Linux environment, when I want to run this command: mvn clean verify -DskipTests=true

**I have received this error messages:**

WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.google.inject.internal.cglib.core.$ReflectUtils$1 (file:/usr/share/maven/lib/guice.jar) to method java.lang.ClassLoader.defineClass(java.lang.String,byte[],int,int,java.security.ProtectionDomain)
WARNING: Please consider reporting this to the maintainers of com.google.inject.internal.cglib.core.$ReflectUtils$1
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
[INFO] Scanning for projects...
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[INFO] Computing target platform for MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.3.1-SNAPSHOT @ /home/smartaindustrial/arduino-eclipse-plugin/io.sloeber.feature/pom.xml
[INFO] Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.10/
[INFO] Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.10/
[INFO] Adding repository http://download.eclipse.org/eclipse/updates/4.10
[INFO] Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.10/R-4.10-201812060815/
[INFO] Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.10/R-4.10-201812060815/
[INFO] Fetching p2.index from http://download.eclipse.org/releases/2018-12/
[INFO] Fetching p2.index from http://download.eclipse.org/releases/2018-12/
[INFO] Adding repository http://download.eclipse.org/releases/2018-12
[INFO] Fetching p2.index from https://download.eclipse.org/technology/epp/packages/2018-12/
[INFO] Fetching p2.index from https://download.eclipse.org/technology/epp/packages/2018-12/
[INFO] Fetching p2.index from http://download.eclipse.org/releases/2018-12/201812191000/
[INFO] Fetching p2.index from http://download.eclipse.org/releases/2018-12/201812191000/
[INFO] Fetching p2.index from http://download.eclipse.org/nebula/releases/latest/
[INFO] Fetching p2.index from http://download.eclipse.org/nebula/releases/latest/
[INFO] Adding repository http://download.eclipse.org/nebula/releases/latest
[INFO] Fetching p2.index from http://gnu-mcu-eclipse.netlify.com/v4-neon-updates/ (172B)
[INFO] Adding repository http://gnu-mcu-eclipse.netlify.com/v4-neon-updates
[INFO] Resolving dependencies of MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.3.1-SNAPSHOT @ /home/smartaindustrial/arduino-eclipse-plugin/io.sloeber.feature/pom.xml
[INFO] {osgi.os=linux, osgi.ws=gtk, org.eclipse.update.install.features=true, osgi.arch=x86_64}
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: io.sloeber.feature.feature.group 4.3.1.qualifier
[ERROR]   Missing requirement: org.eclipse.tools.templates.freemarker.java11 1.0.0.201812111206 requires 'osgi.ee; (&(osgi.ee=JavaSE)(version=11))' but it could not be found
[ERROR]   Cannot satisfy dependency: io.sloeber.feature.feature.group 4.3.1.qualifier depends on: org.eclipse.equinox.p2.iu; org.eclipse.cdt.feature.group 9.5.0
[ERROR]   Cannot satisfy dependency: org.eclipse.cdt.feature.group 9.6.0.201812111211 depends on: org.eclipse.equinox.p2.iu; org.eclipse.cdt.platform.feature.group [9.6.0.201812111211,9.6.0.201812111211]
[ERROR]   Cannot satisfy dependency: org.eclipse.cdt.platform.feature.group 9.6.0.201812111211 depends on: org.eclipse.equinox.p2.iu; org.eclipse.tools.templates.freemarker.java11 [1.0.0.201812111206,1.0.0.201812111206]
[ERROR] 
[ERROR] See http://wiki.eclipse.org/Tycho/Dependency_Resolution_Troubleshooting for help.
[ERROR] Cannot resolve dependencies of MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.3.1-SNAPSHOT @ /home/smartaindustrial/arduino-eclipse-plugin/io.sloeber.feature/pom.xml: See log for details -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MavenExecutionException

rupam-deb, 2019-07-22:
jantje jantje, could you please confirm me on Linux.

rupam-deb, 2019-07-22:
@jantje , could you please confirm me on Linux.

jantje, 2019-07-22:
I have repurposed my linux setups. I do not have any llinux machine available.
Looking at the logs you can clearly see that both builds use a different CDT version
```
[ERROR] Cannot satisfy dependency: io.sloeber.feature.feature.group 4.3.1.qualifier depends on: org.eclipse.equinox.p2.iu; org.eclipse.cdt.feature.group 9.5.0
[ERROR] Cannot satisfy dependency: org.eclipse.cdt.feature.group 9.7.0.201903092251 depends on
```: 

```
[ERROR] Cannot satisfy dependency: io.sloeber.feature.feature.group 4.3.1.qualifier depends on: org.eclipse.equinox.p2.iu; org.eclipse.cdt.feature.group 9.5.0
[ERROR] Cannot satisfy dependency: org.eclipse.cdt.feature.group 9.6.0.201812111211 depends on: 
```
A gogle search leads me to this page https://www.eclipse.org/lists/epp-dev/msg05368.html which makes me think this is a java 11 and cdt related issue

rupam-deb, 2019-07-23:
Thanks for your reply @jantje . I follow your steps from the GITHUB. I also noticed it before with two different CDT (9.5.0 and 9.6.0).

Could you please suggest how can I get rid off these errors?

jantje, 2019-07-23:
Try with java8

rupam-deb, 2019-07-24:
@jantje , Try with Java8 and same errors. Don't get the point where is the problem?

jantje, 2019-08-08:
I don't know what the problem is. travis build works fine with java8 and that is on linux.

GHPS, 2019-08-19:
>  travis build works fine with java8 and that is on linux.

Well - looks like Java10 to me [1] - and that works fine
with osgi.

At least the problem is wide spread, so there are
some clues how to solve it [2].

>Try with Java8 and same errors. 

@rupam-deb How did you try to run maven with Java 8?
Here osgi should indeed work fine.

1: https://oss.sonatype.org/content/repositories/snapshots/com/github/brodykenrick/arduino-eclipse-plugin/io.sloeber.product/4.3.1-SNAPSHOT/io.sloeber.product-4.3.1-20190808.141233-83-p2metadata.xml
2: http://www.descher.at/descher-vu/2019/01/java-11-jaxb-and-osgi/

jantje, 2019-08-19:
> Well - looks like Java10 to me [1] - and that works fine

I'm not the expert on this so I won't disagree. The reason I state there is a need for java8 is because of #1100 issue

GHPS, 2019-08-20:
> The reason I state there is a need for java8 is because of #1100 issue

 @jantje Good to see you back :-)

The sad truth is that Java projects are getting increasingly difficult to
maintain. There are numerous Java 8 projects out there with developers
trying to figure out how to migrate - only - to Java 11.

Back to the question at hand: Chances are good that the osgi issues
is a Java 11 problem only - any version including Java 10 should be
fine.

However, it's not a trivial task to convince maven to stick to Java 8 when
there is also Java 11 installed (the default, for example, on Ubuntu 18.04).
I spend an fruitless hour on this problem - but I'm only a Python developer...

Therefore the question:  @rupam-deb How did you manage to run maven 
with Java 8?

GHPS, 2019-08-20:
I just checked: The issue is solved with the latest commits.

@wimjongman  Thanks for the great work!

wimjongman, 2019-08-20:
@GHPS you're welcome. I expect there are no issues but if there are they should be in the area of networking. We have replaced the http4 client with the http45 client.

Please re-open if you find any issues.

closed 2019-08-20 by wimjongman

uzi18, 2019-08-20:
@wimjongman still can't build:
```
$ mvn clean verify -PSDK,latest -DskipTests=true
[INFO] Scanning for projects...
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[INFO] Computing target platform for MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.3.1-SNAPSHOT @ /home/users/linaja/arduino-eclipse-plugin/io.sloeber.feature/pom.xml
[INFO] Resolving dependencies of MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.3.1-SNAPSHOT @ /home/users/linaja/arduino-eclipse-plugin/io.sloeber.feature/pom.xml
[INFO] {osgi.os=linux, osgi.ws=gtk, org.eclipse.update.install.features=true, osgi.arch=x86_64}
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: io.sloeber.feature.feature.group 4.3.1.qualifier
[ERROR]   Missing requirement: io.sloeber.feature.feature.group 4.3.1.qualifier requires 'org.eclipse.equinox.p2.iu; org.eclipse.cdt.feature.group 9.5.0' but it could not be found
[ERROR] 
[ERROR] See http://wiki.eclipse.org/Tycho/Dependency_Resolution_Troubleshooting for help.
[ERROR] Cannot resolve dependencies of MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.3.1-SNAPSHOT @ /home/users/linaja/arduino-eclipse-plugin/io.sloeber.feature/pom.xml: See log for details -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MavenExecutionException

```

jantje, 2019-08-20:
@wimjongman 
Thanks Wim :-)

uzi18, 2019-08-20:
@jantje @wimjongman how to totally clean build env?

GHPS, 2019-08-20:
@uzi18 Did you try to clear the maven cache? 
`rm -rf ~/.m2`

uzi18, 2019-08-20:
will try, I don't know maven/java stuff
my build env before worked for linux32 and now for linux64 have such errors
thanks for hint

wimjongman, 2019-08-21:
@uzi18 fixed the SDK build as well. Apologies for the oversight.

mvn clean verify -PSDK,latest -DskipTests=true

GHPS, 2019-08-21:
@wimjongman Since your patches not only solves the issue discussed here 
but also gives us the latest Eclipse version 4.12 I suggest to advance to
Sloeber version number 4.4.

uzi18, 2019-08-21:
have cleaned up repository for mvn, but still get : 
```
[INFO] Computing target platform for MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.3.1-SNAPSHOT @ /home/users/linaja/arduino-eclipse-plugin/io.sloeber.feature/pom.xml
[INFO] Resolving dependencies of MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.3.1-SNAPSHOT @ /home/users/linaja/arduino-eclipse-plugin/io.sloeber.feature/pom.xml
[INFO] {osgi.os=linux, osgi.ws=gtk, org.eclipse.update.install.features=true, osgi.arch=x86_64}
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: io.sloeber.feature.feature.group 4.3.1.qualifier
[ERROR]   Missing requirement: io.sloeber.feature.feature.group 4.3.1.qualifier requires 'org.eclipse.equinox.p2.iu; org.eclipse.cdt.feature.group 9.5.0' but it could not be found
[ERROR] 
[ERROR] See http://wiki.eclipse.org/Tycho/Dependency_Resolution_Troubleshooting for help.
[ERROR] Cannot resolve dependencies of MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.3.1-SNAPSHOT @ /home/users/linaja/arduino-eclipse-plugin/io.sloeber.feature/pom.xml: See log for details -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MavenExecutionException

```

wimjongman, 2019-08-22:
Please comb through the whole log to see what is going on. There must be some other clues in there.

It works for me. I did:

* cd
* rm -R .m2
* mvn clean verify -PSDK,latest,linux64 -DskipTests=true

Result:

> [INFO] ------------------------------------------------------------------------
> [INFO] Reactor Summary:
> [INFO]
> [INFO] io.sloeber.parent .................................. SUCCESS [ 39.215 s]
> [INFO] io.sloeber.application ............................. SUCCESS [ 38.564 s]
> [INFO] io.sloeber.core .................................... SUCCESS [  1.027 s]
> [INFO] io.sloeber.core.nl1 ................................ SUCCESS [  0.041 s]
> [INFO] io.sloeber.ui ...................................... SUCCESS [  0.403 s]
> [INFO] io.sloeber.ui.nl1 .................................. SUCCESS [  0.019 s]
> [INFO] io.sloeber.feature ................................. SUCCESS [  0.101 s]
> [INFO] io.sloeber.product.sdk ............................. SUCCESS [ 33.192 s]
> [INFO] io.sloeber.tests ................................... SUCCESS [  0.372 s]
> [INFO] root ............................................... SUCCESS [  0.188 s]
> [INFO] ------------------------------------------------------------------------
> [INFO] BUILD SUCCESS
> [INFO] ------------------------------------------------------------------------
> [INFO] Total time: 09:27 min
> [INFO] Finished at: 2019-08-22T11:27:49+02:00
> [INFO] Final Memory: 124M/1386M
> [INFO] ------------------------------------------------------------------------
>

wimjongman, 2019-08-22:
If you can't find it pipe the whole log to a file and attach it here (not copy paste).

mvn clean verify -PSDK,latest,linux64 -DskipTests=true > mvn.log

uzi18, 2019-08-22:
[mvn.log](https://github.com/Sloeber/arduino-eclipse-plugin/files/3531477/mvn.log)

wimjongman, 2019-08-23:
Weird. Please attach the output of the following commands here

git status
git rev-parse HEAD
git log

uzi18, 2019-08-23:
@wimjongman here you are:

```
[linaja@pldmachine arduino-eclipse-plugin]$ git status
On branch master
Your branch is up to date with 'uzi18/master'.

nothing to commit, working tree clean
[linaja@pldmachine arduino-eclipse-plugin]$ git rev-parse HEAD
572fedd84ee838d46fa29b17ca841468a60c9b75
[linaja@pldmachine arduino-eclipse-plugin]$ git log
commit 572fedd84ee838d46fa29b17ca841468a60c9b75 (HEAD -> master, uzi18/master, origin/master, origin/HEAD)
Author: Wim Jongman <wim.jongman@remainsoftware.com>
Date:   Wed Aug 21 11:57:07 2019 +0200

    I can not compile the project. #1088

commit 06983d15e5db59387d52167ea6338a23f0f15de4
Author: Wim Jongman <wim.jongman@remainsoftware.com>
Date:   Tue Aug 20 15:44:13 2019 +0200

    I can not compile the project. #1088
    
    Attempt two for travis

commit e6f78be63e1274a468770f336f2e214b82a87c90
Author: Wim Jongman <wim.jongman@remainsoftware.com>
Date:   Tue Aug 20 15:41:44 2019 +0200

    I can not compile the project. #1088
    
    Fixed the travis build to not use old stuff

commit cc52c2b710f62bbb865c26779aad64e3ad5c2fa1
Author: Wim Jongman <wim.jongman@remainsoftware.com>
Date:   Tue Aug 20 15:33:52 2019 +0200

    I can not compile the project. #1088

commit 8587e051b594621a6067b57e1da6efa31da20b34
Author: jantje <eclipse@baeyens.it>
Date:   Thu Aug 8 20:33:51 2019 +0200

    Fixing warnings cause by eclipse upgrade

commit b27841752e691cc3672bc6b39b2c951f6c8da253
Merge: 5ca7c154 8db54848
Author: jantje <eclipse@baeyens.it>
Date:   Thu Aug 8 15:44:57 2019 +0200

    Merge branch 'master' of https://github.com/Sloeber/arduino-eclipse-plugin.git

commit 5ca7c15405b3a1792b95b990b59e94a7c06b18cb
Author: jantje <eclipse@baeyens.it>
Date:   Thu Aug 8 15:44:31 2019 +0200

    Added dist: trusty

```

uzi18, 2019-08-23:
have cloned clean repository one more time, but same error.

wimjongman, 2019-08-23:
Weird. I don't know what is going on. Looking at the log, it stops after Tycho (which is the Eclipse build engine for maven) is installed. 

the -P parameter activates the repositories. You can see this in the io.sloener.parent/pom.xml file. 

The questions I have are: On what machine are you? Is it a 32 bit machine? What maven version did you install (mvn --version) we would like to see 3.6.0 + here.
Are you behind some corporate firewall? 

please try the non-sdk build:

mvn -clean -verify -DskipTests=true

uzi18, 2019-08-23:
no firewall just NAT, but all rest files are fetched, have tried to find file org.eclipse.cdt.feature.group* here: 
http://download.eclipse.org/tools/cdt/releases/9.5 but cant find it

mirror problem?

maven version and system arch below:

```
$ mvn -clean -verify -DskipTests=true
Apache Maven 3.6.1 (d66c9c0b3152b2e69ee9bac180bb8fcc8e6af555; 2019-04-04T19:00:29Z)
Maven home: /usr/share/maven
Java version: 1.8.0_222, vendor: IcedTea, runtime: /usr/lib64/jvm/icedtea8-3.13.0/jre
Default locale: pl_PL, platform encoding: UTF-8
OS name: "linux", version: "4.20.2-1", arch: "amd64", family: "unix"
[linaja@pldmachine arduino-sloeber]$ mvn clean verify -DskipTests=true
[INFO] Scanning for projects...
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[INFO] Computing target platform for MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.3.1-SNAPSHOT @ /home/users/linaja/arduino-sloeber/io.sloeber.feature/pom.xml
[INFO] Resolving dependencies of MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.3.1-SNAPSHOT @ /home/users/linaja/arduino-sloeber/io.sloeber.feature/pom.xml
[INFO] {osgi.os=linux, osgi.ws=gtk, org.eclipse.update.install.features=true, osgi.arch=x86_64}
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: io.sloeber.feature.feature.group 4.3.1.qualifier
[ERROR]   Missing requirement: io.sloeber.feature.feature.group 4.3.1.qualifier requires 'org.eclipse.equinox.p2.iu; org.eclipse.cdt.feature.group 9.5.0' but it could not be found
[ERROR] 
[ERROR] See http://wiki.eclipse.org/Tycho/Dependency_Resolution_Troubleshooting for help.
[ERROR] Cannot resolve dependencies of MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.3.1-SNAPSHOT @ /home/users/linaja/arduino-sloeber/io.sloeber.feature/pom.xml: See log for details -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MavenExecutionException

```

wimjongman, 2019-08-23:
It does not seem to start loading. It looks like to does not see any repositories. Crazy.

To force not to use mirrors try this: 

mvn -clean -install -DskipTests=true -Dtycho.disableP2Mirrors=true -Dtycho.localArtifacts=ignore

uzi18, 2019-08-24:
@wimjongman just downloaded+installed maven,jdk 8, git etc.
cloned one more time but on completely different host with windows 7 64bit.
and have the same error, so something is wrong.

have also tested there "mvn clean install -DskipTests=true -Dtycho.disableP2Mirrors=true -Dtycho.localArtifacts=ignore"

uzi18, 2019-08-24:
```
$ mvn clean verify -Platest,linux64 -DskipTests=true
[INFO] Scanning for projects...
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Build Order:
[INFO] 
[INFO] io.sloeber.parent                                                  [pom]
[INFO] root                                                               [pom]
[INFO] 
[INFO] --< com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.parent >--
[INFO] Building io.sloeber.parent 4.3.1-SNAPSHOT                          [1/2]
[INFO] --------------------------------[ pom ]---------------------------------
[INFO] 
[INFO] --- maven-clean-plugin:2.5:clean (default-clean) @ io.sloeber.parent ---
[INFO] 
[INFO] --- maven-enforcer-plugin:1.2:enforce (enforce-maven) @ io.sloeber.parent ---
[INFO] 
[INFO] --------< com.github.brodykenrick.arduino-eclipse-plugin:root >---------
[INFO] Building root 2.2.0                                                [2/2]
[INFO] --------------------------------[ pom ]---------------------------------
[INFO] 
[INFO] --- maven-clean-plugin:2.5:clean (default-clean) @ root ---
[INFO] 
[INFO] --- maven-enforcer-plugin:1.0:enforce (enforce-maven) @ root ---
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary:
[INFO] 
[INFO] io.sloeber.parent 4.3.1-SNAPSHOT ................... SUCCESS [  0.631 s]
[INFO] root 2.2.0 ......................................... SUCCESS [  0.163 s]
[INFO] ------------------------------------------------------------------------
[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  2.436 s
[INFO] Finished at: 2019-08-24T07:03:50Z                                                                                                                                                                                         
[INFO] ------------------------------------------------------------------------ 
```

but no target directory created

jantje, 2019-08-24:
>$ mvn clean verify -Platest,linux64 -DskipTests=true

this is building for linux. Try a windows build
$ mvn clean verify -Platest -DskipTests=true

uzi18, 2019-08-24:
yes have tried under windows same error, but on linux latest,linux64 does not create any product or new files in product dir

wimjongman, 2019-08-26:
You must be in the root directory of the repository to build. It looks like you are in io.sloeber.parent