Sloeber/arduino-eclipse-plugin #1090: Create new sketch fails for ESP8266 with version 4

Harry4Offies, 2019-05-21:
**Describe the bug**
Installed latest V4.3.1_win64.2018-10-10_08-21-58.tar.gz. 
Creation of a new ESP8266 sketch failed after a few seconds with

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Arduino Creat new Sketch and click on it
2. Configure like screenshot
[image]

3. Wait a few seconds
4. See error (sreen shot and logging)
[image]
[Logging.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/3204441/Logging.txt)

**Expected behavior**
A simple empty sketch to be created

**Desktop (please complete the following information):**
 - OS: Windows 10
 - N.A.
 - N.A.

jonnieZG, 2019-05-28:
Same here. I have tested with WEMOS D1 R2 mini and several others, but the same problem persist when I try to use 2.5.1 or 2.5.2. I'd say it's a Java version problem related to the Regex Pattern resolution.

[image]

And here's the entire log of the event. Just rewind to its start and you'll see a StackOverflowError caused due to an internal Java regex Pattern resolution error. I've seen it happen before on several occasions in other Java apps, so my solution was to rephrase a problematic regex pattern.

[log.zip](https://github.com/Sloeber/arduino-eclipse-plugin/files/3226048/log.zip)

smartmicrodevel, 2019-06-18:
Same issue here, would love to use 2.5.2!
Any workaround?

jonnieZG, 2019-06-18:
> Same issue here, would love to use 2.5.2!
> Any workaround?

I couldn't find any, so I'm stuck on 2.5.0.

BbIKTOP, 2019-06-18:
"Latest" nightly build works ok but does not compile due to some empty parameters it tries to pass to the esptool.py. Tried to modify ExternalCommandLauncher constructor that does String#split with regex, but cannot build it. So I changed esptool.py to eliminatine empty args:
```
import sys
import os

i=0
while(i<len(sys.argv)):
    if str(sys.argv[i])=='':
        sys.argv.pop(i)
    else:
        i=i+1

sys.argv.pop(0) # Remove executable name
```

But looks like nobody maintains Sloeber now. It's sad. Probably it's time to move to the PlatformIO (

smartmicrodevel, 2019-06-19:
> > Same issue here, would love to use 2.5.2!
> > Any workaround?
> 
> I couldn't find any, so I'm stuck on 2.5.0.

Tried increasing stack size of JVM via -Xss512m but even that won't help. What happend to the guys who maintained this?

jonnieZG, 2019-06-19:
> Tried increasing stack size of JVM via -Xss512m but even that won't help.
There is a recursive loop that Regex parser falls into. Increasing memory won't help, the Regex has to be rephrased.

> What happend to the guys who maintained this?
I guess that Life is what happened. ;) Maintaining an open source is OK when you are young and you don't have any other worries, but later on in life all these things fall down on your list of priorities, especially if they don't pay, even more if they require additional investments on your behalf.

tvand, 2019-06-21:
I tried to debug the problem. It is in /io.sloeber.core/src/io/sloeber/core/tools/Helpers.java#setTheEnvironmentVariablesPostProcessing()

The problem happens when variables are used twice in a recipe. I will try to provide a patch but meanwhile you may work around the problem by changing 
`tools.esptool.upload.pattern="{cmd}" "{runtime.platform.path}/tools/upload.py" --chip esp8266 --port "{serial.port}" --baud "{upload.speed}" "{upload.verbose}" {upload.erase_cmd} --end --chip esp8266 --port "{serial.port}" --baud "{upload.speed}" "{upload.verbose}" write_flash 0x0 "{build.path}/{build.project_name}.bin" --end`

to

`tools.esptool.upload.pattern="{cmd}" "{runtime.platform.path}/tools/upload.py" --chip esp8266 --port "{serial.port}" --baud "{upload.speed}" "{upload.verbose}" {upload.erase_cmd} --end --chip esp8266 --port "{serial.port}" --baud "{upload.speed}" write_flash 0x0 "{build.path}/{build.project_name}.bin" --end`

in platform.txt.

tvand, 2019-06-21:
> I guess that Life is what happened. ;) Maintaining an open source is OK when you are young and you don't have any other worries, but later on in life all these things fall down on your list of priorities, especially if they don't pay, even more if they require additional investments on your behalf.

The latest activity in the repository was the removal of requests for donations. Go figure.

tvand, 2019-06-21:
Fixed in #1073

jonnieZG, 2019-06-22:
Ok, seems to be working now. 

1. Go to Window >> Preferences >> Install/Update >> Available Software Sites and add this location (you can name it Sloeber Nightly or whatever you like) **http://eclipse.baeyens.it/update/V4/nightly/**
2. Help >> Check for Updates
3. Select **only** **Sloeber** and **Sloeber Plugin** and install that.

Should be working fine when done.

amirna2, 2019-09-02:
Still failing on Mac OSX using latest Eclipse CDT Version: 2019-06 (4.12.0) and the latest nightly plugin build version 4.3.1.201902260606.
Used the steps above: https://github.com/Sloeber/arduino-eclipse-plugin/issues/1090#issuecomment-504675603

erdemuncuoglu, 2019-09-05:
Hi @amirna2
Your issue has a solution. Check my comment in #1102.

closed 2019-10-07 by jantje