Sloeber/arduino-eclipse-plugin #1091: Enum types

alidaf, 2019-06-04:
I cannot compile a program that has an enumerated type passed as a parameter:

```
enum ActuatorCommand {ACT_STOP = 0, ACT_CLOSE, ACT_OPEN, ACT_CMDS};
ActuatorCommand actuatorCmd = ActuatorCommand::ACT_STOP;

void printCommand(ActuatorCommand cmd)
{
  stuff
}

void setup()
{
  stuff
  printCommand(actuatorCmd);
}

void loop()
{
  stuff
}
```
The build error is ...

../sloeber.ino.cpp:11:19: error: variable or field 'printCommand' declared void
 void printCommand(ActuatorCommand cmd) ;
                   ^
../sloeber.ino.cpp:11:19: error: 'ActuatorCommand' was not declared in this scope
subdir.mk:18: recipe for target 'sloeber.ino.cpp.o' failed
make: *** [sloeber.ino.cpp.o] Error 1

This works in the Arduino IDE so I can't figure out the problem here.

alidaf, 2019-06-04:
I managed a workaround by moving the enumerated type into a separate header file and including a compile guard.

erdemuncuoglu, 2019-09-02:
IMAO this is more a benefit than an issue and your "workaround" is in fact what you should do (Let me explain).

Arduino IDE creates the header files with all includes and declarations for you and does this right so you don't need to care for declaration/definition. This is a very juicy decision for attracting (new) bees into the platform.

On the other hand Sloeber is for serious coders. It is also supporting .ino files with a similar approach but not completely. If you look inside the automatically created  `sloeber.ino.cpp` file, you'll see that the `printCommand` function has been declared but enum type hasn't.

What I suggest you:

1. If you're doing light coding (like teaching arduino lessons) then it is better to go with .ino files under Arduino IDE.
1. If you'd like to do serious things then go with Sloeber using header and source files.

jantje, 2019-09-04:
duplicate of #775

closed 2019-09-04 by jantje