Sloeber/arduino-eclipse-plugin #1094: Latest nightly does not upload compiled sketch

BbIKTOP, 2019-06-14:
Latest nightly build failed to upload sketch into esp8266

**To Reproduce**
Create any/empty sketch, then upload it

**Expected behavior**
Sketch shall be uploaded but it's not

**Screenshots**
```
Starting upload
Uploading project "test" with "ESPTOOL"
Starting reset using DTR toggle process
Continuing to use "/dev/cu.SLAB_USBtoUART"
Ending reset

Launching: /Applications/Eclipse.app/Contents/Eclipse//arduinoPlugin/packages/esp8266/tools/python/3.7.2-post1/python /Applications/Eclipse.app/Contents/Eclipse//arduinoPlugin/packages/esp8266/hardware/esp8266/2.5.2/tools/upload.py --chip esp8266 --port /dev/cu.SLAB_USBtoUART --baud 921600  erase_flash --end --chip esp8266 --port /dev/cu.SLAB_USBtoUART --baud 921600  write_flash 0x0 /Users/viktor/eclipse-workspace/test/Release/test.bin --end 
Output:
usage: esptool [-h] [--chip {auto,esp8266,esp32}] [--port PORT] [--baud BAUD]
               [--before {default_reset,no_reset,no_reset_no_sync}]
               [--after {hard_reset,soft_reset,no_reset}] [--no-stub]
               [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,version}
               ...
esptool: error: argument operation: invalid choice: '' (choose from 'load_ram', 'dump_mem', 'read_mem', 'write_mem', 'write_flash', 'run', 'image_info', 'make_image', 'elf2image', 'read_mac', 'chip_id', 'flash_id', 'read_flash_status', 'write_flash_status', 'read_flash', 'verify_flash', 'erase_flash', 'erase_region', 'version')
The execution of command "python" is done.

```

**Desktop (please complete the following information):**
MacOS 10.14.5
Eclipse 4.11.0 20190314-1200
Sloeber, the Eclipse Arduino IDE (plugin version)	4.3.1.201902260606
esp8266 platform 2.5.2

**Additional context**
When I launch the failed command from the command line (bash) it works just fine:
```
viktor@imac:~/eclipse-workspace/test $ /Applications/Eclipse.app/Contents/Eclipse//arduinoPlugin/packages/esp8266/tools/python/3.7.2-post1/python /Applications/Eclipse.app/Contents/Eclipse//arduinoPlugin/packages/esp8266/hardware/esp8266/2.5.2/tools/upload.py --chip esp8266 --port /dev/cu.SLAB_USBtoUART --baud 921600  erase_flash --end --chip esp8266 --port /dev/cu.SLAB_USBtoUART --baud 921600  write_flash 0x0 /Users/viktor/eclipse-workspace/test/Release/test.bin --end
esptool.py v2.6
Serial port /dev/cu.SLAB_USBtoUART
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: cc:50:e3:0c:c9:a1
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Erasing flash (this may take a while)...
Chip erase completed successfully in 6.8s
Hard resetting via RTS pin...
esptool.py v2.6
Serial port /dev/cu.SLAB_USBtoUART
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: cc:50:e3:0c:c9:a1
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 276784 bytes to 195039...
Wrote 276784 bytes (195039 compressed) at 0x00000000 in 4.6 seconds (effective 479.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

```

Here's the sketch:
```
viktor@imac:~/eclipse-workspace/test $ cat test.ino
#include "Arduino.h"
//The setup function is called once at startup of the sketch
void setup()
{
// Add your initialization code here
}

// The loop function is called in an endless loop
void loop()
{
//Add your repeated code here
}
```

BbIKTOP, 2019-06-14:
As expected, there's an empty argument passed. I've added the arguments print to the python script , here it is:
```
Number of arguments: 21 arguments.
Argument List: ['/Applications/Eclipse.app/Contents/Eclipse//arduinoPlugin/packages/esp8266/hardware/esp8266/2.5.2/tools/upload.py', '--chip', 'esp8266', '--port', '/dev/cu.SLAB_USBtoUART', '--baud', '921600', '', 'erase_flash', '--end', '--chip', 'esp8266', '--port', '/dev/cu.SLAB_USBtoUART', '--baud', '921600', '', 'write_flash', '0x0', '/Users/viktor/eclipse-workspace/test/Release/test.bin', '--end']

```

erdemuncuoglu, 2019-08-27:
This issue seems to have a solution in #1073 or #1090.
Any of them includes a workaround.

wimjongman, 2019-08-30:
You are not working on the latest nightly:

Sloeber, the Eclipse Arduino IDE (plugin version) 4.3.1.201902260606

2019/02/26 06:06

closed 2019-08-30 by wimjongman