Sloeber/arduino-eclipse-plugin #1096: Failed to create project

hoo2, 2019-07-07:
**Description**
Whenever I try to create a new Arduino project it fails with the same error.

I have install the "The Arduino Eclipse plugin named Sloeber" from marketplace. I use eclipse version: 2019-06 (4.12.0)

**To Reproduce**
Steps to reproduce the behavior (In my system):
1) right click on Project explorer
2) New->Create Arduino Sketch
3) Fill any configuration and click Next
4) Select Default (ino ar cpp) file and click Finish
5) See the error.

**Screenshots**
[image]
[image]
[image]

**Desktop:**
 - OS: Win7 64bit
 - eclipse version: 2019-06 (4.12.0)
 - Soeber version: V4 I can not find the exact version number but the installation date is 2019-07-07

**Error details - Log:**
```
eclipse.buildId=4.12.0.I20190605-1800
java.version=1.8.0_161
java.vendor=Oracle Corporation
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=en_US
Framework arguments:  -product org.eclipse.epp.package.cpp.product
Command-line arguments:  -os win32 -ws win32 -arch x86_64 -product org.eclipse.epp.package.cpp.product

io.sloeber.core.ui
Error
Sun Jul 07 19:35:20 EEST 2019
Failed to create project

java.lang.reflect.InvocationTargetException
	at org.eclipse.jface.operation.ModalContext.runInCurrentThread(ModalContext.java:449)
	at org.eclipse.jface.operation.ModalContext.run(ModalContext.java:354)
	at org.eclipse.jface.wizard.WizardDialog.run(WizardDialog.java:1035)
	at io.sloeber.ui.wizard.newsketch.NewSketchWizard.performFinish(NewSketchWizard.java:99)
	at org.eclipse.jface.wizard.WizardDialog.finishPressed(WizardDialog.java:833)
	at org.eclipse.jface.wizard.WizardDialog.buttonPressed(WizardDialog.java:472)
	at org.eclipse.jface.dialogs.Dialog.lambda$0(Dialog.java:622)
	at org.eclipse.swt.events.SelectionListener$1.widgetSelected(SelectionListener.java:84)
	at org.eclipse.swt.widgets.TypedListener.handleEvent(TypedListener.java:252)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4131)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1056)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:3944)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3547)
	at org.eclipse.jface.window.Window.runEventLoop(Window.java:823)
	at org.eclipse.jface.window.Window.open(Window.java:799)
	at org.eclipse.ui.internal.navigator.wizards.WizardShortcutAction.run(WizardShortcutAction.java:102)
	at org.eclipse.jface.action.Action.runWithEvent(Action.java:474)
	at org.eclipse.jface.action.ActionContributionItem.handleWidgetSelection(ActionContributionItem.java:580)
	at org.eclipse.jface.action.ActionContributionItem.lambda$4(ActionContributionItem.java:412)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4131)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1056)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:3944)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3547)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1173)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1062)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:155)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:635)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:559)
[... 57 lines cut ...]
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1056)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:3944)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3547)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1173)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1062)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:155)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:635)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:559)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:150)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:155)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:137)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:107)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:400)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:660)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:597)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1468)
```

alidaf, 2019-07-11:
I have this problem too. It started today after using it for the past few weeks with no problems.

judalvarezca, 2019-07-16:
I have the same behaviour since today. It was working until last week.

jantje, 2019-07-16:
I guess you upgraded cdt to a newer version

rookie10, 2019-07-30:
Thats true after the update to 2019.06 it dosent work anymore.
is there any fixes possible

erdemuncuoglu, 2019-08-06:
To workaround this you need to downgrade CDT. Latest version which worked for me was 9.7.2 while I'm using Sloeber nightly version.

Steps to apply the workaround.
1. Add CDT 9.7 repository to Eclipse. (update site: http://download.eclipse.org/tools/cdt/releases/9.7)
2. Uninstall Sloeber plugin. Be sure that CDT is uninstalled too.
3. When asked restart Eclipse.
4. Install CDT 9.7 (9.7.2 is the latest).
5. Install Sloeber.

Issues with this workaround:
1. You will be locked inside CDT version 9.7.
2. CDT is in main Eclipse repository so you can not disable it. If you fail to prevent Eclipse Updater/Installer from updating CDT you have to re-apply it again. 
3. This is not the most beautiful workaround yet doing the job.

jantje, 2019-08-08:
I just upgraded my development environment, ran a project creation test and it ran fine with CDT 9.8.0.201906071757 and eclipse Version: 4.12.0.v20190605-1801 using java 8 (recommended and 9 may be the root cause!!)
I used the genuino uno as a test.
I have no clue what is causing this behaviour at your side. I except the code is far from top notch and as such may cause "weird behaviour" (but I haven't found a better way, and I did look)

jantje, 2019-08-08:
Refect java 9 search
https://dzone.com/articles/java-19-reflection-and-package-access-changes

erdemuncuoglu, 2019-08-11:
Will this update show in nightly repository?

jantje, 2019-08-11:
> 
> 
> Will this update show in nightly repository?

no only on the travis/sonatov(?) site. I don't quite remember the url.
I learned the hard way that people who do not want to invest time ins looking a solution are also not willing to pay someone who does. But I learned my lesson now.

jantje, 2019-08-22:
Did you test with java8 and what was the outcome?

erdemuncuoglu, 2019-08-22:
In a nutshell it was failed with `java.lang.reflect.InvocationTargetException`.
I've openjdk version 1.8.0_222. I'm using plugin version of Sloeber (thus I setup me a repo to install from).

    java.version=1.8.0_222
    java.vendor=Oracle Corporation
    BootLoader constants: OS=linux, ARCH=x86_64, WS=gtk, NL=en_DK
    Command-line arguments:  -os linux -ws gtk -arch x86_64

jantje, 2019-08-22:
openjdk is not from  oracle.
I guess things are not what the seem.

hoo2, 2019-08-23:
@jantje 
In the head post you can see that `java.lang.reflect.InvocationTargetException` is failing in Oracle's java also.
```
eclipse.buildId=4.12.0.I20190605-1800
java.version=1.8.0_161
java.vendor=Oracle Corporation
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=en_US
Framework arguments:  -product org.eclipse.epp.package.cpp.product
Command-line arguments:  -os win32 -ws win32 -arch x86_64 -product org.eclipse.epp.package.cpp.product

io.sloeber.core.ui
Error
Sun Jul 07 19:35:20 EEST 2019
Failed to create project

java.lang.reflect.InvocationTargetException
	at org.eclipse.jface.operation.ModalContext.runInCurrentThread(ModalContext.java:449)
        ....
```
For now I use Sloeber, but I'll try with java 9 and I'll post the results.

jantje, 2019-08-23:
Right now java is a mess. Java 9 and 10 are already "out of date" but it introduces issues and InvocationTargetException  is one of them. 
Note that the standalone sloeber on windows comes with a java included.

erdemuncuoglu, 2019-08-23:
> openjdk is not from oracle.
> I guess things are not what the seem.

I believe that if it should be fixed with Oracle's version, it would be too with OpenJDK.

Devuan Ascii sources also have OpenJDK 11 (which would be the default if I had upgraded my distro) so I installed it from backports. Nothing changes.

    eclipse.buildId=4.12.0.I20190605-1800
    java.version=11.0.4
    java.vendor=Debian
    BootLoader constants: OS=linux, ARCH=x86_64, WS=gtk, NL=en_DK
    Command-line arguments:  -os linux -ws gtk -arch x86_64

    io.sloeber.core.ui
    Error
    Fri Aug 23 12:55:45 EET 2019
    Failed to create project

    java.lang.reflect.InvocationTargetException
        at org.eclipse.jface.operation.ModalContext.runInCurrentThread(ModalContext.java:449)
        ...

Downgrading CDT is not a bad solution for me right now.
Also I have a Photon installation so I can create a project there and then import it into current one.

I use Eclipse for my everyday needs (both in my job and in my hobbies when I can find time) and Sloeber is a great addition on to it.
I also gave a try to other IDEs but even they have a beautiful chromatic paint on them they are just better than the original one.
Sloeber is a very powerful weaapon for real FREE warriors and a few cracks on it will not hinder a perfect cut.

uzi18, 2019-08-24:
there is also eclipse-cdt-arduino plugin but don't know how to use it and not sure if it is comparable with Sloeber :)

jantje, 2019-08-24:
eclipse-cdt-arduino plugin has nothing to do with Sloeber. I have no experiences with it.

hoo2, 2019-08-25:
@erdemuncuoglu You can install the bundled sloeber version in a separate place and have both the latest eclipse and a functioning version of sloeber. This is what I did.

wimjongman, 2019-08-26:
For the record. It was caused by:

Caused by: java.lang.NoSuchMethodError: org.eclipse.cdt.managedbuilder.internal.core.Configuration.<init>(Lorg/eclipse/cdt/managedbuilder/internal/core/ManagedProject;Lorg/eclipse/cdt/managedbuilder/internal/core/ToolChain;Ljava/lang/String;Ljava/lang/String;)V
	at io.sloeber.core.tools.ShouldHaveBeenInCDT.setCProjectDescription(ShouldHaveBeenInCDT.java:72)
	at io.sloeber.core.api.BoardDescriptor.createProject(BoardDescriptor.java:543)
	at io.sloeber.ui.wizard.newsketch.NewSketchWizard.createProjectWrapper(NewSketchWizard.java:127)
	at io.sloeber.ui.wizard.newsketch.NewSketchWizard$1.execute(NewSketchWizard.java:94)
	at org.eclipse.ui.actions.WorkspaceModifyOperation.lambda$0(WorkspaceModifyOperation.java:110)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2295)

closed 2019-08-26 by wimjongman

erdemuncuoglu, 2019-08-26:
Sloeber (plugin version) is running as smooth as it used to be.

There were some updates by @wimjongman for #1088 which I've missed before posting this:

> In a nutshell it was failed with `java.lang.reflect.InvocationTargetException.`

It seems to be fixed since build 4.3.1.201908201345.

    eclipse.buildId=4.12.0.I20190605-1800
    java.runtime.name=OpenJDK Runtime Environment
    java.runtime.version=11.0.4+11-post-Debian-1bpo91
    Command-line arguments:  -os linux -ws gtk -arch x86_64
    Sloeber, the Eclipse Arduino IDE (plugin version) 4.3.1.201908201345

@hoo2

> You can install the bundled sloeber version in a separate place

Thank you for the hint.
The reason I've never made it that way was a habit: Single instance of eclipse (over)loaded with plugins.

hasikp1, 2020-01-22:
I am also getting this error.
`eclipse.buildId=4.14.0.I20191210-0610
java.version=1.8.0_201
java.vendor=Oracle Corporation
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=en_US
Framework arguments:  -product org.eclipse.epp.package.cpp.product -product org.eclipse.epp.package.cpp.product
Command-line arguments:  -os win32 -ws win32 -arch x86_64 -product org.eclipse.epp.package.cpp.product -data file:/D:/Mobitel/eclipseArm/ -product org.eclipse.epp.package.cpp.product`

wimjongman, 2020-01-22:
Please open a new issue and attach your .log. The .log can be exported from the error log view 

This issue is closed.