Sloeber/arduino-eclipse-plugin #1097: [Question for deployment] How do I set default values for Private Hardware and Private Libraries

madhephaestus, 2019-07-12:
When Sloeber standalone ins deployed to a new computer, it generates the 2 config files in the users workspace:

eclipse-workspace/.metadata/.plugins/org.eclipse.core.runtime/.settings/io.sloeber.core.ui.prefs
eclipse-workspace/.metadata/.plugins/org.eclipse.core.runtime/.settings/io.sloeber.arduino.prefs

These files contain the default data for the location of the Arduino sketchbook and Arduino toolchain directories. On out lab machines, these values are incorrect since user sketchbooks are on a network drive. As a result, each student needs to change the values, all to the same default, causing confusion and problems. I would like to know how to modify the standalone Sloeber for deployment with custom string defaults for these values.

uzi18, 2019-07-12:
maybe you can just provide own modified copy of eclipse-workspace, before installation?

madhephaestus, 2019-07-12:
As i said, this me trying to maintain a campus full of machines with this install. I do not hav the ability in intercept each user and provide out-of-pipeline instructions. They will sit at a computer and open the programs i have installed on the machines for them. If it doesn't work with a fresh user on a fresh home dir, then the install is flatly broken. Arduino allows you to specify default paths in the install directory of the application, which is how we point Arduino at the users network file system for the sketchbook. I need sloeber's default install the be configurable so when the user loads it on a fresh install, the paths for the sketchbook are targeted at a specific directory.

uzi18, 2019-07-12:
other solution is to provide autopatcher for these files or even own installer

madhephaestus, 2019-07-12:
i am literally trying to write the installer, and am asking what files in the default install can be changed to alter the default values that Sloeber uses when it writes into these files on the event of a new workspace.

madhephaestus, 2019-07-12:
I have greped the entire directory of all files in the default install, and can not find where these default values come from in the first place. When a new workspace is created, these values get reset to defaults, where are those defaults?

jantje, 2019-07-13:
It is "hardcoded" here.
https://github.com/Sloeber/arduino-eclipse-plugin/blob/296aedb05df3478f06d2687f951a0d03c664f18c/io.sloeber.core/src/io/sloeber/core/api/Defaults.java#L33

madhephaestus, 2019-07-13:
Well this brings up another, much easier to solve, issue #1098

my initial solution to the windows paths it to put them in line with the Defaults from the Arduino itself:

https://github.com/Sloeber/arduino-eclipse-plugin/pull/1099 

Do you have a link to a place in code where you load config strings from a file, if the file exists, and use default strings if the file does not exist? I would like to further modify that section to optionally load the default values from a file.

jantje, 2019-07-13:
>Do you have a link to a place in code where you load config strings from a file,

Yes, Sloeber is mostly configuration file based. Key value pairs.
There is a utility class that reads these kind of arduino formatted files and allows you to query them.
https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.core/src/io/sloeber/core/tools/TxtFile.java
However it is probably easier to create a stream and read it.

jantje, 2019-08-08:
As the pull request s acceted and the travis build is fine now, I'm closing this issue.

closed 2019-08-08 by jantje