Sloeber/arduino-eclipse-plugin #1098: On windows, the private library path and private hardware path point to the wrong place

madhephaestus, 2019-07-13:
**Describe the bug**
The default location for Arduino to place the sketchbook is <UserHome>Documents/Arduino/

the default generated by the plugin is <UserHome>/Arduino/

**To Reproduce**
Steps to reproduce the behavior:
1. Open a standard Arduino install and install a library using the library manager
2. open Sloeber and attempt to use theat library
3. the library is not found, and Sloeber attempts to download it

**Expected behavior**
I would expect Sloeber to look for libraries and toolchains in the same default directory as Arduino by default

**Desktop (please complete the following information):**
 - Windows 10 Arduino 1.6.x and 1.8.x

**Additional context**
This is the line that is wrong:

https://github.com/Sloeber/arduino-eclipse-plugin/blob/296aedb05df3478f06d2687f951a0d03c664f18c/io.sloeber.core/src/io/sloeber/core/api/Defaults.java#L33

jantje, 2019-07-13:
Looks like they use the  "Documents" folder now on windows like on mac. probably also on Linux. So this should be a really easy fix (remove the mac test).
I guess the same applies for the hardware path.

madhephaestus, 2019-07-13:
no, not on linux, its still in $HOME on linux

PR #1099 (2019-08-08): Adding the Documents folder to the default path for Sloeber on Windows
On windows the default is for Arduino to put the sketchbook folder in the users Documents folder. 

this is to resolve #1098
closed 2019-08-08 by jantje