Sloeber/arduino-eclipse-plugin #1102: Fresh install eclipse 64 bit + sloeber plugin --> failed to created project

matthijsfh, 2019-09-01:
Fresh download of eclipse 64 bit (2019-6) and then via market place install the plugin.
Install esp8266 boards.

Create project and on "finish" I get an exception "failed to create project".

matthijsfh, 2019-09-01:
BTW running on a VMPlayer machine. Dont know it that matters.

Grts Matthijs

amirna2, 2019-09-02:
Had the same issue on a fresh install. Here is the full stack trace
```
!ENTRY io.sloeber.core.ui 4 0 2019-09-02 13:09:48.239
!MESSAGE Failed to create project
!STACK 0
java.lang.reflect.InvocationTargetException
	at org.eclipse.jface.operation.ModalContext.runInCurrentThread(ModalContext.java:449)
	at org.eclipse.jface.operation.ModalContext.run(ModalContext.java:354)
	at org.eclipse.jface.wizard.WizardDialog.run(WizardDialog.java:1035)
	at io.sloeber.ui.wizard.newsketch.NewSketchWizard.performFinish(NewSketchWizard.java:99)
	at org.eclipse.jface.wizard.WizardDialog.finishPressed(WizardDialog.java:833)
	at org.eclipse.jface.wizard.WizardDialog.buttonPressed(WizardDialog.java:472)
	at org.eclipse.jface.dialogs.Dialog.lambda$0(Dialog.java:622)
	at org.eclipse.swt.events.SelectionListener$1.widgetSelected(SelectionListener.java:84)
	at org.eclipse.swt.widgets.TypedListener.handleEvent(TypedListener.java:252)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4364)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1512)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1535)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1520)
	at org.eclipse.swt.widgets.Widget.notifyListeners(Widget.java:1324)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4151)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3768)
	at org.eclipse.jface.window.Window.runEventLoop(Window.java:823)
	at org.eclipse.jface.window.Window.open(Window.java:799)
	at org.eclipse.ui.internal.handlers.WizardHandler$New.executeHandler(WizardHandler.java:260)
	at org.eclipse.ui.internal.handlers.WizardHandler.execute(WizardHandler.java:280)
	at org.eclipse.ui.internal.handlers.HandlerProxy.execute(HandlerProxy.java:283)
	at org.eclipse.ui.internal.handlers.E4HandlerProxy.execute(E4HandlerProxy.java:95)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:58)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invokeUsingClass(InjectorImpl.java:320)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invoke(InjectorImpl.java:254)
	at org.eclipse.e4.core.contexts.ContextInjectionFactory.invoke(ContextInjectionFactory.java:173)
	at org.eclipse.e4.core.commands.internal.HandlerServiceHandler.execute(HandlerServiceHandler.java:156)
	at org.eclipse.core.commands.Command.executeWithChecks(Command.java:498)
	at org.eclipse.core.commands.ParameterizedCommand.executeWithChecks(ParameterizedCommand.java:488)
	at org.eclipse.e4.core.commands.internal.HandlerServiceImpl.executeHandler(HandlerServiceImpl.java:213)
	at org.eclipse.ui.internal.handlers.LegacyHandlerService.executeCommand(LegacyHandlerService.java:392)
	at org.eclipse.ui.internal.actions.CommandAction.runWithEvent(CommandAction.java:142)
	at org.eclipse.jface.action.ActionContributionItem.handleWidgetSelection(ActionContributionItem.java:580)
	at org.eclipse.jface.action.ActionContributionItem.lambda$4(ActionContributionItem.java:412)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4364)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1512)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1535)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1520)
	at org.eclipse.swt.widgets.Widget.notifyListeners(Widget.java:1324)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4151)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3768)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1173)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1062)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:155)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:635)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:559)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:150)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:155)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:137)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:107)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:400)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[... 69 lines cut ...]
	at org.eclipse.swt.widgets.Widget.notifyListeners(Widget.java:1324)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4151)
[... cut ...]
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:660)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:597)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1468)
```

amirna2, 2019-09-02:
It's unable to find a signature for the method invoked here:
https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.core/src/io/sloeber/core/tools/ShouldHaveBeenInCDT.java#L71

Probably the method signature has changed in the latest Eclipse CDT.

matthijsfh, 2019-09-02:
Hai,

There seems to be an answer in the other issues. Something with double regex arguments. Went back to my previous installation (32 bit eclipse) and added the 2 links to update the sloeber (stable & nightly). Now I seem to have 2.5.3 running stable.

amirna2, 2019-09-02:
I am using MacOSX. And installing the nightly build doesn't appear to fix the issue.
This is the version I have currently installed ```Sloeber, the Eclipse Arduino IDE (plugin version)	4.3.1.201902260606```

amirna2, 2019-09-02:
Tried also installing the Eclipse + plugin bundle version 4.3.1 and that failed as well

erdemuncuoglu, 2019-09-03:
**This is a duplicate.**
Same problem has been reported in many other issues and currently fixed in nightly (ref #1096).
Update sites (stable/nightly) are not maintained right now for well known reasons but thanks to @Theremingenieur that he is eager for taking the responsibility (ref #1085).

You can download latest builds from here https://oss.sonatype.org/content/repositories/snapshots/com/github/brodykenrick/arduino-eclipse-plugin/io.sloeber.product/4.3.1-SNAPSHOT/

xyybob, 2019-10-03:
When I browse to that link I get:
404 - Request is marked as local-only, remote access not allowed from M2Repository(id=snapshots)
Request is marked as local-only, remote access not allowed from M2Repository(id=snapshots)

What do I need to do?

wimjongman, 2019-10-07:
Download the repo first and then add it as a local repo.

jantje, 2019-10-07:
I made a new release but didn't do extensive testing.
So please download and test https://github.com/Sloeber/arduino-eclipse-plugin/releases/tag/V4_3_2 and report problems.

erdemuncuoglu, 2019-10-09:
Hi @jantje,

I made 3 quick tests to build and upload a sketch to an ESP8266 nodemcu board.

**Note 1:** As of testing nightly update site version was `4.3.1.201910062116` but snapshot version was `4.3.1.201910071516` so I used my private update site.
**Note 2:** There is one workaround needed to successfully upload to an ESP8266 which was discussed in #1073.
**Note 3:** Linux 64bit OS

* TEST 1
   * Fresh install v4.3.2 (nighly) (4.3.1.201910071516)
   * Setup for ESP8266 (platform v2.5.2)
   * Build - PASS
   * Upload - PASS
* TEST 2
   * Fresh install v4.3.1 (stable) (4.3.1.201810100607)
   * Setup for ESP8266 (platform v2.5.2)
   * Upgrade via private nightly repository (4.3.1.201910071516)
   * Build - PASS
   * Upload - PASS
* TEST 3
   * Already installed eclipse 4.12 (2019-09)
   * Already installed Sloeber plugin version (4.3.1.201908231618)
   * Already setup for ESP8266 (platform v2.5.2)
   * Upgrade via private nightly repository (4.3.1.201910071516)
   * Build - PASS
   * Upload - PASS

jantje, 2019-10-09:
Thanks for testing.
As to note 1. I had some issues on my build server so I had to rerun the build. As there were no git changes the update site did not get updated. So there is no issue there.
I'm surprised test 2 and 3 worked. On my system test 2 failed.
I'll change the website so new users will get the new version.

closed 2019-12-11 by jantje