Sloeber/arduino-eclipse-plugin #1103: Automatic generation of "sloeber.ino.cpp" don't start always

Alfystar, 2019-09-10:
**Describe the bug**
When i add, remove or move some function from the .ino files to any other c++ classic files, "sloeber.ino.cpp" not update immediately.
To my point of view, in fact, is like it is regenerate randomically

**To Reproduce**
Steps to reproduce the behavior:
1. Go to any project with same c++ library inside
2. move some function inside a normal library (.h and .cpp files)
3. Run the compiler
4. Very often the errors are:
"error: 'foo' was not declared in this scope"
or
"error: call of overloaded 'foo()' is ambiguous"

**Expected behavior**
The sloeber.ino.cpp have to regenerate when start the compiling any time the .ino files was modify 

**Desktop (please complete the following information):**
 - OS: linux mint 64bit 19.2
 - Version (shold add inside welcome view the version of the ide )

**Additional context**
To solve this problem i think are just enough add the manually updating of "sloeber.ino.cpp"

jantje, 2019-09-10:
>To solve this problem i think are just enough add the manually updating of "sloeber.ino.cpp"

you can delete it in the explorer.

Alfystar, 2019-09-10:
And it regenerate at the next compiling?

closed 2019-09-11 by jantje