Sloeber/arduino-eclipse-plugin #1104: create esp8266 sketch error

moyaying, 2019-09-24:
**Describe the bug**
1. Create a new Arduino sketch.
2.platform folder: esp8266/2.5.2, Board: NodeMcu 1.0, other setting...
3.finish
Then cpu 100%....

stack:
eclipse.buildId=unknown
java.version=1.8.0_221
java.vendor=Oracle Corporation
BootLoader constants: OS=macosx, ARCH=x86_64, WS=cocoa, NL=zh_CN
Framework arguments:  -perspective io.sloeber.application.perspective -keyring /Users/aipai/.eclipse_keyring
Command-line arguments:  -os macosx -ws cocoa -arch x86_64 -perspective io.sloeber.application.perspective -keyring /Users/aipai/.eclipse_keyring

io.sloeber.core.ui
Error
Tue Sep 24 14:43:10 CST 2019
Failed to create project

java.lang.reflect.InvocationTargetException
	at org.eclipse.jface.operation.ModalContext.runInCurrentThread(ModalContext.java:448)
	at org.eclipse.jface.operation.ModalContext.run(ModalContext.java:353)
	at org.eclipse.jface.wizard.WizardDialog.run(WizardDialog.java:974)
	at io.sloeber.ui.wizard.newsketch.NewSketchWizard.performFinish(NewSketchWizard.java:99)
	at org.eclipse.jface.wizard.WizardDialog.finishPressed(WizardDialog.java:772)
	at org.eclipse.jface.wizard.WizardDialog.buttonPressed(WizardDialog.java:411)
	at org.eclipse.jface.dialogs.Dialog.lambda$0(Dialog.java:619)
	at org.eclipse.swt.events.SelectionListener$1.widgetSelected(SelectionListener.java:81)
	at org.eclipse.swt.widgets.TypedListener.handleEvent(TypedListener.java:249)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:86)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4247)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1508)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1531)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1516)
	at org.eclipse.swt.widgets.Widget.notifyListeners(Widget.java:1320)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4091)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3712)
	at org.eclipse.jface.window.Window.runEventLoop(Window.java:818)
	at org.eclipse.jface.window.Window.open(Window.java:794)
	at org.eclipse.ui.internal.navigator.wizards.WizardShortcutAction.run(WizardShortcutAction.java:99)
	at org.eclipse.jface.action.Action.runWithEvent(Action.java:473)
	at org.eclipse.jface.action.ActionContributionItem.handleWidgetSelection(ActionContributionItem.java:565)
	at org.eclipse.jface.action.ActionContributionItem.lambda$4(ActionContributionItem.java:397)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:86)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4247)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1508)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1531)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1516)
	at org.eclipse.swt.widgets.Widget.notifyListeners(Widget.java:1320)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4091)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3712)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1170)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1059)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:153)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:667)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:597)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:148)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:152)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:656)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:592)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1498)
Caused by: java.lang.StackOverflowError
	at java.util.regex.Pattern$CharProperty.match(Pattern.java:3778)
	at java.util.regex.Pattern$Slice.match(Pattern.java:3974)
	at java.util.regex.Pattern$CharProperty.match(Pattern.java:3779)
	at java.util.regex.Pattern$Slice.match(Pattern.java:3974)
	at java.util.regex.Pattern$CharProperty.match(Pattern.java:3779)
	at java.util.regex.Pattern$Slice.match(Pattern.java:3974)
	at java.util.regex.Pattern$CharProperty.match(Pattern.java:3779)
	at java.util.regex.Pattern$Slice.match(Pattern.java:3974)
	at java.util.regex.Pattern$CharProperty.match(Pattern.java:3779)

**Desktop (please complete the following information):**
 - OS: [mac os]
 - Version [sloeber 4.3.1]

agendek78, 2019-09-24:
Issue exists on : esp8266/2.5.2 and : esp8266/2.5.1
Latest working SDK is : esp8266/2.5.0

erdemuncuoglu, 2019-09-25:
This is a **duplicate** of #1073, #1090 and #1102.

Solution is [here](https://github.com/Sloeber/arduino-eclipse-plugin/issues/1102#issuecomment-527656989).

closed 2019-09-25 by jantje