Sloeber/arduino-eclipse-plugin #1108: "non-existent" errors in Problems view

bergmannm, 2019-10-13:
Hi,
I've created new Arduino Sketch Project, platform AVR 1.8.1, with sample code from Adafruit PCD8544 Nokia 5110 LCD library 1.1.0 .  Then I've added Adafruit GFX Library 1.5.7 to the project. The project compiles and uploads to device correctly, even though there is 107 errors in problem view. 

Description	Resource	Path	Location	Type
Invalid arguments '
Candidates are:
long int random(long int, long int)
long int random()
long int random(long int)
'	pcdtest.ino	/testPcd	line 75	Semantic Error
Method 'width' could not be resolved	pcdtest.ino	/testPcd	line 75	Semantic Error
Method 'drawBitmap' could not be resolved	pcdtest.ino	/testPcd	line 90	Semantic Error
Method 'drawBitmap' could not be resolved	pcdtest.ino	/testPcd	line 114	Semantic Error
Method 'height' could not be resolved	pcdtest.ino	/testPcd	line 118	Semantic Error
Invalid arguments '
Candidates are:
long int random(long int, long int)
long int random()
long int random(long int)
'	pcdtest.ino	/testPcd	line 119	Semantic Error
Method 'width' could not be resolved	pcdtest.ino	/testPcd	line 119	Semantic Error

How can I get rid of these "non-existent" errors?

I'm running Sloeber 4.3.1 on Win10.

I've attached my project.
[testPcd.zip](https://github.com/Sloeber/arduino-eclipse-plugin/files/3721532/testPcd.zip)

jantje, 2019-10-13:
See http://eclipse.baeyens.it/faq.shtml#/troubleshooting ->Trouble shooting->Eclipse tells me there are bugs in my code but it compiles ok.

bergmannm, 2019-10-14:
Thanks, resolved by deleting sloeber 4.3.1.

closed 2019-10-14 by bergmannm