Sloeber/arduino-eclipse-plugin #1109: Failed to install V4 Plugin in Yakindu SCT

rmjjanssen1, 2019-10-13:
Hi,

I installed Yakindu SCT version 3.5.7 as product and added the V4 Sloeber Arduino IDE plugin via Help Add software on Mac OS Mojave. Somehow I cannot get the plugin to work properly. Configuration/installation fails to complete successfully. Some of the core bundles do not install as per screenshot below. Yakindu SCT is moved to application folder. Strange thing is that Sloeber as a product does install successful.

Can you please take a look what is wrong here?

[image]

jantje, 2019-10-13:
Did you install the sloeber product or the plugin?

rmjjanssen1, 2019-10-13:
Hi Jan,

No, I installed the Sloeber plugin in Yakindu SCT. One thing I forgot to mention. This problem occurs when using v1.8.1 of the Arduino AVR Boards. As experiment I installed the v1.6.23 and here I’m able to compile the code successfully. 

I also tried to install the Sloeber as product and Yakindu as plugin. Also here I ran into problems and ended up with not able to install the product at all due to broken package.

jantje, 2019-10-13:
The product will give problems for sure. As you have noticed.
io.sloeber.application errors are probably not important. 
Error loading provider class io.sloeber.core.... is probably more interesting to look at.
But it is clear the sloeber plugin install failed for some reason and that is causing a problem.
Note that the latest version is with eclipse 2019/09 which may be to new for Yakindu SCT.

jantje, 2020-01-18:
As this issue has been stale for a long time, I'm closing it. please reopen if you want to spend time/energy on this.

closed 2020-01-18 by jantje