Sloeber/arduino-eclipse-plugin #1110: cannot download: https://eclipse.baeyens.it/download/examples_Arduino_1_6_7.zip

wimjongman, 2019-10-24:
Just a casual user (Toddler), but noticed this on trying to install latest stable MacOS version of Beryllium.

Failed to configer Sloeber
Failed to download "https://eclipse.baeyens.it/download/examples_Arduino_1_6_7.zip".
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target

Wondering if links are in fact broken.

_Originally posted by @MarkAllread in https://github.com/Sloeber/arduino-eclipse-plugin/issues/1085#issuecomment-544270259_

MarkAllread, 2019-10-24:
Just some clarification.
The title of this issue is slightly misleading. I have no problem downloading  "https://eclipse.baeyens.it/download/examples_Arduino_1_6_7.zip".
directly from my browsers (Safari or Firefox). 

It only fails during the first launch of Beryllium 4.3.

jantje, 2019-10-24:
4.3? 4.3.1? or 4.3.2?
https://github.com/Sloeber/arduino-eclipse-plugin/releases

MarkAllread, 2019-10-24:
Installations.....
3.1 Installs correctly
4.0 Installs correctly
4.2 fails
4.3.1 fails
4.3.2 fails 

/Users/gizmo/Desktop/Fail.jpg

jantje, 2020-01-03:
I moved the zip file to github. I also changed the install conditions.
It used to be 
1) if libraries are not installed install libraries
2) if no platforms are installed install examples and arduino avr platform
now it should be
1) if libraries are not installed install libraries
2) if examples  are not installed install examples
3) if no platforms are installed install arduino avr platform

This should make the installation of failed examples possible after upgrade

jantje, 2020-01-04:
@MarkAllread 
can you try with the nightly? It should download newer examples from github instead of baeyens.it.

MarkAllread, 2020-01-04:
@jantje
I'd be happy to but I need a pointer to them.
I used to know where your nightlys were jantje....but I can't find them now.
if I try to go from https://eclipse.baeyens.it/nightly.php
I get this :
404 - Path /com/github/brodykenrick/arduino-eclipse-plugin/io.sloeber.product/4.3.2-SNAPSHOT/ not found in local storage of repository "Snapshots" [id=snapshots]

jantje, 2020-01-04:
The link is available on the eclipse.baeyens.it site
http://eclipse.baeyens.it/nightly.php
http://eclipse.baeyens.it/update/V4/nightly

MarkAllread, 2020-01-04:
@jantje
That's where I thought it was and I tried that too but no luck...
If I go here:
http://eclipse.baeyens.it/update/V4/nightly
I get this:
"If you ended up here you took a shortcut and failed. Go back to start. Start is here installAdvice.shtml"

Tried both Safari and Firefox several times.

jantje, 2020-01-04:
You should use this url in eclipse as update site

MarkAllread, 2020-01-04:
I thought that might be the case.
OK, did a nightly update through my (incomplete) installed V4.3.2
The ONLY nightly update available to me was Sloeber 4.3.1.202001040507 (hmmm.....???)
After installing update and restarting, still got a failed configuration. 
It's still going to baeyens with this nightly. 
Shouldn't the nightly have been a V4?
[image]

[image]

MarkAllread, 2020-01-04:
Sorry, last image was incorrect.
[image]

jantje, 2020-01-04:
I messed up the version numbers a bit...
V4.3.1.202001040507 is the latest nightly
It should have said V4.3.2.202001040507 but I forgot to change the version numbers in the plugins to V4.3.2 so V4.3.2 plugins are versioned as V4.3.1 ...
The easiest way to see the latestness of a build is to look at the last part which is the build nr
In this case the build nr is 202001040507 which means build in the year 2020 month 01 day 04 hour 05 minute 07 so that is really new.

MarkAllread, 2020-01-04:
Understood.
Anyway, it did install it and restarted.
Had same issue as first screenshot above shows.
Looks like it's still trying to get it from Baeyens and not Github

jantje, 2020-01-04:
Strange. I tested here and Sloeber downloaded the github stuff.
Do you see the menu item "show known issues"
[image]

MarkAllread, 2020-01-04:
My menu is the same except that I do not have that item.
I'll try a clean install on another boot volume a bit later today and let you know what happens.

jantje, 2020-01-04:
Not having the item means you do not have the nightly. Which also means the zip file comes from baeyens.it and not github.

MarkAllread, 2020-01-04:
Been trying for a while now.
perhaps it is offline.
[image]

jantje, 2020-01-05:
Works here. I tried both http and https. As there are serious security issues with the latest mac version it looks like a good idea to me to try https.
[image]

jantje, 2020-01-05:
FYI I have a "service available tester" running and he didn't report a site down.

MarkAllread, 2020-01-05:
Hi @jantje

I pinged it last night and saw that it was responding
So I don’t know why this made any difference but I used the 4.3.2 release from Github. I was previously using the one from eclipe.baeyens. It was probably just a coincidence but it worked for me this morning. It could also have been my network monitor interfering

I was able to see and install the nightly. (I selected the product)

Everything looked good as the process continued from here but there was a message that said “Your original request has been modified. “Sloeber is already installed, so an update will be performed instead"

Once the download and installation was complete I was asked to restart Sloeber..which I did.
Again the error message appeared as it was trying to get the examples from baeyens.

I then reviewed the updates available and it is telling me that all (Sloeber product) updates have been installed but I still do not have the “show known issues” menu item.

So the install/update must be failing.

Not sure what else I can do. I keep getting the same results so I must be doing something wrong.

I can post screen shots if you want them.

MarkAllread, 2020-01-05:
Perhaps you could send me a direct link  to the nightly?

jantje, 2020-01-05:
@MarkAllread 
It is hard for me to follow your story.
>So I don’t know why this made any difference but I used the 4.3.2 release from Github. I was previously using the one from eclipe.baeyens.

There is no difference between the two. Baeyens.it points to the github download.
[image]

>Everything looked good as the process continued from here but there was a message that said “Your original request has been modified. “Sloeber is already installed, so an update will be performed instead"

Which version did you upgrading to at that time? At the time of writing it should be V4.3.1.202001040507

>So the install/update must be failing.

I agree. I see 3 possible reasons. 
The first is: Eclipse refusing because a missing requirement. 
The second is because there is no pointer to the correct update site. 
The third is intermittent(?) problems with the network to the update site.

>Perhaps you could send me a direct link to the nightly?

I could,... but it would be a hassle and it should work from the update site.
If you have git and maven on your system the easiest is to compile the nightly yourself

MarkAllread, 2020-01-05:
**In simple terms, I was able to initiate the update:**
[image]
**When the update starts I get this message:**
[image]
**Then a request to restart after the update had finished**
[image]
**Then after the restart, I get the usual problem**
[image]
**Then after checking updates, it looks like the 4.3.1 update has been installed**
[image]

All this is using Mac OSX Sierra. I get the same results every time. (Just ignore my inability to connect yesterday, I think that was a problem at my end).
So I do connect, an update appears to happen but the results aren't what is expected.

If a link is a hassle, no worry, please don't go to the trouble. I don't have the tools you mentioned so I can't compile.
Thanks for your patience with me, but it does seem like I will have to stick with Toddler for now (which is working just fine on Sierra)
I know I will have problems with Toddler when I move up to High Sierra (I tried and it crashed randomly). I was unable to install Berrylium on High Sierra.
Happy New Year @jantje

jantje, 2020-01-05:
This is really weird... I'm bafflegasted .... need to think

MarkAllread, 2020-01-05:
BAFFLEGASTED HAHA!
Never heard that one before...excuse me if I borrow it. ;-)

Anyway, I'm setting up a complete new OS installation. Lots of work to get everything together...java, make files, serial driver and so on.
It's behaving a bit different, I'll let you know what happens with V4.

wimjongman, 2020-01-07:
There is a difference between installing a new version and the initial download of boards and stuff.

MarkAllread, 2020-01-07:
Hi @jantje
So I created a clean brand new Mac Sierra boot drive with fresh installs of Java 8, command line tools, serial driver, etc etc and all the things I needed to install and run Sloeber (including the Arduino IDE which I don’t think I need for V4 anyway)
I always use a new unzipped version for my trials so I unzipped V4.3.2 and ran it.
It started the download process and this time it passed the Arduino examples download without error. However the process did not complete and I got the dreaded “Failed to configure Sloeber” problem.
[image]

When I try to build, I get a “make” error because:

...../Applications/Sloeber.app/Contents/Eclipse//arduinoPlugin/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++: No such file or directory
make: *** [sloeber.ino.cpp.o] Error 127

I recall someone else having this issue but i do not know how to fix it.
I checked my board selection under prefs...Platforms and Boards and I've tried 1.8.2, 1.8.1 and 1.6.20
I checked that all the json files were accessible under preferences>Third party index URLs and I could reach them all. I even did the update to Sloeber.

I think I’ve exhausted myself now. Took me hours to set this up and while it appears that I am closer than before, I am still unable to use V4.

jantje, 2020-01-07:
The good news is that we can close the examples download issue :-) (his issue)
You say arduino avr 1.6.20 does not work. According to #1111 it does.
In one of the gifs at least 1.6.2 seems to be installed correctly.
Lets continue the arduino avr 1.8.x mac issues in #1111

closed 2020-01-07 by jantje