Sloeber/arduino-eclipse-plugin #1111: Mac Os Mojave - Java8 Install V431 headache and solution...

github4manu, 2019-11-15:
Just want to share my findings to finally successfully installed Sloeber.app V431 on a Mac Os Mojave - Java8 Install V431 headache and solution... 

Problems/Solutions:
**problem1:** 
Java certificate problem : Failed to download "https://eclipse.baeyens.it/download/examples_Arduino_1_6_7.zip".
**solution1:** 
had to manually add the https - eclipse_baeyens_it certificate to java
sudo keytool -import -alias eclipse_baeyens_it -keystore /Library/Java/JavaVirtualMachines/jdk-13.0.1.jdk/Contents/Home/lib/security/cacerts -file /Users/????/Documents/eclipse_baeyens_it.cer

**problem2:** 
error message: 
arduinoOTA ... 
Failed to configer Sloeber
Failed to install Arduino AVR Boards
No valid system found for "avr-gcc".
No valid system found for "avrdude".
**solution2:** 
install arduinno.app 1.8.1
open folder :
/Users/????/Library/Arduino15/packages
open package content sloeber.app : /Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/packages

copy /Users/manu/Library/Arduino15/packages/arduino directory into /Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/packages

jantje, 2019-11-16:
The first is a duplicate of #1110 
Problem2 ... you need to install the boards in Sloeber. Your solution will cause problems.

github4manu, 2019-11-16:
Hello,
As for problem 2 ("fix"), i don't have any problems up to know, all is working well, even adding other non-arduino boards/platforms!
Removing / Adding arduino board 181 via  preferences/arduino/...board-manager give me the same error (see image attatchments)
[image]

Clean setup error log (
[sloeberV431-macos-mojave.log](https://github.com/Sloeber/arduino-eclipse-plugin/files/3854046/sloeberV431-macos-mojave.log)

github4manu, 2019-11-16:
Further investigation.
I can add via the board manager Arduino 1.6.xxx, but 1.8.1 fails to "unpack" the avrdude and avr-gcc
[image]

jantje, 2019-11-16:
To me it looks like something in version 1.8.1 is wrong or a new feature (not yet supported by Sloeber) is being used.
If you search in the file [sloeber install]/arduinoplugin/package_index.json (which is downloaded from arduino.cc) for 1.8.1 you will see arduino avr version 1.8.1 is dependent on 
```
          "toolsDependencies": [
            {
              "packager": "arduino",
              "name": "avr-gcc",
              "version": "7.3.0-atmel3.6.1-arduino5"
            },
            {
              "packager": "arduino",
              "name": "avrdude",
              "version": "6.3.0-arduino17"
            },
            {
              "packager": "arduino",
              "name": "arduinoOTA",
              "version": "1.3.0"
            }
          ]
```
Based on this info Sloeber tries to install arduinoOTA version 1.3.0 (succeeds), avrdude version 7.3.0-atmel3.6.1-arduino5 (not found for your system) avr-gcc version 7.3.0-atmel3.6.1-arduino5 (not found for your system).
With not found for your system I mean that sloeber searches in this file for the download info for your system and fails to find a proper candidate. It does find what it needs for my system.

In the tools section you will find the arduinoOTA version 1.3.0 and it looks like
```
        {
          "name": "arduinoOTA",
          "version": "1.3.0",
          "systems": [
...
              "host": "i386-apple-darwin11",
              "archiveFileName": "arduinoOTA-1.3.0-darwin_amd64.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/arduinoOTA-1.3.0-darwin_amd64.tar.bz2"
...
```
Note the   "host": "i386-apple-darwin11", I assume sloeber will download this and you can confirm this by checking for the file [sloeberinstall]/arduinoplugin/downloads/arduinoOTA-1.3.0-darwin_amd64.tar.bz2
If you look for the avr dude version you will see
```
        {
          "name": "avrdude",
          "version": "6.3.0-arduino17",
          "systems": [
...
              "host": "x86_64-apple-darwin12",
              "archiveFileName": "avrdude-6.3.0-arduino17-x86_64-apple-darwin12.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino17-x86_64-apple-darwin12.tar.bz2"
            },
...
          ]
        },
```
Here you see the host has a different number "host": "x86_64-apple-darwin12", I assume that Sloeber fails to see this as the package to install.

```
          "name": "avr-gcc",
          "version": "7.3.0-atmel3.6.1-arduino5",
          "systems": [
            {
...
              "host": "x86_64-apple-darwin14",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-arduino5-x86_64-apple-darwin14.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino5-x86_64-apple-darwin14.tar.bz2"

```
Now the only apple host is "host": "x86_64-apple-darwin14", again I assume Sloeber does not find a "good install candidate" on your system
Note that the megaavr boards have used "avr-gcc"   "version": "7.3.0-atmel3.6.1-arduino5" in previous releases.
Can you install 1.8.1 in arduino ide correctly?

github4manu, 2019-11-16:
Hi,
Thank you for your time !!! I don't get all the details you wrote..., but yes,  i installed Arduino 1.8.1.
Thanks again for your help!

I will include these files of the arduino181 install directory (MacOS 
[image])

/Applications/Arduino181.app/Contents/Java/hardware/package_index_bundled.json
/Applications/Arduino181.app/Contents/Java/hardware/platform.txt
/Applications/Arduino181.app/Contents/Java/hardware/platform.keys.rewrite.txt
[platform.keys.rewrite.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/3854557/platform.keys.rewrite.txt)
[platform.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/3854558/platform.txt)
[image]
[image]

github4manu, 2019-11-16:
the json file
[package_esp8266com_index.json.zip](https://github.com/Sloeber/arduino-eclipse-plugin/files/3854572/package_esp8266com_index.json.zip)

github4manu, 2019-11-23:
ok, tried all kind of stuff, finally ended up to do the installation of the 1.8.1 arduino board manually.
dependencies that not get installed correctly
          _"toolsDependencies": [
            {
              "packager": "arduino",
              "name": "avr-gcc",
              "version": "7.3.0-atmel3.6.1-arduino5"
            },
            {
              "packager": "arduino",
              "name": "avrdude",
              "version": "6.3.0-arduino17"
            },_

downloaded the dependencies, extracted, created the directories, moved the extracted contends ...

[image]

jantje, 2019-11-23:
That is basically what the installation does. So that should do the trick.

jschw, 2019-12-29:
I ran into the same problem and can't use Sloeber after updating to Catalina... :(
After I followed github4manu's suggestions, I get this error when trying to compile the project: clang: error: unknown argument: '-mno-skip-bug'
(Currently running MacOS 10.15.2)

Any idea how to fix this? 

Catalina is just a f***ing mess, never had so many problems in the past 10 years after any OS X major update...

jantje, 2020-01-07:
The current nightly fixes #1110 which means problem 1 can be dealt with by upgrading to the nightly.

jantje, 2020-01-07:
I think the problem is here: https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.core/src/io/sloeber/core/managers/ToolSystem.java#L52
Because I have no clue what the logic should be I'll add all the options used here. Though I'm sure that will cause grievances elsewhere.

jantje, 2020-01-07:
Nightly should contain this fix tomorrow CET

jantje, 2020-01-08:
As I see this as is annoying issue for mac users and the workaround (receiving error messages and run update; during installation) is annoying as well. I would like to make a new release for the latest mac changes.
Before I do so a test of the changes and "a all works well" is needed to avoid confusion and extra work.

Apart from "other not yet reported issues" and "things overlooked by me" I'm mostly concerned with the fact that baeyens.it is not a trusted site and as such the update may fail.
This could be the explanation of the behaviour that @MarkAllread described that bafflegasted me https://github.com/Sloeber/arduino-eclipse-plugin/issues/1110#issuecomment-570938899

If so problem1 (from the first post) still exists and solution 1 (from the first post) is still needed.

mac users: your input is required.

MarkAllread, 2020-01-08:
@Jantje
I'll start it off by adding a little more info on my trials yesterday. (which I did several times). Here is a simplified bulleted list.

Mac Pro Sierra 10.12..6
Clean install of Sierra on new boot volume.

- Clean install of V4.3.2
- Did NOT do the update to 4.3.1.202001080506
- Ran Sloeber and it configured without the issue of downloading 	"...examples_Arduino_1_6_7.zip"
- Gave the issue "Failed to install Arduino AVR boards” during configuration.
- Tried a simple “Blink” sketch.
- Failed to build due to 
	“..avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++: No such file or directory”

- I performed the update to 4.3.1.202001080506
- After restart, checked that 4.3.1 had been installed and it HAD.
- Still gave "Failed to install Arduino AVR boards issue"
- Still did NOT have the menu item "Show known issues" that you mentioned yesterday
- Still failed to build due to 
	“..avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++: No such file or directory”

What I also find strange is that Eclipse fails to remember the link to the 4.3.1 update between restarts of Sloeber. I have to manually enter the link again. But I did confirm that it was installed using the “what’s installed” link.

If there is any specific info you think I can provide, please ask. I will leave the clean volume intact for now.

jantje, 2020-01-08:
[io.sloeber.updatesite-4.3.1-SNAPSHOT.zip](https://github.com/Sloeber/arduino-eclipse-plugin/files/4036966/io.sloeber.updatesite-4.3.1-SNAPSHOT.zip)
Can you download this zip file extract and use this as update site?
I'm pretty sure for one reason or another the Sloeber update simply didn't work

mjmeijer, 2020-01-08:
Can do a test build on Mojave, not yet Catalina, later.
Is Sloeber.io a registered mac developer, able to sign executables?
That maybe a problem on Mojave as well.

MarkAllread, 2020-01-08:
Did clean install of 4.3.2
Gave the same issue "Failed to install Arduino AVR boards” during configuration.
Installed update as requested.

[image]

Confirmed update

[image]

No new menu item.

[image]

Build of "Blink" still failed 
    "...arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++: No such file or directory"

Everything seemed to go exactly the same as before.

MarkAllread, 2020-01-08:
Just a quick fyi
I previously (probably about a year ago) have had V4 earlier versions installed and running on the same Mac. Can't recall if it was Sierra, but it was at least El Capitan 10.11.6
I have also tried to install 4.3.2 on High Sierra 10.13.6 with the same problems I am experiencing now.

jantje, 2020-01-08:
I think you did not upgrade the plugin correctly :-s
[image]
Selecting the Sloeber IDE is more important than the product!!!
FI on my system you see both are new.
[image]
Check the plugin section all  sloeber.io plugins must have the same version number and date
[image]

jantje, 2020-01-08:
@mjmeijer welcome back

As there still is no money in opensource and I'm still living from welvare there is no money to sign anything. 
We even may have lost the sloeber.io domain. #1085

MarkAllread, 2020-01-08:
>Selecting the Sloeber IDE is more important than the product!!!

Geesh!!!!!
Didn't do that because its says "plugin version" and I'm using the product.

Anyway, I tried it with your download link and I now have the new menu item. 

I think we have a winner!

This time there was a definite longer download of boards.
I tried both 1.6.20 and 1.8.2 and they both worked.

So the secret is to download both the "product" AND the "plugin version"

Seems like you've fixed it with the nightly 👍 -)

One minor thing left.
Every launch of Sloeber end with this...
[image]
It disappears after a few seconds.

jantje, 2020-01-09:
The product is just a bundle containing the plugin.
I don't know if it is my lack of understanding of the relation between product application and plugin or it is a feature but indeed one would expect a update of a bundle to update the referenced content but that never worked.
The "finishing the installation" is the check to see stuff is installed. On my system it just flashes. I guess it is dependent on the system load and disk read speed.
But I am happy you got things going.

MarkAllread, 2020-01-09:
>But I am happy you got things going.

....and thank you very much @jantje for all your effort.
I also went back to my original working volume and installed the "plugin" update using your nightly and it's all working fine there too.

closed 2020-04-21 by jantje