Sloeber/arduino-eclipse-plugin #1112: v4.3.1 "Product" Won't Install on Win 7 Machine

gfvalvo, 2019-11-15:
Hi. I want to do a full install of v4.3.1 Product on Win 7 machine. I downloaded and extracted the package into its own directory and clicked on 'sloeber-ide.exe'. I also pointed it to a new Workspace directory.

The install seemed to go well. It downloaded and extracted a bunch of stuff. But, now it just sits there at the opening "Welcome" page (picture below). Does nothing else. Left it over night -- nothing.

I did not have this problem installing v4.3. How do I fix it?

Thanks.
[image]

erdemuncuoglu, 2019-11-15:
When you close that "Welcome" screen every thing should be as usual. It won't disappear as you expected. It is related with Eclipse not Sloeber. It also happens whenever you install a new add-on (plugin) into the Eclipse.

gfvalvo, 2019-11-15:
Nope. When I close the Welcome screen the whole Sloeber application exits. If I restart it, the Welcome screen re-appears and just sits there. Does nothing. Any other suggestions?

jantje, 2019-11-16:
I just tried 4.3.2 in windows and I get
[image]
clicking the yellow marked X works like @erdemuncuoglu described.

gfvalvo, 2019-11-16:
Still no joy. This time I deleted both the install and the workspace folders. Then re-extracted the compressed file and clicked sloeber-ide.exe. Just to make things fun, it gave me an error message this time. I've attached a screen shot.

It really shouldn't be this difficult.

Is there anything corrupted from previously, failed attempts that I should clean out? As mentioned, I deleted the folders. Is there something in the Register I should delete to make a "fresh start"?

[image]

gfvalvo, 2019-11-16:
By the way, the install instructions are a little bit vague. Does Step #4 mean click on "eclipsec.exe" or "sloeber-ide.exe"?

[image]

jantje, 2019-11-16:
you should start sloeber-ide.exe
It seems like the download of arduino avr boards failed on your system. Did you allow java internet access?
Anyway sloeber is properly installed you should be able to use the boards manager to reinstall the arduino avr boards or the ones you want.

gfvalvo, 2019-11-16:
OK, tried one more time. Clean download, clean directories ........ Success!!!
Random glitch perhaps, or something dumb on my part.

Anyway, thanks guys.

closed 2019-11-16 by gfvalvo