Sloeber/arduino-eclipse-plugin #1113: ESP8266 upload fails - esptool: error: argument operation: invalid choice

Gord1, 2019-11-21:
This doesn't seem to be fixed yet in the nightly build. Ref: https://github.com/Sloeber/arduino-eclipse-plugin/issues/1073#issuecomment-520211697

Starting upload
Uploading project "xcvvxc" with "ESPTOOL"
Starting reset using DTR toggle process
Continuing to use "/dev/ttyUSB0"
Ending reset

Launching: /home/gordon/Downloads/Sloeber//arduinoPlugin/packages/esp8266/tools/python3/3.7.2-post1/python3 /home/gordon/Downloads/Sloeber//arduinoPlugin/packages/esp8266/hardware/esp8266/2.6.0/tools/upload.py --chip esp8266 --port /dev/ttyUSB0 --baud 921600  erase_flash --end --chip esp8266 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset  write_flash 0x0 /home/gordon/Documents/sloeber-workspaceJunk/xcvvxc/Release/xcvvxc.bin --end 

Output:
usage: esptool [-h] [--chip {auto,esp8266,esp32}] [--port PORT] [--baud BAUD]
               [--before {default_reset,no_reset,no_reset_no_sync}]
               [--after {hard_reset,soft_reset,no_reset}] [--no-stub]
               [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,version}
               ...
esptool: error: argument operation: invalid choice: '' (choose from 'load_ram', 'dump_mem', 'read_mem', 'write_mem', 'write_flash', 'run', 'image_info', 'make_image', 'elf2image', 'read_mac', 'chip_id', 'flash_id', 'read_flash_status', 'write_flash_status', 'read_flash', 'verify_flash', 'erase_flash', 'erase_region', 'version')
The execution of command "python3" is done.

Was using the latest stable Sloeber version and had this problem and moved to current Nightly version though update   Sloeber	4.3.1.201911160506	

Gord_W

Gord1, 2019-11-21:
> Was using the latest stable Sloeber version and had this problem and moved to current Nightly version though update Sloeber 4.3.1.201911160506

To test if it was the automatic update contributed to the problem I downloaded the latest nightly from
https://oss.sonatype.org/content/repositories/snapshots/com/github/brodykenrick/arduino-eclipse-plugin/io.sloeber.product/4.3.1-SNAPSHOT/io.sloeber.product-4.3.1-20191116.013607-92-linux.gtk.x86_64.tar.gz

and tried it out to see if it has the same problem.

The download version does NOT have the same problem, so I consider this issue closed.  

Anyone doing an automatic update should be advised that downloading a new copy might be the better way to go.
Thanks,
Gord_W

closed 2019-11-21 by Gord1

erdemuncuoglu, 2019-11-22:
Hi @Gord1 

In fact this is a known issue and hasn't been fixed yet so a workaround should be applied anyway.

> Anyone doing an automatic update should be advised that downloading a new copy might be the better way to go.

An automatic update test which was reported in [this comment](https://github.com/Sloeber/arduino-eclipse-plugin/issues/1102#issuecomment-540076246) as "TEST 2" went successful. Any detailed feedback is much appreciated regarding your update issue.

Thanks,

Gord1, 2019-11-22:
Hi @erdemuncuoglu 
> Any detailed feedback is much appreciated regarding your update issue.

I was using V4.3.1_linux64.2018-10-10_08-21-58.tar.gz, the latest stable version and ran into the above problem with the ESP8266 upload.  To see if there was a fix in the nightly version I used the automatic update feature in Sloeber/Eclipse. 

-Window>Preference>Update>Available Software Sites and deselected Sloeber Stable and selected Nightly.  
-I also selected Automatic Update. 
-Restarted it and let it do it's thing. Eclipse stuff and Sloeber stuff downloaded and installed.
-I probably restarted Sloeber
-Tested ESP8266 upload and had same problem

That is when I wrote the bug report.  As a further test I download the nightly from the above site and found that it worked.
Gord_W

erdemuncuoglu, 2019-11-22:
Did you ever applied any of the workarounds until it worked?

Gord1, 2019-11-22:
io.sloeber.product-4.3.1-20191116.013607-92-linux.gtk.x86_64.tar.gz  worked out of the box.

ESP8266 2.6.1

erdemuncuoglu, 2019-11-24:
Hi @jantje,
As I roughly inspected that `"...ESP8266 upload fails..."` was fixed in platform 2.6.1 (or maybe in 2.6.0). When @Gord1 had that problem he was probably using 2.5.2 or earlier. With fresh install he automatically got 2.6.1
We may close all related issued mentioning that it was solved with newer version of that platform.

jantje, 2019-11-24:
Sounds like a good idea. I still need to update the site so the latest product points to 4.3.2