Sloeber/arduino-eclipse-plugin #1114: Plugin doesn't follow HTTP redirections when download compiler tools

wmarkow, 2019-12-07:
**Describe the bug**
I have prepared my custom [package_wmarkow_index.json](https://raw.githubusercontent.com/wmarkow/arduino-sandbox/master/package_wmarkow_index.json). It contains two custom **avr-1.8.2** platforms which uses two different compiler tools:
 * **avr-gcc-8.3.0-x64-mingw** hosted as https://blog.zakkemble.net/download/avr-gcc-8.3.0-x64-mingw.zip
 * **avr-gcc-9.1.0-x64-mingw** hosted as https://blog.zakkemble.net/download/avr-gcc-9.1.0-x64-mingw.zip

When I want to use one of these platforms in Sloeber, then compiler is not downloaded correctly, because the downloader doesn't follow the HTTP redirections to reach the destination.

**To Reproduce**
Take a look at the GIF:
[image]

**Expected behavior**
The tools should be downloaded.
As a workaround I downloaded them manually and copied into **arduinoPlugin\downloads** directory.

**Screenshots**
Take a look at the attached GIF above.

**Desktop (please complete the following information):**
 - OS: Windows 7
 - Browser Firefox 71.0

wimjongman, 2019-12-08:
are you able to create a patch for this issue?

jantje, 2019-12-08:
I'm pretty sure redirections have been implemented for something. Not sure if it was only for json or for all downloads.
Not sure why you want to use redirections for the zip files though ....

wmarkow, 2019-12-09:
@wimjongman theoretcally and practically I'm able to create a patch for this issue, but I'm not giving any promisses. Please wait a bit.

@jantje I'm not the owner of this webpage. I just found that the guy hosts there the avr-gcc dostributions so I just used them in my own package json. This webpage uses Wordpress under the hood and redirects links to the download folder.

wimjongman, 2019-12-09:
@wmarkow cool. Thanks!

wmarkow, 2019-12-09:
I had some time to take a look at this. Just put a breakpoint in
**io.sloeber.core.managers.InternalPackageManager.downloadAndInstall(String, String, IPath, boolean, IProgressMonitor)**
and execute the test scenario. The breakpoint will hit twice but continue until it wants to download 
**https://blog.zakkemble.net/download/avr-gcc-8.3.0-x64-mingw.zip**
Now debug what is going on.

@jantje indeed the method **io.sloeber.core.api.PackageManager.myCopy(URL, File, boolean)** handles the redirections - but only once. In my case I had a multi-redirection (like a chain), like below:

**https://blog.zakkemble.net/download/avr-gcc-8.3.0-x64-mingw.zip** redirects with **302** code to:
**http://blog.zakkemble.net/wp-content/uploads/downloads/2019/02/avr-gcc-8.3.0-x64-mingw.zip** redirects with **301** to:
**https://blog.zakkemble.net/wp-content/uploads/downloads/2019/02/avr-gcc-8.3.0-x64-mingw.zip** which can be finally downloaded with code **200**

Proposed fix: when HTTP status is **3xx** then call **io.sloeber.core.api.PackageManager.myCopy(URL, File, boolean)** recursively with a redirect URL.

Possible side effect: what if url **A** redirects to url **B** and url **B** redirects to url **A**? We will have an endless loop and probably Sloeber will crash with out of memory error. Is that even possible?

jantje, 2019-12-09:
>Possible side effect: what if url A redirects to url B and url B redirects to url A? We will have an endless loop and probably Sloeber will crash with out of memory error. Is that even possible?

that will happen for sure

I would prefer to point to https://blog.zakkemble.net/wp-content/uploads/downloads/2019/02/avr-gcc-8.3.0-x64-mingw.zip  in the json file instead of https://blog.zakkemble.net/download/avr-gcc-8.3.0-x64-mingw.zip 
No fix needed and will survive one new redirect

erdemuncuoglu, 2019-12-10:
> Possible side effect: what if url A redirects to url B and url B redirects to url A? We will have an endless loop and probably Sloeber will crash with out of memory error. Is that even possible?

Web browsers overcome this by limiting loop count (by 10 redirects for example).

jantje, 2019-12-10:
If all goes well the nightly should contain this fix tomorrow.
Thanks

wmarkow, 2019-12-13:
> I would prefer to point to https://blog.zakkemble.net/wp-content/uploads/downloads/2019/02/avr-gcc-8.3.0-x64-mingw.zip in the json file instead of https://blog.zakkemble.net/download/avr-gcc-8.3.0-x64-mingw.zip
> No fix needed and will survive one new redirect

I'm sorry @jantje , I missed your replay. That make also sense but I see that you have already accepted the PR. Do we make a loop counter, so after - let's say 5 redirections - it will fail? With a small refactoring of this private **io.sloeber.core.api.PackageManager.myCopy(URL, File, boolean)** method we could achieve this. But what to do when we hit the counter? Silently fail without any error and the file will not be downloaded or just throw some exception? I do not know. Please advice.

wimjongman, 2019-12-13:
You mean a redirect counter. 

You might want to save the redirects URLS in a List and if the URL was encountered before you throw an exception.

jantje, 2019-12-14:
@wmarkow I accepted the change as you simply used the wrapper method I made to fix redirection used for json files. That should be pretty harmless.

wmarkow, 2019-12-31:
@jantje I have just commited a small improvement in the case: a redirection counter, default set to 5. When more redirection met then IOException is thrown.

I have retested this with my test scenario and set in code the MAX_HTTP_REDIRECTINOS to 1. Then I had the following error message in Sloeber:
```
Failed to install Arduino AVR Boards GCC 8.3.0
Failed to download "https://blog.zakkemble.net/download/avr-gcc-8.3.0-x64-mingw.zip".
Too many redirections while downloading file.
```

Please take a look in some free time.

wimjongman, 2020-01-01:
You have to make a pull request

wmarkow, 2020-01-02:
Pull requests have been merged: plugin follows the multiple redirects, but no more than 5 of them.

I'm closing this issue.

closed 2020-01-02 by wmarkow