Sloeber/arduino-eclipse-plugin #1115: At some specific case a wrong compiler version is used

wmarkow, 2019-12-09:
**Describe the bug**
If multiple platforms are selected in the **Platforms and Boards** preferences dialog, then - at some specific conditions - a compiler from wrong platform is used to compile the code. 
We have for example two platforms:
 * an **avr** platform **1.6.23** has a compiler in version **5.4.0-atmel3.6.1-arduino2**
 * an **avr** platform **1.8.2** has a compiler in version **7.3.0-atmel3.6.1-arduino5**

Now go to the next section of this bug report to see how to reproduce.

**To Reproduce**
Steps to reproduce the behavior:
1. Compile the project for platform **1.6.23**. Arduico core **1.6.23** sources will be used (which is correct). **5.4.0-atmel3.6.1-arduino2** will be used (which is correct).
2. Compile the project for platform **1.8.2**. Arduico core **1.8.2** sources will be used (which is correct). **7.3.0-atmel3.6.1-arduino** will be used (which is correct).
3. Compile again the project for platform **1.6.23**. Arduico core **1.6.23** sources will be used (which is correct). **7.3.0-atmel3.6.1-arduino** will be used (which is wrong).

**Expected behavior**
The correct compiler from the platform definition should be always used.

**Screenshots**
See the gif below how to reproduce:
[image]

**Desktop (please complete the following information):**
 - OS: Windows 7
 - Browser Firefox 71.0
 - Sloeber V4.3.1_win64.2018-10-10_08-21-58

jantje, 2019-12-10:
Actually this is compliant with the arduino ide behaviour. Originally Sloeber used the versions as specified in the json file. But that meant some stuff didn't work because json files were not updated and assumed newer avr compilers.
I had a discussion about this with Arduino people and to my surprise this was the "wanted" behaviour.
To accommodate on how I think it should be versus how arduino thinks it should be I created the option arduino->preferences->arduino->"Chose the toolchain like the arduino ide" (Freely translated from dutch)
This option is default on.
Turn it off to force the versions as specified in the json files.
Note that you have to select the preferences of the project and apply to take into account this change.
Also not that for this functionality (compiling for multiple versions) you are probably better of using multiple configurations.

wmarkow, 2019-12-11:
Thanks @jantje for explanations. It works when I turned off this option.

I close this issue then.

closed 2019-12-11 by wmarkow