Sloeber/arduino-eclipse-plugin #1117: Cannot create sketches with 2.5.2

rapasal, 2019-12-11:
**Describe the bug**
When creating a new sketch with **Sloeber 4.3.1.201810101903** (64 bit product bundle from http://eclipse.baeyens.it/stable.php?OS=Windows) and **ESP8266 2.5.2** on **Windows 10** I get `java.lang.InvocationTargetException` error.

If I try to change an existing 2.4.2 sketch to use 2.5.2 I get `java.lang.StackOverflowError`. When using 2.4.2 it is fine.

**To Reproduce**
Create a new sketch using ESP8266 2.5.2.

**Expected behavior**
Project created, or ESP core version updated.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser N/A
 - Version N/A

erdemuncuoglu, 2019-12-11:
Hi @rapasal,
`java.lang.InvocationTargetException` This is a known issue (#1102) and fixed in Sloeber nightly (which is the preferred version nowadays). On the other hand there is an other issue (#1073) with 2.5.2 which is fixed in 2.6.1.

`java.lang.StackOverflowError` This may also disappear with latest nightly.

jantje, 2019-12-11:
Version 3.4.2 is now officially released.

closed 2019-12-11 by jantje

rapasal, 2019-12-11:
Thanks! 2.6.2 works a treat in Sloeber 3.4.1.