Sloeber/arduino-eclipse-plugin #1122: Compilation fails

larsskj, 2019-12-30:
I have written the following code as part of a project:

`struct command {
	char code;
	int size;
	int (* callback)(char *inbuf);
	command *next { nullptr };
};

command *commands { nullptr };

command * getCommand(char code) {
	static command *current = commands;

	while (current->code != code && current->next) {
		current = current->next;
	}

	if (current->code == code) {
		return current;
	} else {
		return nullptr;
	}
}
`

Whilst the code should be valid - it is as it compiles and runs using other tools - it won't compile using Sloeber.

Sloeber adds forward declarations to the file _sloeber.ino.cpp_ but only for the functions: The struct is missing from _sloeber.ino.cpp_ and the compiler stops with an error as the function cannot return an unknown type.

Adding the struct to the top op the list in _sloever.ino.cpp_ makes the project compile, but the struct is removed again whenever I modify the code.

jantje, 2019-12-30:
I guess this is a duplicate of #775 
Workaround: put the struct in a header file and include that in your ino file

larsskj, 2019-12-30:
Yes, this looks as a duplicate - sorry for not searching closed bugs.

jantje, 2019-12-30:
> 
> 
> Yes, this looks as a duplicate - sorry for not searching closed bugs.

No problem. It is logical to search only open issues. However it is annoying to keep "known issues" open.
To alleviate this I added links to github issues directly from Sloeber under the arduino menu item. But few people seem to find those.

jantje, 2020-01-03:
Seems I was mistaken and there is no "known issue" in the menu.
So I'll add it.

jantje, 2020-01-03:
@rlogiacco 
Can you provide a Italian translation for

command.know_limitations.info.label=Show the Known limitations of Sloeber ...
command.known_limitations.info.description=Opens a browser page that lists all known limitations (check before creating an issue)

closed 2020-01-18 by jantje