Sloeber/arduino-eclipse-plugin #1124: Could neither enter setup() nor set breakpoints

JochenKoehler69, 2020-01-04:
Hi,
just learning to work with Sloeber and debugging ESP8266 devices.
First examples were quiet successful (however I was just able to enter the source code in the loop() function).
But with the third trial (self written project) I'm not able to debug anyhow :-(.

This is the dump of the debugger:
```
GNU gdb (GDB) 7.11
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=i686-pc-mingw32 --target=xtensa-lx106-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
(trace) gdbarch_init()
(trace) gdbarch_init()
(trace) xtensa_unwind_pc (next_frame = 0x636a768)
(info ) [xtensa_unwind_pc] pc = 0x4010025b
(trace) xtensa_alloc_frame_cache ()
(trace) call0_analyze_prologue (start = 0x4010025b, pc = 0x4010025b, ...)
(verb ) [call0_analyze_prologue] stopped at instr addr 0x4010025b, succeeded
gdbstub_do_break_breakpoint_addr () at D:\sloeber\arduinoPlugin\packages\Jantje\hardware\esp8266\1.0.0\libraries\gdb\gdbstub-entry.S:399
399		break 0,0
Warning: the current language does not match this frame.
(trace) xtensa_skip_prologue (start_pc = 0x4020a92c)
(trace) xtensa_breakpoint_from_pc (pc = 0x4020a939)
(trace) xtensa_skip_prologue (start_pc = 0x401076bc)
(trace) xtensa_breakpoint_from_pc (pc = 0x401076c3)
(trace) xtensa_skip_prologue (start_pc = 0x40107658)
(trace) xtensa_breakpoint_from_pc (pc = 0x40107682)
(trace) xtensa_skip_prologue (start_pc = 0x40107658)
(trace) xtensa_skip_prologue (start_pc = 0x40107658)
(trace) xtensa_breakpoint_from_pc (pc = 0x4010765f)
(trace) xtensa_breakpoint_from_pc (pc = 0x4010765f)
(trace) xtensa_breakpoint_from_pc (pc = 0x4010765f)
(trace) xtensa_breakpoint_from_pc (pc = 0x401076c3)
(trace) xtensa_breakpoint_from_pc (pc = 0x4020a939)
Warning:
Cannot insert breakpoint 1.
Cannot access memory at address 0x4020a939
```
Output of the serial monitor is ending with "$T05#b9"

Code looks like this (gdb.h is also included):

```
void RAMFUNC setup(){
	// Debug Addition
	Serial.begin(115200);
#ifdef DEBUG
	uart_div_modify(0, UART_CLK_FREQ / 115200);
	gdbstub_init();
#endif

```
I'm using Sloeber 4.3.

What additional information do you need to help?

jantje, 2020-01-04:
I would stop debugging, clean all breakpoints set the ones I want and restartthe debugging, 
But I'm not an expert

JochenKoehler69, 2020-01-05:
Thanks for your hint. I tried it once more with all breakpoints deleted.
Afterwards I was able again to step into the loop() function.
However it seems to be impossible to set any breakpoints!
And also it's not possible to do any "step overs". The debugger enters all functions ending up in any "low level" functions without source code and then I stuck ...
Any advice?

Thanks!

jantje, 2020-01-05:
I think you are hitting the limitations of ESP8266 debugging.

freeck, 2020-01-05:
Did you use the RAMFUNC directive in the header of the functions you try to debug?
If not you first have to delete all breakpoints before single stepping. As Jantje stated the debugging function of 8266 is limited. This is due to the fact that it can only handle one hardware breakpoint at a time not using the RAMFUNC.

JochenKoehler69, 2020-01-06:
Hey freek,

thanks for your hint. That's it! I missed the RAMFUNC in some class methods ...
Now it works really fine. This is a great tool!
However, is there a limitation in the number of functions modified by RAMFUNC?

freeck, 2020-01-06:
Sure, as long as you do not exceed the available iram size.
So do not debug big chuncks of code, containing a RAMFUNC directive in the header, but only small functions.

closed 2020-01-07 by jantje