Sloeber/arduino-eclipse-plugin #1129: STM32 issue again?

dferbas, 2020-01-14:
Sloeber does not (by default) link Arduino sketch on Stm32duino platform (Eclipse on Windows 10).

---
I browsed several forums and trackers prior to bothering in this forum.
Here #220 seems to fix it. I think not yet fully.

I installed stm32duino via https://raw.githubusercontent.com/stm32duino/BoardManagerFiles/master/STM32/package_stm_index.json both to Arduino IDE 1.8.10 (AIDE) and via Sloeber 4.3.1 plugin, installed through Marketplace to Eclipse 2019-12 C++.

---
Sketch linking works directly with AIDE, I was able to compile everything with Sloeber, but linking was not successful here. So it was some library issue. I started to browse for missing symbols in source files, then I compared link parameters to these in AIDE - nearly the same. I found AIDE was compiling much more files, and they all come from SrcWrapper library.

Once I added ...\arduinoPlugin\packages\STM32\hardware\stm32\1.8.0\libraries\SrcWrapper\src as a linked folder to my project, linking succeeded.

I realized same can be achieved by adding **SrcWrapper as an Arduino library** via Arduino menu. Much better. There is still a disadvantage - it fixes a project to STM32 platform. I cannot quickly demonstrate to students nice Arduino feature - working on different platforms without an effort (only by changing project build configurations from build button combo).

I encourage to share a note of this - I think, it can save some time to others.

---
I checked **platform.txt** and here is
`recipe.hooks.prebuild.3.pattern.windows=cmd /c echo #include ^<SrcWrapper.h^> > "{build.src_wrapper.path}" `

I checked **environment variables in project preferences** and this hook variable is not in my project environment. I checked plugin for the pattern and found it [here](https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.core/src/io/sloeber/core/tools/Helpers.java) at end of setTheEnvironmentVariablesPostProcessing().

**Any advice?**

jantje, 2020-01-14:
This is a know issue as a leftover of #927 
See this comment https://github.com/Sloeber/arduino-eclipse-plugin/issues/927#issuecomment-394680972

jantje, 2020-01-14:
You can work around your issue with the exclude from build for all non stm32 build configs

closed 2020-01-18 by jantje

dferbas, 2020-01-20:
Thanks for handling this and pointing to the relevant issue.