Sloeber/arduino-eclipse-plugin #1130: Arduino AVR Boards v1.8.x not working

dvsingletary, 2020-01-18:
On an initial install of Sloeber 4.3.2, it attempts to use the latest version of Arduino AVR Boards v1.8.2, but installation fails with a modal error window titled "Problem Occurred" which reads: "'Adopting Arduino Board Platforms' has encountered a problem.  Installing Arduino Board Playforms".  And when you click on "Details>>" it has many "OK" statements, with this in the middle of the log:
Failed to install Arduino AVR Boards
No valid system found for "avr-gcc".
No valid system found for "avrdude".

This bug, as well as a rather involved workaround for it is described accurately here on the eclipse community forums: https://www.eclipse.org/forums/index.php/r/th_id/1100629/1814918/0/0/??SQ=533f76a20e9982a6879558c84e391bf6&S=ae19a7b18d623e412d64f6faaaa557c4

This is on Mac OS Catalina 10.15 and Mojave 10.14.  Arduino AVR Boards 1.6.23 (last release of 1.6.x) works fine, but any of the 1.8.x releases have this problem.  This effectively prevents using Sloeber on Catalina since Catalina requires 64 bit tools which are only provided with 1.8.x.

jantje, 2020-01-18:
The bug in eclipse you reference to is not for sloeber but for another eclipse arduino plugin (which has been stale for 2 years now)
But the root cause is the same.
I guess your issue is a duplicate of #1111. workaround: upgrade to latest nightly.
Please report progress. I'm planning to make a new release but I can't test on mac.

dvsingletary, 2020-01-19:
Updating to the nightly resolved the issue - Via the Sloeber->preferences->Arduino menu I'm able to check the box for Arduino AVR Boards 1.8.2 and it installs and works as expected.  Thanks!

jantje, 2020-01-19:
Thanks for the feedback.
I'll close this issue marking it as duplicate and hope to make a release soon.

closed 2020-01-19 by jantje