Sloeber/arduino-eclipse-plugin #1131: Linux Product V4 doesn't run

Mark-81, 2020-01-20:
I'm experience the same issues of this one:

https://github.com/Sloeber/arduino-eclipse-plugin/issues/812

but I'm under Ubuntu 19.10.
I extracted the compressed file (with `tar`) but it says "permission denied".
I tried to install JRE and JDK even if it seems they are not needed.

Of course the `sloeber-ide` has execution permissions.

jantje, 2020-01-20:
are you on a 64 bit system?

Mark-81, 2020-01-21:
Yes, but I've just figured out why: it has nothing to do with the code.
For the records: the mount point where the directory lives, was mounted by default with the `noexec` option.

closed 2020-01-21 by Mark-81

jantje, 2020-01-21:
Thanks for the feedback.
"permission denied" is one of those error messages that seem very clear but give you absolutely no clue about what might be wrong.