Sloeber/arduino-eclipse-plugin #1133: Unterminated quoted string when using another STM32 Core

Mark-81, 2020-01-21:
**Describe the bug**
Building a project based upon the STM32Duino Core results in the following error:

```
14:38:37 **** Incremental Build of configuration Release for project Test1 ****
make -j4 all 
make --no-print-directory pre-build
bash -c "[ -f /home/mark/Test1/Release/sketch ] || mkdir -p /home/mark/Test1/Release/sketch"
bash -c "[ -f /build_opt.h ] || touch /home/mark/Test1/Release/sketch/build_opt.h"
/bin/sh: 1: Syntax error: Unterminated quoted string
make[1]: *** [makefile:159: pre-build] Error 2
make: *** [makefile:118: all] Error 2
bash -c "echo  > /home/mark/Test1/Release/sketch/SrcWrapper.cpp" "
"make -j4 all" terminated with exit code 2. Build might be incomplete.
```

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new Arduino sketch
2. Select `STM32/hardware/stm32/1.8.0` as platform's directory
3. Build
4. See error in the Console output

**Expected behavior**
The correct build of the sketch.

**Desktop (please complete the following information):**
 - OS: Ubuntu 19.10 64-bit

jantje, 2020-01-21:
Seems like an issue with the prebuild command.
It is trying to create the file /home/mark/Test1/Release/sketch/build_opt.h but that fails because there is a double quote at the end.
I assume that as a workaround you can run the command 
touch /home/mark/Test1/Release/sketch/build_opt.h
and then the build should no longer give this issue.

Mark-81, 2020-01-21:
I don't know if this the root cause, but I found this:

    environment/project/io.sloeber.core.toolChain.release.1353533361/A.RECIPE.HOOKS.PREBUILD.3.PATTERN/value=bash -c "echo $0 > ${A.BUILD.SRC_WRAPPER.PATH}" "

in `.settings/org.eclipse.cdt.core.prefs`.
Removing the trailing `"` leads to go ahead in the build (with other errors, though).

jantje, 2020-01-21:
I strongly advice not to fiddle around in the eclipse files.
You can also remove the " in eclipse->project properties->c/c++ build->environment
but this will be overwritten when sloeber decides that a update is needed.
best place to tackle is the platform.txt recipe.hooks.prebuild variable

Mark-81, 2020-01-22:
> It is trying to create the file /home/mark/Test1/Release/sketch/build_opt.h but that fails because there is a double quote at the end.
> I assume that as a workaround you can run the command
> touch /home/mark/Test1/Release/sketch/build_opt.h
> and then the build should no longer give this issue.

I'm not sure about this. The `build_opt.h` file is created.
In fact, if you look at the output above, trailing quotes is placed on the next command:

```
bash -c "[ -f /build_opt.h ] || touch /home/mark/Test1/Release/sketch/build_opt.h"
/bin/sh: 1: Syntax error: Unterminated quoted string
make[1]: *** [makefile:159: pre-build] Error 2
make: *** [makefile:118: all] Error 2
bash -c "echo  > /home/mark/Test1/Release/sketch/SrcWrapper.cpp" "
```
It's about the `SrcWrapper.cpp` file. But this instruction is called anyway, even if the file is already there.

Inspecting the `platform.txt` file I see:

    recipe.hooks.prebuild.1.pattern.linux=bash -c "[ -f {build.path}/sketch ] || mkdir -p {build.path}/sketch"
    recipe.hooks.prebuild.2.pattern.linux=bash -c "[ -f {build.opt.sourcepath} ] || touch {build.opt.path}"
    recipe.hooks.prebuild.3.pattern.linux=bash -c "echo $0 > {build.src_wrapper.path}" "#include <SrcWrapper.h>"

It seems the last command is incomplete. 
Is there any reason for that? Or a temporary workaround to fix it?

jantje, 2020-01-22:
>I'm not sure about this. The build_opt.h file is created.

you are right.

Looking at the platform.txt and the project properties you provided t looks as if #include <SrcWrapper.h>" has dropped off of A.BUILD.SRC_WRAPPER.PATH.
It also looks like $0 is not working as expected.
What is the command 
bash -c "echo  $0> /home/mark/Test1/Release/sketch/SrcWrapper.cpp" " #include <SrcWrapper.h>"
Supposed to do?

FYI; this hook thing (heavily used by the esp guys) is making me very nervous.  Why do you need to make an empty header file in the build output folder?

closed 2020-01-25 by Mark-81