Sloeber/arduino-eclipse-plugin #1134: How to #include external libreries

Mark-81, 2020-01-22:
I'm not sure I'm doing it right.
In the Preferences I see the path for external libraries: `/home/mark/Arduino/libraries`.

But it seems it doesn't find libraries there.
For example I cloned the [FlexyStepper](https://github.com/Stan-Reifel/FlexyStepper) repo (I know it's available from the library manager, but it's just an example) into the above directory.

Still it cannot resolve the `#include`.
Am I missing something?

jantje, 2020-01-22:
Did you try to build again?
It takes some time/builds before the library gets added.
You can manually add a library with arduino->add library (if it is not there sloeber will not auto add it)
There may  also be a conflict. look in the error log (window->show view->search for error)

Mark-81, 2020-01-22:
I rebuilt the project a lot of times and I also tried the Project > C/C++ Index > Re-solve Unresolved Includes.

The error log contains some error about eclipse, but nothing to do about the offending #include.

jantje, 2020-01-22:
What does the include browser say? https://www.youtube.com/watch?v=5Yi5xeeqlsk
You can manually add a library with arduino->add library (if it is not there sloeber will not auto add it)
Did you turn off auto include?

Mark-81, 2020-01-25:
Sorry I gave up. There are so many other issues in the IDE (at least in my setup, I'm not saying in your code) I cannot spent more time. I sadly keep the default IDE right now. Anyway, thanks for the hints and for the kind support.

closed 2020-01-25 by Mark-81