Sloeber/arduino-eclipse-plugin #1136: Bad Compile command when using private hardware definitions

ksstille, 2020-01-22:
**Describe the bug**
When using a Board from a private hardware path the compile command is beginning with the Include of the private hardware path. Anything before (gcc ....) is missing. This leads to the impossibility to compile.

**To Reproduce**
1. Set a private hardware path
2. Select a board from there
3. Build the sketch

**Expected behavior**
Clean compile

**System**
 
OS: Linux, Ubuntu 19.10

**Error output**

`22:42:23 **** Incremental Build of configuration Release for project Ampel_INT_Eclipse ****
make all 
Building file: ../sloeber.ino.cpp
Starting C++ compile
I"/home/hd/karl-ste/Projekte/Arduino/hardware/avr/Eigene/cores/arduino" -I"/home/hd/karl-ste/Projekte/Arduino/hardware/avr/Eigene/variants/eightanaloginputs" -I"/home/karl-ste/Projekte/Arduino/libraries/Adafruit_PWM_Servo_Driver_Library" -I"/home/karl-ste/Projekte/Arduino/libraries/PinChangeInterrupt/src" -I"/home/karl-ste/Projekte/Arduino/libraries/UDEV" -I"/opt/Sloeber/arduinoPlugin/packages/arduino/hardware/avr/1.8.2/libraries/EEPROM/src" -I"/opt/Sloeber/arduinoPlugin/packages/arduino/hardware/avr/1.8.2/libraries/Wire/src" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../sloeber.ino.cpp"  "sloeber.ino.cpp.o"
/bin/sh: 1: I/home/hd/karl-ste/Projekte/Arduino/hardware/avr/Eigene/cores/arduino: not found
make: [subdir.mk:20: sloeber.ino.cpp.o] Fehler 127 (ignoriert)
Finished building: ../sloeber.ino.cpp
 
Building file: /opt/Sloeber/arduinoPlugin/packages/arduino/hardware/avr/1.8.2/libraries/Wire/src/utility/twi.c
Starting C compile
I"/home/hd/karl-ste/Projekte/Arduino/hardware/avr/Eigene/cores/arduino" -I"/home/hd/karl-ste/Projekte/Arduino/hardware/avr/Eigene/variants/eightanaloginputs" -I"/home/karl-ste/Projekte/Arduino/libraries/Adafruit_PWM_Servo_Driver_Library" -I"/home/karl-ste/Projekte/Arduino/libraries/PinChangeInterrupt/src" -I"/home/karl-ste/Projekte/Arduino/libraries/UDEV" -I"/opt/Sloeber/arduinoPlugin/packages/arduino/hardware/avr/1.8.2/libraries/EEPROM/src" -I"/opt/Sloeber/arduinoPlugin/packages/arduino/hardware/avr/1.8.2/libraries/Wire/src" -MMD -MP -MF"libraries/Wire/src/utility/twi.c.d" -MT"libraries/Wire/src/utility/twi.c.o" -D__IN_ECLIPSE__=1 "/opt/Sloeber/arduinoPlugin/packages/arduino/hardware/avr/1.8.2/libraries/Wire/src/utility/twi.c"  "libraries/Wire/src/utility/twi.c.o"
/bin/sh: 1: I/home/hd/karl-ste/Projekte/Arduino/hardware/avr/Eigene/cores/arduino: not found
Finished building: /opt/Sloeber/arduinoPlugin/packages/arduino/hardware/avr/1.8.2/libraries/Wire/src/utility/twi.c
make: [libraries/Wire/src/utility/subdir.mk:20: libraries/Wire/src/utility/twi.c.o] Fehler 127 (ignoriert)
 
Building file: /opt/Sloeber/arduinoPlugin/packages/arduino/hardware/avr/1.8.2/libraries/Wire/src/Wire.cpp
Starting C++ compile
I"/home/hd/karl-ste/Projekte/Arduino/hardware/avr/Eigene/cores/arduino" -I"/home/hd/karl-ste/Projekte/Arduino/hardware/avr/Eigene/variants/eightanaloginputs" -I"/home/karl-ste/Projekte/Arduino/libraries/Adafruit_PWM_Servo_Driver_Library" -I"/home/karl-ste/Projekte/Arduino/libraries/PinChangeInterrupt/src" -I"/home/karl-ste/Projekte/Arduino/libraries/UDEV" -I"/opt/Sloeber/arduinoPlugin/packages/arduino/hardware/avr/1.8.2/libraries/EEPROM/src" -I"/opt/Sloeber/arduinoPlugin/packages/arduino/hardware/avr/1.8.2/libraries/Wire/src" -MMD -MP -MF"libraries/Wire/src/Wire.cpp.d" -MT"libraries/Wire/src/Wire.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "/opt/Sloeber/arduinoPlugin/packages/arduino/hardware/avr/1.8.2/libraries/Wire/src/Wire.cpp"  "libraries/Wire/src/Wire.cpp.o"
/bin/sh: 1: I/home/hd/karl-ste/Projekte/Arduino/hardware/avr/Eigene/cores/arduino: not found
Finished building: /opt/Sloeber/arduinoPlugin/packages/arduino/hardware/avr/1.8.2/libraries/Wire/src/Wire.cpp
make: [libraries/Wire/src/subdir.mk:20: libraries/Wire/src/Wire.cpp.o] Fehler 127 (ignoriert)
 
Building file: /home/karl-ste/Projekte/Arduino/libraries/UDEV/UDEV.cpp
Starting C++ compile
I"/home/hd/karl-ste/Projekte/Arduino/hardware/avr/Eigene/cores/arduino" -I"/home/hd/karl-ste/Projekte/Arduino/hardware/avr/Eigene/variants/eightanaloginputs" -I"/home/karl-ste/Projekte/Arduino/libraries/Adafruit_PWM_Servo_Driver_Library" -I"/home/karl-ste/Projekte/Arduino/libraries/PinChangeInterrupt/src" -I"/home/karl-ste/Projekte/Arduino/libraries/UDEV" -I"/opt/Sloeber/arduinoPlugin/packages/arduino/hardware/avr/1.8.2/libraries/EEPROM/src" -I"/opt/Sloeber/arduinoPlugin/packages/arduino/hardware/avr/1.8.2/libraries/Wire/src" -MMD -MP -MF"libraries/UDEV/UDEV.cpp.d" -MT"libraries/UDEV/UDEV.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "/home/karl-ste/Projekte/Arduino/libraries/UDEV/UDEV.cpp"  "libraries/UDEV/UDEV.cpp.o"
/bin/sh: 1: I/home/hd/karl-ste/Projekte/Arduino/hardware/avr/Eigene/cores/arduino: not found
Finished building: /home/karl-ste/Projekte/Arduino/libraries/UDEV/UDEV.cpp
make: [libraries/UDEV/subdir.mk:20: libraries/UDEV/UDEV.cpp.o] Fehler 127 (ignoriert)
 
Building file: /home/karl-ste/Projekte/Arduino/libraries/PinChangeInterrupt/src/PinChangeInterrupt.cpp
Starting C++ compile
I"/home/hd/karl-ste/Projekte/Arduino/hardware/avr/Eigene/cores/arduino" -I"/home/hd/karl-ste/Projekte/Arduino/hardware/avr/Eigene/variants/eightanaloginputs" -I"/home/karl-ste/Projekte/Arduino/libraries/Adafruit_PWM_Servo_Driver_Library" -I"/home/karl-ste/Projekte/Arduino/libraries/PinChangeInterrupt/src" -I"/home/karl-ste/Projekte/Arduino/libraries/UDEV" -I"/opt/Sloeber/arduinoPlugin/packages/arduino/hardware/avr/1.8.2/libraries/EEPROM/src" -I"/opt/Sloeber/arduinoPlugin/packages/arduino/hardware/avr/1.8.2/libraries/Wire/src" -MMD -MP -MF"libraries/PinChangeInterrupt/src/PinChangeInterrupt.cpp.d" -MT"libraries/PinChangeInterrupt/src/PinChangeInterrupt.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "/home/karl-ste/Projekte/Arduino/libraries/PinChangeInterrupt/src/PinChangeInterrupt.cpp"  "libraries/PinChangeInterrupt/src/PinChangeInterrupt.cpp.o"
/bin/sh: 1: I/home/hd/karl-ste/Projekte/Arduino/hardware/avr/Eigene/cores/arduino: not found
Finished building: /home/karl-ste/Projekte/Arduino/libraries/PinChangeInterrupt/src/PinChangeInterrupt.cpp
make: [libraries/PinChangeInterrupt/src/subdir.mk:32: libraries/PinChangeInterrupt/src/PinChangeInterrupt.cpp.o] Fehler 127 (ignoriert)
 
Building file: /home/karl-ste/Projekte/Arduino/libraries/PinChangeInterrupt/src/PinChangeInterrupt0.cpp
Starting C++ compile
I"/home/hd/karl-ste/Projekte/Arduino/hardware/avr/Eigene/cores/arduino" -I"/home/hd/karl-ste/Projekte/Arduino/hardware/avr/Eigene/variants/eightanaloginputs" -I"/home/karl-ste/Projekte/Arduino/libraries/Adafruit_PWM_Servo_Driver_Library" -I"/home/karl-ste/Projekte/Arduino/libraries/PinChangeInterrupt/src" -I"/home/karl-ste/Projekte/Arduino/libraries/UDEV" -I"/opt/Sloeber/arduinoPlugin/packages/arduino/hardware/avr/1.8.2/libraries/EEPROM/src" -I"/opt/Sloeber/arduinoPlugin/packages/arduino/hardware/avr/1.8.2/libraries/Wire/src" -MMD -MP -MF"libraries/PinChangeInterrupt/src/PinChangeInterrupt0.cpp.d" -MT"libraries/PinChangeInterrupt/src/PinChangeInterrupt0.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "/home/karl-ste/Projekte/Arduino/libraries/PinChangeInterrupt/src/PinChangeInterrupt0.cpp"  "libraries/PinChangeInterrupt/src/PinChangeInterrupt0.cpp.o"
/bin/sh: 1: I/home/hd/karl-ste/Projekte/Arduino/hardware/avr/Eigene/cores/arduino: not found
Finished building: /home/karl-ste/Projekte/Arduino/libraries/PinChangeInterrupt/src/PinChangeInterrupt0.cpp
make: [libraries/PinChangeInterrupt/src/subdir.mk:39: libraries/PinChangeInterrupt/src/PinChangeInterrupt0.cpp.o] Fehler 127 (ignoriert)
 
Building file: /home/karl-ste/Projekte/Arduino/libraries/PinChangeInterrupt/src/PinChangeInterrupt1.cpp
Starting C++ compile
I"/home/hd/karl-ste/Projekte/Arduino/hardware/avr/Eigene/cores/arduino" -I"/home/hd/karl-ste/Projekte/Arduino/hardware/avr/Eigene/variants/eightanaloginputs" -I"/home/karl-ste/Projekte/Arduino/libraries/Adafruit_PWM_Servo_Driver_Library" -I"/home/karl-ste/Projekte/Arduino/libraries/PinChangeInterrupt/src" -I"/home/karl-ste/Projekte/Arduino/libraries/UDEV" -I"/opt/Sloeber/arduinoPlugin/packages/arduino/hardware/avr/1.8.2/libraries/EEPROM/src" -I"/opt/Sloeber/arduinoPlugin/packages/arduino/hardware/avr/1.8.2/libraries/Wire/src" -MMD -MP -MF"libraries/PinChangeInterrupt/src/PinChangeInterrupt1.cpp.d" -MT"libraries/PinChangeInterrupt/src/PinChangeInterrupt1.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "/home/karl-ste/Projekte/Arduino/libraries/PinChangeInterrupt/src/PinChangeInterrupt1.cpp"  "libraries/PinChangeInterrupt/src/PinChangeInterrupt1.cpp.o" [... cut ...]
make: [libraries/Adafruit_PWM_Servo_Driver_Library/subdir.mk:20: libraries/Adafruit_PWM_Servo_Driver_Library/Adafruit_PWMServoDriver.cpp.o] Fehler 127 (ignoriert)
 
make: /home/hd/karl-ste/Projekte/Arduino/Ampel_INT_Eclipse/Release/arduino.ar: Kommando nicht gefunden
Starting archiver
make: *** [makefile:78: /home/hd/karl-ste/Projekte/Arduino/Ampel_INT_Eclipse/Release/arduino.ar] Fehler 127
/home/hd/karl-ste/Projekte/Arduino/Ampel_INT_Eclipse/Release/arduino.ar 
"make all" terminated with exit code 2. Build might be incomplete.

22:42:23 Build Failed. 11 errors, 0 warnings. (took 544ms)`

jantje, 2020-01-23:
Please be more verbose in the "Describe the bug" and the "To Reproduce" section.
Also explain why you want to use the private hardware.

ksstille, 2020-01-23:
I use an Arduino Nano at 8 Mhz. I defined it in a private hardware folder (hardware/avr/Eigene) under the Arduino home and added this place to the private hardware folders. Find the boards.txt attached.
When choosing this privately defined board, the first part of the compile command is missing. By this gcc is not started. See output from the first post.

[boards.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/4103775/boards.txt)

**To Reproduce**

- Save attached boards.txt under subfolder (ARDUINO PROJECT HOME)/hardware/avr/something below the Arduino project directory.
- Add "(ARDUINO PROJECT HOME)/hardware" under Window>Preferences>Arduino as Private hardware path 
- Under Project > Properties > Arduino from Platform folder choose the folder added before
- Choose Arduino Nano 3.3V 8MHz from the Boards
- Build any sketch

jantje, 2020-01-23:
again:
Also explain why you want to use the private hardware.

ksstille, 2020-01-23:
The private hardware is needed as I change the Arduino Nanos to 3.3V for interoperability with a 3.3V I2C bus. At this voltage the maximum frequency is 8MHz and the fuses are changed to turn on the internal resonator and define the board as private hardware.

But this bug report is not about the private hardware. It is about the software error constructing the compilation commands when choosing this option in the software. 

I assume that in the construction of the compilation command string the string for the private hardware path is not appended but overwritten.

jantje, 2020-01-23:
Private hardware is important because which compiler to use is defined by the json file used to install the platform. Private hardware->no json file->no context
Also it seems you arre "making your own board" so you are fiddling in the boards.txt and platform.txt which means you may have broken things.
The only private hardware I know that runs out of the box is teensy. This is only so because Paul of Teensy modded Teensyduino.
What I'm trying to say is that using private hardware is hard and requires a lot of understanding of the arduio framework and how it is implemented in Sloeber.
I myself try to avoid it.

ksstille, 2020-01-23:
It was a problem of a platform.txt which only contained additional options as accepted by the Arduino IDE. Copying the contents of the original platform.txt  into the private one solved the issue.

jantje, 2020-01-23:
good to see you fixed it.

closed 2020-01-23 by jantje