Sloeber/arduino-eclipse-plugin #1137: external libraries - syntax error

Sebastiangka, 2020-01-23:
Hi everyone!

I start using Sloeber some weeks ago so I don't have much idea of everything. I'm running Sloeber in Windows and after having some problems linking extern and my libraries, have now the following problem.

I tried some of the examples of de SdFat-Beta library from greiman in Arduino IDE and works. I want to use the same library in Sloeber and after compiling I got some errors like **not declared variables in this scope, uninitialized reference member in... or expected '}' at end of input**.

Since I didn´t modify this library at all and it worked before, I suppose that I'm missing something about the configuration or something.

Do any of you have have an idea how can I solve this problem?

Thanks in advance

jantje, 2020-01-23:
Do not cross post!!!!!

>Do any of you have have an idea how can I solve this problem?

What you are asking is: "How do I do problem analyses".
This is a very broad question. but here are some handles for the "code doesn't build in Sloeber but it does in Arduino IDE problem".
1)Clean your project and then build it.

2)Focus on the very first error in the build output (the console). This is probably the easiest thing to fix as following errors may be a result of the first error.

3)Build the problem again and see whether you get the same first error.

4)Try to understand what is going on. This is hard! Change the code a bit and build ..... (experiment)

5)If you don't get it compare the build command of this specific file between arduino IDE and Sloeber.
If these are different check the sloeber known issues https://github.com/Sloeber/arduino-eclipse-plugin/issues?utf8=%E2%9C%93&q=label%3A%22status%3A+known+limitation%22+
If there is no known issue explaining the difference create a issue explaining the behaviour.

6)If there is no difference in the command and you still don't get it check the library and board provider support channels searching for sloeber and eclipse.

7)If you are still not getting it create a issue on what you think is the most appropriate support channel and be explicit and verbose.
Tell the big story and zoom in on the details of what the problem is and what you tried.
Hope for someone capable and willing to help.
In the mean time restart with 1. You should have learned a lot and you might have overlooked something.

8 ) If someone tries to help: be grateful and open to the solution proposed. The proposal may sound bunkers but don't forget "You don't have a clue and a complete stranger has invested time and energy to help you. The least you can do is check the proposal out."

9) Respond to the responder. Don't be a taker. You have received help, show you appreciate it.

Sebastiangka, 2020-01-23:
Thanks for your fast answer and your time. 
I did try the first 4 steps before I started this issue. Will take some time to work on 5) or explain (add precise information) more about this issue, in case that 5) wouldn't work well.

erdemuncuoglu, 2020-01-24:
@jantje :1st_place_medal: 
Some how we should hang your post some where in the README like under the FAQ. I will go on it.

jantje, 2020-01-24:
> 
> 
> @jantje 🥇
> Some how we should hang your post some where in the README like under the FAQ. I will go on it.

I was thinking the same :-) Seems the faq troubleshooting is a good place and also needs a cleanup. Be my guest,

Sebastiangka, 2020-01-29:
Well, every start is not that easy, I appreciate your time and advice. It looks like I need to be much more careful when adding some libraries in Sloeber. I recommend for people like me, that doesn't have much experience with Sloeber, after creat your project choose only the libraries you need and build your project to check that everything is fine. Then you can copy or open one of the example codes and run it.
My mistake was that I chose a library from the Arduino-List and not the Third-Party-Library that I wanted to use.
Here an example :
[image]

closed 2020-01-29 by Sebastiangka

jantje, 2020-01-29:
Thanks for searching. Indeed when you use many libraries you have to be careful to pick the right one.