Sloeber/arduino-eclipse-plugin #1140: Failed to create project for stm32

hasikp1, 2020-01-28:
Unable to create a new project.
Failed to create project java.lang.reflect.InvocationTargetException.
I have attached the error log and the screenshot of the project settings.
 
[eclipserr.log](https://github.com/Sloeber/arduino-eclipse-plugin/files/4120996/eclipserr.log)

[image]

jantje, 2020-01-30:
Are you using the sloeber product? It looks as if there is a conflicting CDT Sloeber plugin

hasikp1, 2020-02-05:
Which one should I use?. There are two in the eclipse market place.
[image]

erdemuncuoglu, 2020-02-05:
Second one: "The Arduino Eclipse plugin named Sloeber V4"

Alternatively you can add nightly repository into eclipse:
Open from menu `Help -> Install New Software...` then press button `Add..`
Put below into form
Name: Sloeber nightly
Location: http://eclipse.baeyens.it/update/V4/nightly

jantje, 2020-02-08:
You should download from http://eclipse.baeyens.it/

closed 2020-02-22 by jantje