Sloeber/arduino-eclipse-plugin #1141: MacOS Product Bundle is corrupt

StevieDee123, 2020-01-30:
The current 64-bit MacOS bundle is corrupt. When I download V4.3.2_mac64.2019-10-07_08-26-30.tar.gz and extract it, the Sloeber executable shows the following dialog.

[image]

I have tried this multiple times over the past few days.

StevieDee123, 2020-01-31:
I figured this out. The .tar.gz file didn't like being unpacked with the default Mac Archive Utility. It works fine when I do command line:
gunzip -c V4.3.2_mac64.2019-10-07_08-26-30.tar.gz | tar xopf -

Thanks. Sorry to bother you :)

erdemuncuoglu, 2020-01-31:
You're welcome.

jantje, 2020-01-31:
Would it be possible to make the tar mac archive compatible?
If not; should we not add this info to the installation instructions.
Or is the mac archive utility something non-standard?

liebman, 2020-01-31:
Note that even `tar zxf V4.3.2_mac64.2019-10-07_08-26-30.tar.gz` fails.  only `gunzip -c V4.3.2_mac64.2019-10-07_08-26-30.tar.gz | tar xf -` works.

StevieDee123, 2020-01-31:
The default Mac archive utility is pretty standard. I've never seen this with other gzipped tarball files before. I'm not sure what is causing it. I would suggest at a minimum, add the command line details to the installation instructions.

jantje, 2020-01-31:
I'm not sure how all this works but the code that does the tarring is here
https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.product/pom.xml#L38

and the build output for the tar is here 
```
Installing org.eclipse.rse.core.feature.group 4.5.100.201907301734.
Installing org.eclipse.cdt.native.feature.group 9.9.0.201909051303.
Operation completed in 18915 ms.
[INFO] 
[INFO] --- tycho-p2-director-plugin:1.3.0:archive-products (archive-products) @ io.sloeber.product ---
[INFO] Building tar: /share/homes/admin/.jenkins/jobs/arduino eclipse plugin build/workspace/io.sloeber.product/target/products/sloeber-ide-sloeber_release-win32.win32.x86_64.tar.gz
[INFO] Building tar: /share/homes/admin/.jenkins/jobs/arduino eclipse plugin build/workspace/io.sloeber.product/target/products/sloeber-ide-sloeber_release-linux.gtk.x86_64.tar.gz
[INFO] Building tar: /share/homes/admin/.jenkins/jobs/arduino eclipse plugin build/workspace/io.sloeber.product/target/products/sloeber-ide-sloeber_release-macosx.cocoa.x86_64.tar.gz
[JENKINS] Archiving disabled
[INFO]                                                                         
[INFO] ------------------------------------------------------------------------
[INFO] Building io.sloeber.updatesite 4.3.1-SNAPSHOT
[INFO] ------------------------------------------------------------------------
```
There might be something "wrong" with the tar that is used on my build machine.

wimjongman, 2020-02-05:
> I'm not sure how all this works but the code that does the tarring is here
> https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.product/pom.xml#L38

Why not use zip? tar.gz seems overly complicated

jantje, 2020-02-08:
> 
> Why not use zip? tar.gz seems overly complicated

Zip for windows and mac? Also for linux?
No idea how to tackle his in a pom.

liebman, 2020-02-08:
Zip is available on linux as well.  I'd even bet that on the common desktop distro's it may be installed by default most of the time.....

wimjongman, 2020-02-09:
Just replace tar.gz with zip in the pom you linked earlier.

TrippyLighting, 2020-04-19:
Hmmm.... so the status says "fixed in nightly". But then following the nightlies one gets greeted with:
"If you ended up here you took a shortcut and failed. Go back to start. Start is here installAdvice.shtml"

So, in other words, the fix is not available to mere mortals and I have to resort to the stable and unpack this using the method @StevieDee123 posted above?
:/

jantje, 2020-04-19:
@TrippyLighting Happy to see you again :-)
The nightly must be installed using the eclipse update site.
As I'm in the middle of making a release I could offer you a beta product download.

jantje, 2020-04-19:
Here is a pre release https://github.com/Sloeber/arduino-eclipse-plugin/releases/tag/V4_3_3_beta

TrippyLighting, 2020-04-19:
@jantje it seems that zipping it as suggested by @wimjongman did not resolve the issue.
Depending on how users have configured their mac setup, when they click on the green download button the download will automatically go into the downloads folder. There it is by default (Or so I believe. THis can be changed of course) automatically unzipped using the standard unarchive utility on macOS. Even with the pre-release you just did, that is still the case and users will receive the same "Sloeber is damaged ..." message.

I was able to unpack the stable version using the command line and gunzip. I've not tried that yet with the beta.

Now I have to figure out how to get it to work again with the newer Teensyduino releases. Wish me luck!

TrippyLighting, 2020-04-19:
Aside from the unpacking issue, this was the easiest update to use teensy boards ever!

jantje, 2020-04-19:
@TrippyLighting Am I correct to assume you succeeded to install the 4.3.3 beta but needed a special unpack command?

TrippyLighting, 2020-04-19:
@jantje nope ;-)
I was able to unpack the 4.3.2 stable with gunzip and.
I was not able to get 4.3.3 to unpack into a usable version. I still receive the same "Sloeber is damaged ..." message. That is with the macOS Archive Utility. 
No success unzipping this with gunzip. Maybe I haven't; tried hard enough ...

jantje, 2020-04-19:
Disappointing to me :-(
I will need to look for a better solution.

jantje, 2020-04-19:
Can you check the checksum of the file you downloaded? This is what I get with the original file and the uploaded/downloaded version.
Extracts perfectly as well.
```
Name: V4.3.3_mac64.2020-04-19_06-43-35.zip
Size: 185525081 bytes (176 MiB)
CRC64: 3571B005CB2856D9
```

TrippyLighting, 2020-04-19:
How do I do that?
A simple:
cksum V4.3.3_mac64.2020-04-19_06-43-35.zip

returns this:
3305885082 185525081 V4.3.3_mac64.2020-04-19_06-43-35.zip

jantje, 2020-04-20:
According to this article cksum calculates the CRC https://en.wikipedia.org/wiki/Cksum
The numbers do not match what my checksum tool calculates. Not even in length
```
Name: V4.3.3_mac64.2020-04-19_06-43-35.zip
Size: 185525081 bytes (176 MiB)
CRC32: 11B6DE88
CRC64: 3571B005CB2856D9
SHA256: 0AB4E3004E8DB1085E438AB92890AA40846DC03FE01E8E9263FB191D4199E26C
SHA1: 3BC54A7A74BA9D451F7327A27C9DC96748EDA106
BLAKE2sp: D26DFF159BFE633E55DDC16F66482C6CBD03ECE598F9799AE7F6528081E21CC0
```

IMHO this document states that `cksum -o 3 V4.3.3_mac64.2020-04-19_06-43-35.zip` should give you the CRC32 https://ss64.com/osx/cksum.html

wimjongman, 2020-04-21:
Maybe it is related to this:

https://www.eclipse.org/forums/index.php/t/1101775/

https://bugs.eclipse.org/bugs/show_bug.cgi?id=558570

wimjongman, 2020-04-21:
There is also this thread. 

https://github.com/MacDownApp/macdown/issues/1106

jantje, 2020-04-21:
@wimjongman thanks for the input. It confirms my thinking it is a mac security issue.
Seems that right clicking and ignoring the sercurity should do the trick though I have no idea where to right click.

TrippyLighting, 2020-04-21:
@jantje 
SHA1:	3bc54a7a74ba9d451f7327a27c9dc96748eda106 
SHA256:	0ab4e3004e8db1085e438ab92890aa40846dc03fe01e8e9263fb191d4199e26c

Looks OK to me.

TrippyLighting, 2020-04-21:
@jantje @wimjongman I agree that this appears to b a macOS security "thing". Following one of the links @wimjongman provided it is suggested to remove the quarantine flag from the manually downloaded .app-file.

So after unpacking the 4.33 zip file with the macOS native Archive Utility I did just that:

`sudo xattr -r -d com.apple.quarantine Sloeber.app`

Now it runs fine!

jantje, 2020-04-21:
@TrippyLighting so this turns out to be another security invoked issue on mac 😠 
If I understand correctly there are 3 ways of working around this problem
1.1)download with mac browser
1.2)unpacked the sloeber ide zip file with the macOS native Archive Utility 
1.3)sudo xattr -r -d com.apple.quarantine Sloeber.app

2.1)download with mac browser
2.2)unzip with gunzip

3.1) download with  wget (currently wget https://github.com/Sloeber/arduino-eclipse-plugin/releases/download/V4_3_3_beta/V4.3.3_mac64.2020-04-19_06-43-35.zip)
3.2) unpack with gunzip or  the macOS native Archive Utility 

I need to document "this mac feature". I think the first one is the most future proof as I suspect gunzip and wget will be brought into compliance to the security requirements.
Do you agree this is the best option?

TrippyLighting, 2020-04-21:
@jantje #1 is probably the most common way mac users would download and unzip a file.
It would definitely help to add a few lines of explanation and the  sudo xattr- .... to the installation documentation.
Not everyone will dig through issues on GitHub to find a solution ;-)

jantje, 2020-04-21:
Web site should be updated soon :-) (that is if build succeeds)

closed 2020-04-21 by jantje

tvand, 2020-07-20:
The current bundle V4.3.3_mac64.2020-04-23_11-36-51.zip additionally required explicit specification of a -vm path in sloeber-ide.ini in my case (macOS 10.15.5)

jantje, 2024-02-08:
Seems Sloeber 4.4.3 has install issues as well #1624.
As there are multiple Sloeber mac users here I post this message hoping someone can take a look.
Jantje