Sloeber/arduino-eclipse-plugin #1142: Compile Error #include expects "FILENAME" or <FILENAME>

Nexus1212, 2020-02-08:
There is a Bug in the compiling of the MD5.h files.
Path:
C:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.4\tools\sdk\include\mbedtls\mbedtls

see issue:
https://github.com/espressif/arduino-esp32/issues/1327
https://github.com/Sloeber/arduino-eclipse-plugin/issues/1028

If I changed the MD5.h:
#if !defined(MBEDTLS_CONFIG_FILE)
#include "config.h"
#else
#include MBEDTLS_CONFIG_FILE
#endif

to:
#if !defined(MBEDTLS_CONFIG_FILE)
#include "config.h"
#880 
#define VAL(str) #str
#define TOSTRING(str) VAL(str)
#include TOSTRING(MBEDTLS_CONFIG_FILE)
#endif

it works! Do you know a better sulotion?

jantje, 2020-02-08:
> Do you know a better solution?

From my point of view the workaround is the way it should be done.

closed 2020-02-08 by jantje

madhephaestus, 2020-07-12:
PR of this workaround can be found here: https://github.com/espressif/arduino-esp32/pull/4159

jantje, 2020-07-12:
As far as I know this is fixed in the latest Sloeber and no workaround is needed

PappyZero, 2023-06-07:
#include SPI.h&gt;
#include &lt;Wire.h&gt;
#include &lt;ESP8266WiFi.h&gt; 
#include &lt;SoftwareSerial.h&gt; 
#include &lt;ESP8266WebServer.h&gt;
#include &lt;ESP8266HTTPClient.h&gt;
#include &lt;Adafruit_GFX.h&gt; 
#include &lt;Adafruit_SSD1306.h&gt; 
#include &lt;Adafruit_Fingerprint.h&gt; 

Please this is the code that I giving me an error. 

And here are the error messages: 

C:\Users\DELL\AppData\Local\Temp\.arduinoIDE-unsaved202357-2320-1fc3or9.1k34\sketch_jun7a\sketch_jun7a.ino:23:10: error: #include expects "FILENAME" or <FILENAME>
   23 | #include SPI.h&gt;
      |          ^~~
C:\Users\DELL\AppData\Local\Temp\.arduinoIDE-unsaved202357-2320-1fc3or9.1k34\sketch_jun7a\sketch_jun7a.ino:24:10: error: #include expects "FILENAME" or <FILENAME>
   24 | #include &lt;Wire.h&gt;
      |          ^
C:\Users\DELL\AppData\Local\Temp\.arduinoIDE-unsaved202357-2320-1fc3or9.1k34\sketch_jun7a\sketch_jun7a.ino:25:10: error: #include expects "FILENAME" or <FILENAME>
   25 | #include &lt;ESP8266WiFi.h&gt; // If you face any problem with this library vist here:
      |          ^
C:\Users\DELL\AppData\Local\Temp\.arduinoIDE-unsaved202357-2320-1fc3or9.1k34\sketch_jun7a\sketch_jun7a.ino:26:10: error: #include expects "FILENAME" or <FILENAME>
   26 | #include &lt;SoftwareSerial.h&gt; // https://youtu.be/oq3_xy2rCyk
      |          ^
C:\Users\DELL\AppData\Local\Temp\.arduinoIDE-unsaved202357-2320-1fc3or9.1k34\sketch_jun7a\sketch_jun7a.ino:27:10: error: #include expects "FILENAME" or <FILENAME>
   27 | #include &lt;ESP8266WebServer.h&gt;
      |          ^
C:\Users\DELL\AppData\Local\Temp\.arduinoIDE-unsaved202357-2320-1fc3or9.1k34\sketch_jun7a\sketch_jun7a.ino:28:10: error: #include expects "FILENAME" or <FILENAME>
   28 | #include &lt;ESP8266HTTPClient.h&gt;
      |          ^
C:\Users\DELL\AppData\Local\Temp\.arduinoIDE-unsaved202357-2320-1fc3or9.1k34\sketch_jun7a\sketch_jun7a.ino:29:10: error: #include expects "FILENAME" or <FILENAME>
   29 | #include &lt;Adafruit_GFX.h&gt; //Library link:https://github.com/adafruit/Adafruit-GFX-Library
      |          ^
C:\Users\DELL\AppData\Local\Temp\.arduinoIDE-unsaved202357-2320-1fc3or9.1k34\sketch_jun7a\sketch_jun7a.ino:30:10: error: #include expects "FILENAME" or <FILENAME>
   30 | #include &lt;Adafruit_SSD1306.h&gt; //Library link:https://github.com/adafruit/Adafruit_SSD1306
      |          ^
C:\Users\DELL\AppData\Local\Temp\.arduinoIDE-unsaved202357-2320-1fc3or9.1k34\sketch_jun7a\sketch_jun7a.ino:31:10: error: #include expects "FILENAME" or <FILENAME>
   31 | #include &lt;Adafruit_Fingerprint.h&gt; //Library link:https://github.com/adafruit/Adafruit-Fingerprint-Sensor-Library
      |          ^

exit status 1

Compilation error: #include expects "FILENAME" or <FILENAME>

jantje, 2023-06-08:
`#include SPI.h>`
Is not correctly formatted code
`#include <SPI.h>`

amanbindra1979, 2023-10-03:
Combination of single quotes, double quotes and / should work:
`make -j CFLAGS='-DMBEDTLS_CONFIG_FILE=\"mbedtls_config.h\"'`