Sloeber/arduino-eclipse-plugin #1146: Build failure  : subdir.mk:18: recipe for target 'sloeber.ino.cpp.o' failed

MangyDogUK, 2020-03-01:
Fresh install of Sloeber 
Basic arduino boards dont show this error, 
But when setting the project for STM32 boards (using (but not limited to) rogerclarks core) Im getting this error in the release folder.

I have hardware and libraries linked to the correct fodlers. What more do i need to do?

erdemuncuoglu, 2020-03-02:
Did you see issue #1143. Despite it is "Closed", is a highly active one. And yours seems to be related.

jantje, 2020-03-02:
looks like a duplicate indeed
Closed does not mean "fixed".
As I explained somewhere in the issue V1.8.0 is not fixable and a workaroud needs to be applied.

closed 2020-03-02 by jantje

MangyDogUK, 2020-03-02:
I'm sorry should have mentioned I managed to fix it by reattaching the
hardware libraries in preferences....

I think this can be closed....

uzi18, 2020-03-02:
@jantje  rogerclarks core for stm32 is not same core as 1.8.0, so it is not duplicate, not prebuild related
just info :)

jantje, 2020-03-02:
@uzi18 thanks for the info.
With more than 700 boards in arduino space; people should provide json files and version numbers 😒

uzi18, 2020-03-02:
maybe it is possible to add core path as info string to console log before launching make
something like "building with: %s" 

so we can always ask for buildlog

jantje, 2020-03-02:
The buildlog already contains the info because there are includes to the core folders.
Most people do not provide full build logs however.

MangyDogUK, 2020-03-02:
Like i say i managed to fix it... One of the other things i didnt do was to install the SAM packages through sloeber  

But I do still need to link each individual library folder in the library properties. Just linking to the parent library folder isnt working.

jantje, 2020-03-02:
>But I do still need to link each individual library folder in the library properties. Just linking to the parent library folder isnt working.

I have no clue what you are talking about.
Supported way to add arduino libraries are:
adding #include to the code
or with arduino->add library

The only supported action in the libraries folder is "deleting libraries" and "exclude from build"

MangyDogUK, 2020-03-02:
Nope, That doesn’t work...If I have a #include it remains unresolved unless I go into the add library command from the sloeber menu... Also IN order for that to work I also need to add the library folder to the library list under the Arduino library menu. It wont read the sub folders from just having the parent library folder listed...   Sent from Mail for Windows 10 From: jantjeSent: 02 March 2020 14:47To: Sloeber/arduino-eclipse-pluginCc: MangyDogUK; AuthorSubject: Re: [Sloeber/arduino-eclipse-plugin] Build failure : subdir.mk:18: recipe for target 'sloeber.ino.cpp.o' failed (#1146) But I do still need to link each individual library folder in the library properties. Just linking to the parent library folder isnt working.I have no clue what you are talking about.Supported way to add arduino libraries are:adding #include to the codeor with arduino->add libraryThe only supported action in the libraries folder is "deleting libraries" and "exclude from build"