Sloeber/arduino-eclipse-plugin #1147: Arduino Mega elf build is not compliant with Arduino IDE command line

jantje, 2020-03-05:
Arduino mega "c to elf" command should have the -relax option for big programs
However arduino IDE does not have this
To work around that (needed for marvin) Sloeber overwrites the platform.txt setting
https://github.com/Sloeber/arduino-eclipse-plugin/blob/5140b46e262fc43e1234dad78a4960598b666fe7/io.sloeber.core/config/post_processing_boards_default.txt#L2-L3
However it would be better to have a compiler option "c to elf" and have the user add the -relax option if needed

uzi18, 2020-03-05:
related to #270

jantje, 2020-03-05:
It would be great if you can point me to how arduino IDE is adding the --relax

jantje, 2020-03-05:
Found it here
https://github.com/arduino/arduino-cli/blob/master/legacy/builder/phases/linker.go#L82
Hardcoded based on BUILD_PROPERTIES_BUILD_MCU == "atmega2560" behind BUILD_PROPERTIES_COMPILER_C_ELF_FLAGS

uzi18, 2020-03-05:
what a workaround ;)

closed 2020-03-08 by jantje