Sloeber/arduino-eclipse-plugin #1153: folder

jandrassy, 2020-03-07:
Nightly plugin version 4.3.3.202003070521 and current HEAD generates to makefile
```
AR += \
/home/duro/eclipse/-workspace/Test3/Release/arduino.ar \

```
```
/home/duro/eclipse/-workspace/Test3/Release/arduino.ar: $(AR_OBJ)
	@echo 'Starting archiver'
```
the right folder should be `eclipse-workspace`. folder `/home/duro/eclipse` exists too.

for other folder with `-` the makefile is ok.
```
AR += \
/home/duro/runtime-EclipseApplication/Blink/Release/arduino.ar \
```

jantje, 2020-03-07:
As I have seen - being used in both project name and in folder names I guess this problem only occurs in workspace names.

jandrassy, 2020-03-08:
the problem is in .settings/org.eclipse.cdt.core.prefs
`A.BUILD.PATH/value=${eclipse_home}-workspace/Test3/Release`

part of the path is missmatched as a variable

jantje, 2020-05-04:
I can not reproduce the issue here
[image]

jandrassy, 2020-05-05:
Eclipse is installed in
/home/duro/eclipse/
and workspace is
/home/duro/eclipse-workspace/
so something matches ${eclipse_home} in /home/duro/eclipse-workspace/ and replaces it to
${eclipse_home}-workspace/
resulting in
/home/duro/eclipse/-workspace/

jantje, 2020-05-05:
I'll try this specific combination.
```
Eclipse is installed in XXX/eclipse/
and workspace is  XXX/eclipse-workspace/
```
Seems like you are on unix.
If I can't reproduce in windows I'll try linux.
IMHO this seems to be a very specific case.

jandrassy, 2020-05-05:
Yes I am on Linux. 
The location and name of folder eclipse-workspace is default.
I don't know how many people install eclipse in home folder and name it eclipse. The current installer offers a folder named for example java2020-03 and creates a subfolder named eclipse in it. I deleted the java2020-03 part so I ended up with folder named eclipse. (And I name my Eclipse installation folder always eclipse.)
And it is possible, that the bug could match other variables to other paths too.

uzi18, 2020-05-05:
have linux and use almost the same workspace is in eclipse-workspace or arduino-workspace 
but in facts did not created new project for long time

jantje, 2020-06-30:
Tested on linux and windows.
I can't recreate the problem

jantje, 2020-07-07:
Anyone could recreate this issue?

closed 2020-10-23 by jantje